LibreELEC/LibreELEC.tv #6719: [BUG] samba 4.16.x

heitbaum, 2022-07-19:
2 items to be addresses with samba 4.16

- use embedded heimdal to build
- fix browse issue

```
smbclient -U libreelec -L \\localhost
Password for [WORKGROUP\libreelec]:

        Sharename       Type      Comment
        ---------       ----      -------
SMB1 disabled -- no workgroup available
```

pretoriano80, 2022-07-19:
I wonder if these are somehow related ,i mean if using embedded heimdal could also fix the browse issue .

heitbaum, 2022-07-19:
> I wonder if these are somehow related ,i mean if using embedded heimdal could also fix the browse issue .

I’m somewhat hoping. 🤞So that needs to be the first order of business.

heitbaum, 2022-08-26:
Open wet build has samba dependent on samba:host. Use this as a base to split the build of heimdal out. https://github.com/openwrt/packages/blob/master/net/samba4/Makefile

Portisch, 2022-08-31:
The solution was there all the time, found by unnamed CoreELEC member within a few hours:
```text
Note that when Samba is configured to run as an Active Directory Domain Controller, the samba binary that supplies the AD code will continue to provide DCERPC services while allowing samba-dcerpcd to provide services like SRVSVC in the same way as smbd did previously.
```
https://github.com/CoreELEC/CoreELEC/commit/d07ac771047f28b015fd65d397747caef4c511c5
Thank you for your cooperation and respect of solution origins!

heitbaum, 2022-08-31:
> The solution was there all the time, found by unnamed CoreELEC member within a few hours by reading the change log:
> 
> ```
> Note that when Samba is configured to run as an Active Directory Domain Controller, the samba binary that supplies the AD code will continue to provide DCERPC services while allowing samba-dcerpcd to provide services like SRVSVC in the same way as smbd did previously.
> ```
> 
> [CoreELEC/CoreELEC@d07ac77](https://github.com/CoreELEC/CoreELEC/commit/d07ac771047f28b015fd65d397747caef4c511c5) Thank you for your cooperation and respect of solution origins!

Hi @Portisch - without going into a he says/she says… I can categorically state that the code I submitted is my own. 

It has been on my work list for a long time; in my branch, mailing lists, logs, and in the forum, in this issue and the PR. Every bump of samba has had me looking at it again. In terms of your quoted comment - actually it is not the reason for the issue - the samba compile is not done as an “Active Directory Domain Controller”. Though you are right - that the answer was staring us all in the face… “samba-dcerpcd”

I can share the details on my working (straces/logs/…) - the lights on moment with discovery of the solution was firstly the /tmp/srvsvc in the log which did send me to read the change log again. But just adding samba-dcerpcd into the compile list = no success…. sbin - no success, then local_np.c: dyn_SAMBA_LIBEXECDIR via strings on the /usr/lib/…

So whilst timing is a coincidence - I reiterate - my work is original.

I go out of my way to ensure that attribution is correct, and update the authorship from myself to the author when it is not from me or co-authored. I have worked with @vpeter4 on PRs previously and hope to continue doing so. I do not and would not blatantly rip other peoples code off. Have a look at my branches if you need to validate my statement.

I would appreciate your confirmation and understanding that the submissions of these PRs are coincidence. 

Note: I don’t have a watch on the coreelec or other downstream repos; so was unaware of the submission until ray commented on my PR.

@vpeter4 - I would be very happy for you to submit your PR to upstream (instead of mine) yours was first after all :-)

Portisch, 2022-08-31:
I am sorry that we left you in dark for months and just 24h after we fixed it you published your "own" solution. It just shows LE behavior did not and will not change in future. Get what ever is possible and give a f* about others....

Again, I am sorry, we are just a fork, not member of LE.

closed 2022-09-01 by heitbaum