LibreELEC/LibreELEC.tv #6723: Problem installing LibreElec via Noobs on RaspberryPi 4

Engztrom, 2022-07-20:
I get an error saying "Error downloading meta file", for "https://releases.libreelec.tv/noobs/LibreELEC_RPi4/LibreELEC_RPi4.png" 
and other files.
If instead trying to open this file directly in the built-in brower it says the certificate is too old, 
but if ignoring that I can open it there. 
But how do I ignore a too old certificate when installing? 
(I assume it's the same problem for the installation, even though it only says "Error downloading meta file" there).

[image]

cunlem, 2023-01-08:
For me the link resolves to
<https://ftp.fau.de/libreelec/noobs/LibreELEC_RPi4/LibreELEC_RPi4.png>
and produces no error

Perhaps a mirror in your location has an expired certificate and needs to be either fixed or removed from the list.

HiassofT, 2023-01-08:
thanks a lot for checking and reporting back, I'm closing this issue now as it's no longer reproducible

closed 2023-01-08 by HiassofT