LibreELEC/LibreELEC.tv #6737: [BUG] Rock Pi 4 Bluetooth adapter not found

vasilenkoalexey, 2022-07-24:
### Describe the bug
After installation and the first boot, everything works fine, BT devices are detected and connected. But after the first reboot - got the message "Bluetooth adapter not found". This is the situation for versions 10.0.0 \10.0.1\10.0.2. For version 9.2.6 everything works fine.

### To Reproduce
Steps to reproduce the behavior:
1. Reboot after installation

### Informations
 - 10.0.0\10.0.1\10.0.2
 - Rock Pi 4

### Log file
[kodi.log](https://pastebin.com/sMrEex7D)

### Additional context

heitbaum, 2022-07-25:
Could you please try the latest nightly LE10 (you can move between the 10.0.0/1/2 and the nightly without issue of database version issues.)
- https://test.libreelec.tv/10.0/Generic/Generic/LibreELEC-Generic.x86_64-10.0-nightly-20220723-27bd68e.img.gz
- https://test.libreelec.tv/10.0/Generic/Generic/

vasilenkoalexey, 2022-07-25:
> * https://test.libreelec.tv/10.0/Rockchip/rock-pi-4/LibreELEC-RK3399.arm-10.0-nightly-20220724-27bd68e-rock-pi-4.img.gz

Still reproduced [kodi.log](https://pastebin.com/2KreK2uY) Also [dmesg](https://pastebin.com/QbnaHg1V) output

heitbaum, 2022-07-25:
Please check from prompt
```
bluetoothctl
  list
  show
  devices
  scan on
```

vasilenkoalexey, 2022-07-25:
After installation and **BEFORE** first reboot (and with pairing my controller):
```
Agent registered
[bluetooth]#   list
Controller 6C:21:A2:37:C5:AD LibreELEC [default]
[bluetooth]#   show
Controller 6C:21:A2:37:C5:AD (public)
        Name: LibreELEC
        Alias: LibreELEC
        Class: 0x000c0000
        Powered: yes
        Discoverable: no
        DiscoverableTimeout: 0x000000b4
        Pairable: yes
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v1D6Bp0246d0538
        Discovering: yes
        Roles: central
        Roles: peripheral
Advertising Features:
        ActiveInstances: 0x00 (0)
        SupportedInstances: 0x05 (5)
        SupportedIncludes: tx-power
        SupportedIncludes: appearance
        SupportedIncludes: local-name
[bluetooth]#   devices
Device 98:B6:E9:B9:35:AC Pro Controller
[bluetooth]#   scan on
Discovery started
```
**AFTER** first reboot
```
[bluetooth]#   list
[bluetooth]#   show
No default controller available
[bluetooth]#   devices
No default controller available
Agent registered
[bluetooth]#   scan on
No default controller available
```

chewitt, 2022-07-25:
Please use the 11.0 nightly as this has newer kernels and everything. If the same behaviour still exists, run "dmesg | paste" on first install and then after reboot, and share the URLs generated.

heitbaum, 2022-07-25:
Is this only a second boot issue? 
If a cold reboot does it behave th3 same way?

if you have a spare SD card - it would be good to test LE11 nightly.

vasilenkoalexey, 2022-07-25:
Not only a second boot issue, but all after first. BT adapter is lost **COMPLETELY** after first reboot, and next reboots does not solve this.
Ok, I will test last LE11.

vasilenkoalexey, 2022-07-25:
Still reproduced on last LE11 nightly

Before first reboot:
[dmesg](https://pastebin.com/cNCNX06p)
[kodi.log](https://pastebin.com/Sa6sk9YQ)

After first reboot:
[dmesg](https://pastebin.com/1WqQ4jaM)
[kodi.log](https://pastebin.com/4YQqyuAx)

chewitt, 2022-07-25:
Before first reboot:

```
[    7.026728] Bluetooth: hci0: BCM: chip id 130
[    7.027146] Bluetooth: hci0: BCM: features 0x0f
[    7.029280] Bluetooth: hci0: BCM4345C5
[    7.029294] Bluetooth: hci0: BCM4345C5 (003.006.006) build 0000
[    7.075841] Bluetooth: hci0: BCM4345C5 'brcm/BCM4345C5.hcd' Patch
[    7.962841] Bluetooth: hci0: BCM4345C5 Ampak_CL1.5 UART 37.4 MHz BT 5.0 [Version: Version: 0033.0080]
[    7.962857] Bluetooth: hci0: BCM4345C5 (003.006.006) build 0080
```

After first reboot:
```
[    5.686600] Bluetooth: hci0: BCM: chip id 130
[    5.686914] Bluetooth: hci0: BCM: features 0x0f
[    5.688982] Bluetooth: hci0: BCM4345C5
[    5.689012] Bluetooth: hci0: BCM4345C5 (003.006.006) build 0000
[    5.748757] Bluetooth: hci0: BCM4345C5 'brcm/BCM4345C5.hcd' Patch
[    6.592594] Bluetooth: hci0: BCM4345C5 Ampak_CL1.5 UART 37.4 MHz BT 5.0 [Version: Version: 0033.0080]
[    6.592623] Bluetooth: hci0: BCM4345C5 (003.006.006) build 0080
[    6.593302] Bluetooth: hci0: BCM: Using default device address (43:45:c5:00:1f:ac)
```

The only difference I can see in the boot log is this (otherwise boot looks clean):
```
[    6.593302] Bluetooth: hci0: BCM: Using default device address (43:45:c5:00:1f:ac)
```

If you do install and then unplug the PSU so that second boot is a 'cold' boot (not a reboot resulting in warm boot) does the issue still happen?

vasilenkoalexey, 2022-07-25:
After 'cold' boot issue doesn't happen, all works fine.

chewitt, 2022-07-25:
The one thing I did notice when looking at device trees is this:

https://github.com/torvalds/linux/blob/master/arch/arm64/boot/dts/rockchip/rk3399-rock-pi-4b.dts#L26
vs.
https://github.com/torvalds/linux/blob/master/arch/arm64/boot/dts/rockchip/rk3399-rock-pi-4b-plus.dts#L34

For kicks you could try compiling an image/kernel that modifies the 4b dts to use the same line as the 4b-plus.

However the better approach is probably to report the issue to the maintainers via the linux-rockchip mailing list. As the issue only exhibits with warm (re)boot the issue resides in either the kernel or perhaps something in the boot firmware (u-boot, and vendor things that contribute to it) - perhaps something not being (re)set correctly during the power down or power on process for SDIO devices.

vasilenkoalexey, 2022-07-27:
I just try Armbian 22.08.0 minimal image from [https://github.com/radxa-build/rockpi-4b/releases/tag/20220701-0226](https://github.com/radxa-build/rockpi-4b/releases/tag/20220701-0226) and BT work fine after warm reboot. Moreover, I am sure that bt will work on other images based on the same kernel.
If it is assumed that the problem is in the kernel or boot firmware, and not in Libreelec image, then how can this be proved?

chewitt, 2022-07-27:
Emgineering method (test one change at a time and observe the effect), e.g. boot LE using the Radxa kernel config and patchset, or using the Radxa u-boot version and patchset. Look for differences in power related kernel defconfig (u-boot configs are likely the same upstream ones).

vasilenkoalexey, 2022-07-28:
I found this issue [bluez needs a workaround for HCIs with the HCI_QUIRK_INVALID_BDADDR quirk](https://github.com/bluez/bluez/issues/107) and some info [here](https://forum.armbian.com/topic/13601-orange-pi-4-kernel-5xx-rt5651-sound-and-bluetooth-fixed/?do=findComment&comment=98961). Really, according to [btbcm.c](https://github.com/torvalds/linux/blob/6e7765cb477a9753670d4351d14de93f1e9dbbd4/drivers/bluetooth/btbcm.c#L59)  and [    `6.593302] Bluetooth: hci0: BCM: Using default device address (43:45:c5:00:1f:ac)` BT device marked as `HCI_QUIRK_INVALID_BDADDR` So... perhaps problem can be solved if replace BCM4345C5.hcd by another from Armbioan image (for example) that contains a different MAC BT address...

vasilenkoalexey, 2022-07-30:
What about to add this workaround patch? [https://github.com/armbian/build/blob/993f2e6bbf72b4846c192649bab3ab0b575b9140/patch/kernel/archive/rockchip-5.18/4016-workaround-broadcom-bt-serdev.patch](https://github.com/armbian/build/blob/993f2e6bbf72b4846c192649bab3ab0b575b9140/patch/kernel/archive/rockchip-5.18/4016-workaround-broadcom-bt-serdev.patch)

chewitt, 2022-07-30:
It would be academically interesting to know if it works/solves the issue for you, but since the patch would be applied to all RK devices we probably wouldn't accept it into our codebase. It would be a good starting point for a discussion with the upstream linux-wireless maintainers though.

knaerzche, 2022-09-12:
@vasilenkoalexey: I'll check if I can include this patch, but as you've said: It's just an workarround.

jernejsk, 2022-09-12:
Proper solution in this case is to improve bootloader (most probably U-Boot) to patch DT with BT address. OrangePi 3 had same issue and this was the solution.

vasilenkoalexey, 2022-09-17:
For me, If you have idea and plans for proper solution, it would be better implement it instead of workaround (even if it takes longer)

heitbaum, 2023-12-02:
@vasilenkoalexey is this still occurring on LE12 nightlies?

heitbaum, 2024-03-11:
Closing as stale. Happy to reopen if this is still and issue and tested with latest LE12 20240311-86dba4b images.

closed 2024-03-11 by heitbaum