LibreELEC/LibreELEC.tv #6742: [BUG][LE11]addon pvr.hts crashes kodi

Flattermann78, 2022-07-27:
### Describe the bug
When pvr.hts installed, LE11 ([LibreELEC-RPi4.arm-11.0-nightly-20220726-db9a5bc) crashes after boot while starting the UI (last thing is the start "bing") and a reboot is executed.

### To Reproduce
Steps to reproduce the behavior:
1. install pvr.hts (addon for tvheadend)
2. start kodi

### Informations
 - LE Version: 11
 - Hardware Platform: RPi4

### Log file
http://ix.io/45QY

Last lines of log:
022-07-27 20:53:45.183 T:1153     INFO <general>: Running the application...
2022-07-27 20:53:45.190 T:1153     INFO <general>: starting zeroconf publishing
2022-07-27 20:53:45.191 T:1180     INFO <general>: ES: Starting UDP Event server on port 9777
2022-07-27 20:53:45.191 T:1180     INFO <general>: UDP: Listening on port 9777 (ipv6 : true)
2022-07-27 20:53:45.191 T:1153    ERROR <general>: JSONRPC Server: Failed to connect to sdpd
2022-07-27 20:53:45.191 T:1153     INFO <general>: JSONRPC Server: Successfully initialized
2022-07-27 20:53:45.195 T:1153     INFO <CWebserver[8080]>: Started
2022-07-27 20:53:45.202 T:1171     INFO <general>: Register - new cec device registered on cec->Linux: CEC Adapter (0000:0000)
2022-07-27 20:53:45.213 T:1153     INFO <general>: CWSDiscoveryListenerUDP::Start - Started
2022-07-27 20:53:45.451 T:1164    ERROR <general>: Repository add-on repository.zachmorris uses old schema definition for the repository extension point! This is no longer supported, please update your addon to use <d>
2022-07-27 20:53:45.451 T:1164    ERROR <general>: Repository add-on repository.zachmorris does not have any directory and won't be able to update/serve addons! Please fix the addon.xml definition
2022-07-27 20:53:46.062 T:1167     INFO <general>: AddOnLog: pvr.hts: starting PVR client
2022-07-27 20:53:46.087 T:1167     INFO <general>: PVR Manager: Starting
2022-07-27 20:53:46.091 T:1198     INFO <general>: AddOnLog: pvr.hts:   Received permissions:
2022-07-27 20:53:46.091 T:1198     INFO <general>: AddOnLog: pvr.hts:   administrator              : 1
2022-07-27 20:53:46.091 T:1198     INFO <general>: AddOnLog: pvr.hts:   HTSP streaming             : 1
2022-07-27 20:53:46.091 T:1198     INFO <general>: AddOnLog: pvr.hts:   HTSP DVR                   : 1
2022-07-27 20:53:46.091 T:1198     INFO <general>: AddOnLog: pvr.hts:   Failed/aborted DVR         : 1
2022-07-27 20:53:46.091 T:1198     INFO <general>: AddOnLog: pvr.hts:   anonymous HTSP only        : 0
2022-07-27 20:53:46.091 T:1198     INFO <general>: AddOnLog: pvr.hts:   global connection limit    : 0
2022-07-27 20:53:46.091 T:1198     INFO <general>: AddOnLog: pvr.hts:   DVR connection limit       : 0
2022-07-27 20:53:46.091 T:1198     INFO <general>: AddOnLog: pvr.hts:   streaming connection limit : 0
2022-07-27 20:53:46.093 T:1198     INFO <general>: AddOnLog: pvr.hts:   Available streaming profiles:
2022-07-27 20:53:46.093 T:1198     INFO <general>: AddOnLog: pvr.hts:   Name: htsp, Comment: HTSP Default Stream Settings
2022-07-27 20:53:46.093 T:1198     INFO <general>: AddOnLog: pvr.hts:   Name: webtv-h264-aac-matroska, Comment: WEBTV profile H264/AAC/Matroska
2022-07-27 20:53:46.093 T:1198     INFO <general>: AddOnLog: pvr.hts:   Name: webtv-h264-vorbis-mp4, Comment: WEBTV profile H264/Vorbis/MP4
2022-07-27 20:53:46.093 T:1198     INFO <general>: AddOnLog: pvr.hts:   Name: htsp, Comment: HTSP Default Stream Settings
2022-07-27 20:53:46.094 T:1198     INFO <general>: AddOnLog: pvr.hts:   Name: webtv-h264-aac-matroska, Comment: WEBTV profile H264/AAC/Matroska
2022-07-27 20:53:46.094 T:1198     INFO <general>: AddOnLog: pvr.hts:   Name: webtv-h264-aac-mpegts, Comment: WEBTV profile H264/AAC/MPEG-TS
2022-07-27 20:53:46.094 T:1198     INFO <general>: AddOnLog: pvr.hts:   Name: webtv-vp8-vorbis-webm, Comment: WEBTV profile VP8/Vorbis/WEBM
2022-07-27 20:53:46.094 T:1198     INFO <general>: AddOnLog: pvr.hts:   Name: htsp, Comment: HTSP Default Stream Settings
2022-07-27 20:53:46.094 T:1198     INFO <general>: AddOnLog: pvr.hts:   Name: webtv-h264-aac-mpegts, Comment: WEBTV profile H264/AAC/MPEG-TS
2022-07-27 20:53:46.094 T:1198     INFO <general>: Skipped 1 duplicate messages..
2022-07-27 20:53:46.094 T:1198     INFO <general>: AddOnLog: pvr.hts:   Name: htsp, Comment: HTSP Default Stream Settings
2022-07-27 20:53:46.094 T:1198     INFO <general>: AddOnLog: pvr.hts:   Name: webtv-h264-aac-mpegts, Comment: WEBTV profile H264/AAC/MPEG-TS
2022-07-27 20:53:46.094 T:1198     INFO <general>: AddOnLog: pvr.hts:   Name: webtv-vp8-vorbis-webm, Comment: WEBTV profile VP8/Vorbis/WEBM
2022-07-27 20:53:46.094 T:1198     INFO <general>: Skipped 1 duplicate messages..
2022-07-27 20:53:46.094 T:1198     INFO <general>: AddOnLog: pvr.hts:   Name: webtv-h264-aac-matroska, Comment: WEBTV profile H264/AAC/Matroska
2022-07-27 20:53:46.094 T:1198     INFO <general>: AddOnLog: pvr.hts:   Name: webtv-vp8-vorbis-webm, Comment: WEBTV profile VP8/Vorbis/WEBM
2022-07-27 20:53:46.094 T:1198     INFO <general>: AddOnLog: pvr.hts:   Name: webtv-h264-aac-matroska, Comment: WEBTV profile H264/AAC/Matroska
2022-07-27 20:53:46.094 T:1198     INFO <general>: AddOnLog: pvr.hts: Request async EPG (3 days)
2022-07-27 20:53:46.097 T:1198     INFO <general>: AddOnLog: pvr.hts: Async updates requested
2022-07-27 20:53:46.124 T:1167     INFO <general>: WEATHER: Downloading weather

### Additional context
Deleting the addon on the SD card leads to a working boot and usable installation.

chewitt, 2022-07-27:
Known and resolved with removing the current binary (and cached copy) and reinstalling the add-on. Nothing to fix as LE11 nightlies are still pre-alpha. Life on the bleeding edge sometimes comes with blood :)

Flattermann78, 2022-07-27:
Thank you! I wasn't aware of the cached copy.

closed 2022-07-27 by Flattermann78