LibreELEC/LibreELEC.tv #6743: Please upgrade or patch the kernel to fix CVE-2022-34918

mossroy, 2022-07-27:
See https://security-tracker.debian.org/tracker/CVE-2022-34918 and/or https://www.randorisec.fr/crack-linux-firewall/. It allows escalation of privileges.

It has been fixed by https://github.com/torvalds/linux/commit/7e6bc1f6cabcd30aba0b11219d8e01b952eacbb6, which is included in version 5.10.130 of mainline linux kernel, see https://mirrors.edge.kernel.org/pub/linux/kernel/v5.x/ChangeLog-5.10.130 (search for netfilter). So it probably affects LibreELEC 10 (kernel 5.10.123)

Upgrading the kernel to a version >=5.10.130 (or manually applying the patch) should be enough.

The fix has also been included in version 5.18.11 (see https://mirrors.edge.kernel.org/pub/linux/kernel/v5.x/ChangeLog-5.18.11), so it probably does not affect LibreELEC 11 (kernel 5.18.14)

heitbaum, 2022-07-28:
> See https://security-tracker.debian.org/tracker/CVE-2022-34918 and/or https://www.randorisec.fr/crack-linux-firewall/. It allows escalation of privileges.
> 
> It has been fixed by [torvalds/linux@7e6bc1f](https://github.com/torvalds/linux/commit/7e6bc1f6cabcd30aba0b11219d8e01b952eacbb6), which is included in version 5.10.130 of mainline linux kernel, see https://mirrors.edge.kernel.org/pub/linux/kernel/v5.x/ChangeLog-5.10.130 (search for netfilter). So it probably affects LibreELEC 10 (kernel 5.10.123)
> 
> Upgrading the kernel to a version >=5.10.130 (or manually applying the patch) should be enough.
> 
> The fix has also been included in version 5.18.11 (see https://mirrors.edge.kernel.org/pub/linux/kernel/v5.x/ChangeLog-5.18.11), so it probably does not affect LibreELEC 11 (kernel 5.18.14)

Does not really impact LE - as the processes are running as root. 
But as part of our regular updates - there is a PR raised #6734 - which we waited for [retbleed ](https://en.wikipedia.org/wiki/Retbleed)

mossroy, 2022-07-28:
OK, great!

closed 2022-08-05 by CvH