LibreELEC/LibreELEC.tv #6754: [BUG] Installing the locale addon breaks the VDR (live plugin)

jhelgert, 2022-07-30:
### Describe the bug
The VDR PVR backend (with installed vdr-live plugin) isn't able to start once the locale addon is installed and configured accordingly:

``` none
Jul 30 09:24:47.447680 LibreELEC vdr[808]: [808] VDR version 2.6.1 started
Jul 30 09:24:47.447702 LibreELEC vdr[808]: [808] switched to user 'root'
Jul 30 09:24:47.450758 LibreELEC vdr[808]: [808] codeset is 'UTF-8' - known
Jul 30 09:24:47.473507 LibreELEC vdr[808]: [808] found 0 locales in /storage/.kodi/addons/service.multimedia.vdr-addon/locale
...
Jul 30 09:24:47.513193 LibreELEC vdr[808]: [808] loading plugin: /storage/.kodi/addons/service.multimedia.vdr-addon/plugin/libvdr-live.so.2.6.1
Jul 30 09:24:47.583529 LibreELEC sh[808]: terminate called after throwing an instance of 'std::runtime_error'
Jul 30 09:24:47.583529 LibreELEC sh[808]:   what():  locale::facet::_S_create_c_locale name not valid
Jul 30 09:24:47.584306 LibreELEC systemd[1]: service.multimedia.vdr-addon.service: Main process exited, code=killed, status=6/ABRT
Jul 30 09:24:47.597232 LibreELEC systemd[1]: service.multimedia.vdr-addon.service: Failed with result 'signal'.
Jul 30 09:24:49.599650 LibreELEC systemd[1]: service.multimedia.vdr-addon.service: Scheduled restart job, restart counter is at 1.
Jul 30 09:24:49.600038 LibreELEC systemd[1]: Stopped vdr.
```

### To Reproduce
Steps to reproduce the behaviour:
1. Install the VDR PVR backend and enable the live plugin
2. Install the locale addon and set locale to `de_DE` and reboot
3. VDR no longer starts

### Informations
 - LE Version: 11.0 nightly-20220729-34b5937
 - Hardware Platform: Generic

### Log file
http://ix.io/465N

### Additional context
Installing the locale addon is required in order to have german time format inside the vdr-live plugin.

mglae, 2022-07-31:
Is working fine for me with German locale after test installing vdr-addon v10.80.7.120 on generic-legacy.

```
Jul 30 09:24:47.473507 LibreELEC vdr[808]: [808] found 0 locales in /storage/.kodi/addons/service.multimedia.vdr-addon/locale
```
This is odd. I'm seeing `found 28 locales`. Please check the path if the locale files are missing and reinstall the addon in this case.

jhelgert, 2022-08-02:
> Please check the path if the locale files are missing and reinstall the addon in this case.

I checked it again on a freshly installed (nightly build) system. The files are there:

[image]

But reinstalling the addon yields the same error message as above once I set the locale inside the plugin.

mglae, 2022-08-06:
Thanks for testing it again.

I can confirm it is failing on Generic (GBM) whereas it is working on Generic-legacy (X11).

mglae, 2022-08-15:
This is a Locale Addon issue not generating locale files on Generic /GBM).

I've committed a fix.

closed 2022-08-15 by CvH

jhelgert, 2022-08-16:
@mglae Thanks! 

Just one question: How can I update the locale addon to the new rev 103? I already updated LE to the latest nightly build [20220816-937690a](https://test.libreelec.tv/11.0/Generic/Generic/) which should include the fix. However, the locale addon is still on 102. And triggering a "search for updates" inside the addons menu doesn't find the new version.

heitbaum, 2022-08-16:
It’s currently queued to build. ~8 hours away - then will need to be pushed to the addons server.

heitbaum, 2022-08-17:
> @mglae Thanks!
>
> Just one question: How can I update the locale addon to the new rev 103? I already updated LE to the latest nightly build [...]

It has been pushed to the addon's server - you should see it soon

jhelgert, 2022-08-17:
> It has been pushed to the addon's server - you should see it soon

Yes, I already installed it and it works like a charm now. Thanks again :)!

PR #6807 (2022-08-15): locale addon: fix call of localedef, update language and settings format, REV 103
Locale addon silently fails to create locale data with localedef below `/storage/.kodi/addons/service.locale/locpath/*`

After enabling output capture the error messages show:
```
gzip: invalid magic
gzip: invalid magic
[error] character map file `UTF-8' not found: No such file or directory
[warning] LC_NAME: field `name_gen' not defined
[warning] LC_ADDRESS: field `country_post' not defined
[warning] LC_ADDRESS: field `country_isbn' not defined
[warning] LC_IDENTIFICATION: field `audience' not defined
[warning]  [... cut]