LibreELEC/LibreELEC.tv #6758: [BUG] Kodi crashes when CPU0 current freq is not available

KOPRajs, 2022-08-02:
### Describe the bug

When Kodi cannot read _/sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq_ it crashes.

On big.LITTLE systems (like the RK3399) it is usually favorable to disable the "LITTLE" part of the CPU and leaving only the "big" cores enabled for using with LibreELEC (you get better speed and less heat). I'm disabling it by running the following script in _/storage/.config/autostart.sh_:
```
echo "powersave" > /sys/devices/system/cpu/cpufreq/policy0/scaling_governor
echo "performance" > /sys/devices/system/cpu/cpufreq/policy4/scaling_governor
echo "0" > /sys/devices/system/cpu/cpu0/online
echo "0" > /sys/devices/system/cpu/cpu1/online
echo "0" > /sys/devices/system/cpu/cpu2/online
echo "0" > /sys/devices/system/cpu/cpu3/online

```

Until recently it was working great, but in the current nightly build (2022-08-02) the Kodi crashes when you try to open the "Settings" -> "System information" -> "CPU".

```
LibreELEC:~ # cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq 
cat: read error: Device or resource busy

```

Probably the new kernel returns "read error" while the older used to return just 0.
Anyway I think it should not cause Kodi to crash.

### To Reproduce
Steps to reproduce the behavior:
1. run "echo "0" > /sys/devices/system/cpu/cpu0/online"
2. open "Settings" -> "System information" -> "CPU"
3. Kodi crashes

### Informations
 - LE Version: 11 (2022-08-02)
 - Hardware Platform: RK3399

### Log file

### Additional context

lrusak, 2022-08-03:
Please report to Kodi instead. Thanks.

heitbaum, 2023-01-15:
Closing as a Kodi issue. Ticket is raised with Kodi.

closed 2023-01-15 by heitbaum