LibreELEC/LibreELEC.tv #6767: [BUG] Video stutter on RPi4

Flattermann78, 2022-08-05:
### Describe the bug

After around 10 minutes, video starts to stutter and losses audio using pvr plugin for tvheadend

### To Reproduce
Steps to reproduce the behavior:
1. Start video using tvheadend (pvr.hts)
2. Wait
3. See error

### Informations
 - LE Version: LibreELEC-RPi4.arm-10.0-nightly-20220725-d6f0921.img.gz, but we see the issues with the stable version as well so we switched to the nightly build
 - Hardware Platform: RPi4

### Log file

https://paste.kodi.tv/okopazumuc
### Additional context
Issues start with line 
2022-08-05 22:32:30.321 T:959     DEBUG <general>: CVideoReferenceClock: detected 1 vblanks, missed 2, refreshrate might have changed
in the log file

Flattermann78, 2022-08-05:
Some more logs
https://drive.google.com/file/d/1UpO3VYyIbOGxQRL60HiZ2LMdEhVON982/view?usp=sharing

Flattermann78, 2022-09-17:
I updated to the most recent version (LibreELEC-RPi4.arm-10.0-nightly-20220916-591da77) but the issues stays the same. The audio and video are out of sync with an ever increasing lack of timing. After some time, video stops completely. This holds true for all videos (DVD, media files or streams from tvheadend).

I created a log from the DVD replay:
https://drive.google.com/file/d/1wXxh7VwAa_Z5bkeKBKeHuqig9-ZOWL1f/view?usp=sharing

Flattermann78, 2022-09-27:
I'm experiencing the issue also with LE 11

heitbaum, 2023-10-16:
Closing. Please reopen if still and issue. Please share an update.