LibreELEC/LibreELEC.tv #6782: [BUILD][ERROR] PCRE2 error when building RPi4 image on Ubuntu 20.04 aarch64

pretoriano80, 2022-08-10:
### Describe the bug
Build fails due to pcre2 error

### Informations
 - LE Version: LE11 
 - Hardware Platform: RPi4
 -  Build system: Ubuntu 20.04 aarch64 

### Log file
[78.log](https://github.com/LibreELEC/LibreELEC.tv/files/9303539/78.log)

heitbaum, 2022-08-10:
Can you please try adding ncurses to the depends

https://github.com/LibreELEC/LibreELEC.tv/blob/16f075c66f0c9c6406ba0304c26defa6b5579e17/packages/devel/pcre2/package.mk#L11

pretoriano80, 2022-08-11:
I had to add ncurses to both host and target depends (it failed when i added only to target ).So far it seem to work ,but the build didn't completed yet,so i will let you know when it does.

pretoriano80, 2022-08-11:
pcre2 build log -> 
[130.log](https://github.com/LibreELEC/LibreELEC.tv/files/9306619/130.log)

`DONE 130 3 install pcre2:target 32.0203 2.8321 117 29.7113 1660197438.6561 1660197468.3674 /home/ubuntu/LibreELEC.tv/build.LibreELEC-RPi4.arm-11.0-devel/.threads/logs/130.log`

pretoriano80, 2022-08-11:
Nope, pcre2:host still fails...

pretoriano80, 2022-08-11:
Here are the logs after adding `PKG_DEPENDS_TARGET="toolchain ncurses"`
[CMakeOutput.log](https://github.com/LibreELEC/LibreELEC.tv/files/9307093/CMakeOutput.log)
[CMakeError.log](https://github.com/LibreELEC/LibreELEC.tv/files/9307095/CMakeError.log)
[232.log](https://github.com/LibreELEC/LibreELEC.tv/files/9307096/232.log)

pretoriano80, 2022-08-11:
@heitbaum and @SupervisedThinking  I've tried all your suggestions but pcre2:host still fails .

What puzzles me is that if i use `PROJECT=RPi DEVICE=RPi4 ARCH=arm scripts/build pcre2` then pcre2 builds just fine.But trying the same with pcre2:host fails.

I tried to reset the master branch,distclean ,etc...no dice.

SupervisedThinking, 2022-08-11:
@pretoriano80 well I guess we need host build opts too 🤦🏻‍♂️😅

heitbaum, 2022-08-12:
@pretoriano80 - can you please check your output on your build host

```
--   Link pcre2grep with libz ........ : Library not found
--   Link pcre2grep with libbz2 ...... : Library not found
--   Link pcre2test with libeditline . : Library not found
--   Link pcre2test with libreadline . : Library not found
```

The build on the my Oracle Ampere node (within the docker) for both :host and :target have the above output.

https://github.com/LibreELEC/LibreELEC.tv/pull/6785#issuecomment-1211847418

pretoriano80, 2022-08-12:
@SupervisedThinking

pretoriano80, 2022-08-12:
I closed it by mistake .

pretoriano80, 2022-08-12:
Just to double check ,I've done another distclean &build image,it's all good and everything compiled just fine.

PR #6785 (2022-08-15): pcre2: fix build on aarch64 without ncurses
- Fixes #6782 build on aarch64 of pcre2

@pretoriano80 please check this works for you too.
closed 2022-08-15 by CvH