LibreELEC/LibreELEC.tv #6783: [BUG] Regression playing certain videos on RPi4

rschupp, 2022-08-11:
### Describe the bug
Regression playing certain videos on RPi4

### To Reproduce
Steps to reproduce the behavior:

1. Play any 2 minute news clip ("Tagesschau") from German public tv like
http://media.tagesschau.de/video/100s/2022/0811/TV-100s-0807.webxl.h264.mp4
(it doesn't matter if Kodi reads it from the web or you download it to a local file)

2. The problem is seen in the last 10 seconds: they show a weather map animation followed by a short outro (similar to the intro). In the bad version you can hear the outro fanfare on audio, but the video still shows the last frame of the weather map. In fact, videoplayer doesn't stop after the end of the clip and continues to show the weather map until stopped explicitly.

### Informations
 - LE Version: LE 11.0 nightly
 - Hardware Platform: RPi4

### Log file
http://ix.io/47gm

### Additional context
This is a regression, using nightly builds from https://test.libreelec.tv/11.0/RPi/RPi4/ I narrowed it down to

good: LibreELEC-RPi4.arm-11.0-nightly-20220723-0da8a26.img.gz 
bad: LibreELEC-RPi4.arm-11.0-nightly-20220724-f32d79d.img.gz

Unfortunately 0da8a26..f32d79d contains both an update for the RPi linux kernel and the RPi specific patch for ffmpeg...

heitbaum, 2023-09-04:
Is this still occurring with le11?/le12?

rschupp, 2023-09-05:
> Is this still occurring with le11?/le12?

Sometimes (when using a local copy). But the above URL doesn't work anymore and news videos shown by e.g. `plugin.video.ardundzdf` exhibit no problems, so closing this issue.