LibreELEC/LibreELEC.tv #6814: bananapi-m2u powers off on boot

nileio, 2022-08-17:
I have tested many images on my board bananapi-m2u (ultra) - AllWinner R40
and none of them worked. I tried the archive versions 9.5 above, all until the final release 10.0.2
all of them have got the same issue below
### Describe the bug
the board powers off on first boot
### To Reproduce
Steps to reproduce the behavior:
the image is flashed, power on , Logo comes along and the print out of paritioning , etc.
It then restarts automatically 
on restart the board power turns off completely ! it simply switches off the power to the board and then nothing.

### Informations
 - LE Version: [e.g. 10.0.2]
 - Hardware Platform: [e.g. bananapi-m2u]

### Log file

### Additional context
I have tried to isolate the problem but there is nothing .. tried HDMI output on 3 different TVs and all produce the same result. I tried the LibreElec Image Maker on Windows and used it to Write the image, I also used my Mac to burn images ..
I got no idea what is the problem. 

Could it be a power issue ? I use the OTG power and it works fine with other images.
If so, why does other images (for example Ubuntu Mate dont have this problem ?) How can I get any debug info.
It is obviosuly a major problem but I tried different power adapters producing required outpout 5 V , 2
and same problem with LibreElec.
I was really hoping to get LibreElec to run on this board !

Thanks for your help

jernejsk, 2022-08-20:
I tested R40 several times, including LE10 before release and it always worked. I never powered it via OTG, though. It's possible that voltage dips below acceptable level when GPU is being used by Kodi. Try with barrel power supply, rated for at least 3 A.

You can get useful (boot) logs via serial console. If you remove "quiet" and add "systemd.debug-shell=1" in extlinux/extlinux.conf, you'll get even more useful logs and possibility to execute commands, if system doesn't crash.

jernejsk, 2022-08-21:
I did some testing and I can confirm that issue is in powering via USB micro. It works fine for me when board is powered via barrel connector. If I power it via OTG port, it fails just as you described on LE10. It works fine in either case with LE11 nightly image.

I'm not sure if Kodi or kernel is at fault, but in either case, that won't be changed in LE10 release. As mentioned above, use LE11 nightly images or better yet, switch to barrel connector for powering the board.

closed 2022-08-21 by jernejsk

nileio, 2022-08-24:
Thanks alot for feedback @jernejsk  .. I will test LE11 nightly promptly. barrel connector is faulty or so on my board and I tried it multiple times the last thing I like to do is buy a new one and it ends up same issue with that connector. Thanks for testing and providing support this board is little flaky to be frank!

nileio, 2022-08-24:
can you pls provide the link to the nightly image you tried ?! @jernejsk I can't find the LE11 nightly for this image , there is bananapi-m2p which is a different board.. please confirm!

nileio, 2022-08-24:
I removed 'quite' as suggested and added "systemd.debug-shell=1" 

here is a screenshot of where things stop

[image]

nileio, 2022-08-25:
@jernejsk awesome thanks heaps!!

jernejsk, 2022-08-25:
does it work for you now?

nileio, 2022-08-25:
@jernejsk It worked with the LE11 nightly and booted up, however, the core issue are: 

1. tried to run a video and it completely froze ( the screen either go just Green and nothing or the system freeze and then I had to reset the board. I tried with two different monitors with HDMI). The video is just simply using my local IPTV URL (using Tvheadend PVR - yet to try with a local video or audio)
2. Wireless is not available even though it says during the boot `[OK] enabled Wireless Service`
Bluetooth seem to work and I think the bluetooth Audio OK (however, haven't tested it yet)

appreciate your help. do you need any logs / etc. ? I am keen to try fix this board to see if it I can use it to run simply TV !

jernejsk, 2022-08-25:
Enable debug logs in Kodi and and open new issue, together with url created with pastekodi command (it will take all relevant logs and put it on pastebin site.)