LibreELEC/LibreELEC.tv #6845: [BUG] Crash with nettle / gnutls 3.7.7

heitbaum, 2022-08-28:
https://gist.github.com/polojoe/b18984a72964b5dd8582cdc76eaf517e

[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
Core was generated by `/usr/lib/kodi/kodi.bin --standalone -fs'.
Program terminated with signal SIGBUS, Bus error.
#0  0xf4ffe774 in _nettle_ghash_set_key () from /usr/lib/libnettle.so.8
[Current thread is 1 (Thread 0xbb5ff2c0 (LWP 1458))]

Thread 1 (Thread 0xbb5ff2c0 (LWP 1458)):
#0  0xf4ffe774 in _nettle_ghash_set_key () from /usr/lib/libnettle.so.8
#1  0xf4ffea0c in nettle_gcm_set_key () from /usr/lib/libnettle.so.8
#2  0xf7a4dcb0 in ?? () from /usr/lib/libgnutls.so.30
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

https://forum.libreelec.tv/thread/26002-kodi-crash-when-starting-iptv-channel/?postID=172254

https://github.com/LibreELEC/LibreELEC.tv/pull/6750

https://git.lysator.liu.se/nettle/nettle/-/compare/nettle_3.8_release_20220602...nettle_3.8.1_release_20220727?from_project_id=12

https://github.com/gnutls/gnutls/compare/3.7.6...3.7.7
https://github.com/gnutls/gnutls/compare/3.7.7...master

heitbaum, 2022-08-28:
There are some changes in the 3.7.6 to 3.7.7 with gnutls and some further changes since 3.7.7. 
From the backtrace - unable to determine the calling function in gnutls.

heitbaum, 2022-08-28:
Currently not able to reproduce the crash. (x86_64), will need to test further.

vpeter4, 2022-08-28:
I think this is responsible commit at least on arm https://github.com/gnutls/gnutls/commit/2f61f102169e4d6652c9b82246353cd276366809#diff-b95ca7b81bed82151a4d3dc277e8282a9009caaa4976d4a9c71056544c778b1e

heitbaum, 2022-08-29:
Image https://test.libreelec.tv/11.0/RPi/RPi4/LibreELEC-RPi4.arm-11.0-nightly-20220829-90a6f3e.img.gz includes the fix.

heitbaum, 2022-08-29:
> I think this is responsible commit at least on arm [gnutls/gnutls@2f61f10#diff-b95ca7b81bed82151a4d3dc277e8282a9009caaa4976d4a9c71056544c778b1e](https://github.com/gnutls/gnutls/commit/2f61f102169e4d6652c9b82246353cd276366809#diff-b95ca7b81bed82151a4d3dc277e8282a9009caaa4976d4a9c71056544c778b1e)

looks like only to be some channels on the RPi - so guessing different crypto may be the issue and you find seems to concur

heitbaum, 2022-08-29:
Tested as working. Thanks @HiassofT - https://forum.libreelec.tv/thread/26002-kodi-crash-when-starting-iptv-channel/?postID=172415#post172415

closed 2022-08-29 by heitbaum