LibreELEC/LibreELEC.tv #6858: [BUG] /usr/bin/zip crashes with "zip error: Not supported (uzoff_t not same size as zoff_t)"

nvdias0, 2022-08-31:
### Describe the bug
calling zip (/usr/bin/zip) from the command line returns error
**"zip error: Not supported (uzoff_t not same size as zoff_t)"**

### To Reproduce
Steps to reproduce the behavior:
1. Login to libreelec 11 trough ssh
2. zip
3. See error

### Informations
 - LE Version: LE 11 - commit 90a6f3e 
 - Hardware Platform: RPi4

### Log file
no info on logs regarding zip

### Additional context
This has been happen from the past weeks. I can not confirm since which commit.

heitbaum, 2022-09-02:
May have found the issue - please test this zip file. https://github.com/LibreELEC/LibreELEC.tv/blob/c17753422d644cc4c971b98d7fff9aa841dfda21/zip

nvdias0, 2022-09-02:
> May have found the issue - please test this zip file. https://github.com/LibreELEC/LibreELEC.tv/blob/c17753422d644cc4c971b98d7fff9aa841dfda21/zip

it iworks  on LE 11 from yesterday.

closed 2022-09-05 by CvH