LibreELEC/LibreELEC.tv #6864: [BUG] RP4: Resuming playback fails with x264 videos encoded with --open-gop option when DRM Prime hardware acceleration is enabled.

Groschi, 2022-09-02:
### Describe the bug
On a RPi 4, resuming of some x264-encoded videos fails when the option `Allow hardware acceleration with DRM PRIME` is enabled - instead of launching fullscreen video playback i get an opaque busy spinner overlay (the spinner itself is not moving). The audio is playing though. It appears to fail on all files encoded with the `--open-gop` option. Files encoded without `--open-gop` seem to work fine. I see this happening with both LE 10.x and current LE 11 nightly builds.

### To Reproduce
Steps to reproduce the behavior:
1. Go into the player settings, enable both `Allow using DRM PRIME decoder` and `Allow hardware acceleration with DRM PRIME`
2. Download the footage i've encoded with and without `--open-gop`:
    https://drive.google.com/file/d/1L6bv6iFOga92KcAzb-axSnAYvO0-fwUk/view?usp=sharing
    https://drive.google.com/file/d/1oErnCSKS6Z2R7NR41m0f8Ew8ZPDjHdH2/view?usp=sharing
3. Play the file file `Big Buck Bunny_no-opengop.mkv`, skip ahead a minute or two, then stop playback.
4. Try resuming playback of the file from its saved position: Playback resumes as it should.
5. Repeat steps 3 & 4 with the file `Big Buck Bunny_opengop.mkv` : You get no video. Audio plays though.

### Informations
 - LE Version: 10.0.2 & 11.x nightly
 - Hardware Platform: RPi4

### Log file
http://ix.io/49rT

HiassofT, 2022-09-04:
Thanks a lot for your detailed bugreport!

I could reproduce the issue and also get similar behavior when playing with mpv (on RPiOS) or when using ffmpeg --ss if v4l2 HW decoding is enabled.

Interestingly LE9.2.8 (which uses MMAL HW decoding) resumes this file fine (also when using latest RPi firmware).

I've pinged our video decoder guru, I suspect it's something in ffmpeg / v4l2

HiassofT, 2022-10-01:
We've got a fix for the issue, it should be included in LE11 builds soon. Backport to LE10 needs a bit more work and will (hopefully) follow later

PR #6948 (2022-10-01): ffmpeg: update kodi patch
This fixes the "resume not working" issue #6864 

Also refreshed and renamed RPi kodi patches and added a patch to fix deinterlace of software decoded videos with latest ffmpeg patches

Runtime tested on RPi3B+ and RPi4

Fixes: #6864
closed 2022-10-01 by CvH