LibreELEC/LibreELEC.tv #6910: [BUG] Storage partition is read-only / cannot be modified in other OSs

erayrafet, 2022-09-22:
### Describe the bug
I am trying to add my media content directly to LibreELEC's Storage partition from my ChromeOS or Android devices and neither can change/add/modify files on it. I ran chmod 775 for the tvshows folder via SSH and still cannot copy anything inside. I factory reset LibreELEC as well.

### Informations
 - LE Version: LibreELEC-Generic.x86_64-11.0-nightly-20220920
 - Hardware Platform: HP Pavilion 13-AN0007NU
 - LibreELEC is installed on an SSD inside a USB caddy.

### Log file

### Additional context
I am running nightly builds since they support my RTL8822BE wifi card (no bluetooth though) and to test out the GBM backend.

chewitt, 2022-09-22:
Closing because this is a bug tracker not a support forum (the support forum is here: https://forum.libreelec.tv).

closed 2022-09-22 by chewitt

erayrafet, 2022-09-22:
> Closing because this is a bug tracker not a support forum (the support forum is here: https://forum.libreelec.tv).

Well, isn't this a bug?

chewitt, 2022-09-22:
Some other OS's inability to handle data write to an ext4 partition is neither a bug or our problem, but if you post in the forum there are probably people who will help investigate what's going on.