LibreELEC/LibreELEC.tv #6929: [BUG]

lbarmen, 2022-09-25:
### Describe the bug
https://www.youtube.com/watch?v=md_fkJDtu1Q
Such a flash occurs every 5-10 minutes, it makes it very difficult to watch

### To Reproduce
I use the elementum plugin and pre-download the movie

### Informations
 - LE 10.2, LE night LibreELEC-RPi4.arm-10.0-nightly-20220919-8c72424 (LE 10.1 this problem does not exist)
 - RPi4 4GB ram
 
### Log file
log file will be added later

lbarmen, 2022-09-25:
[image]

lbarmen, 2022-09-25:
[kodi.log](https://github.com/LibreELEC/LibreELEC.tv/files/9641183/kodi.log)

HiassofT, 2022-09-25:
Your kodi log shows you are using banned addons.

Remove these addons (ideally reproduce it with a complete clean LE installation) and then either use "pastekodi" on the command line or the log upload function of LE settings to post a log.

HiassofT, 2022-09-28:
Closing as banned addons are present and no new logs were provided.

Create a new issue if the issue persists with a clean LE installation

closed 2022-09-28 by HiassofT