LibreELEC/LibreELEC.tv #6930: [BUG] Boot loop after updating external NAS to Ubuntu 22.04

NetDwarf, 2022-09-27:
### Describe the bug
When opening the share of the external NAS LibreELEC crashes. The bootloop only happens if you have added a folder containing series/movies etc otherwise you are not able to do this as it crashes the moment you open the share.

### To Reproduce
I explain the crash alone (bootloop happens only if you had it working prior)

0. Do a fresh install of LE11 (not sure if exclusive to LE11)
1. Add external NAS share (Ubuntu 22.04)
2. Open that share
3. It might crash (depending on what exactly the culprit is)

### Informations
 - LE Version: LE11 (Any version basically)
 - Hardware Platform: Odroid-C2
 
 - LE Version: LE11
 - Hardware Platform: Rock64

Same behavior for both. Both restart on boot.

### Log file
The log is on LE11 220901 however I only tried downgrading to see if the error persists (it does).

https://pastebin.com/7890yh4p

### Additional context
The external NAS is accessible from other computers just fine. 
Samba version is 2:4.15.9+dfsg-0ubuntu0.2

mglae, 2022-09-28:
The logged crash is likely #6846.

Please post **debug** crash log of recent nightly

NetDwarf, 2022-09-28:
> The logged crash is likely #6846.
> 
> Please post **debug** crash log of recent nightly

Does this cause a crash at startup? I mean I test a version prior to this merge to make sure, but the version I used suddenly started failing after working fine for 2 weeks and all other versions/devices as well.

NetDwarf, 2022-09-29:
Okay here is the crash with debug logging enabled:

https://pastebin.com/81wCRUyT

This seems less conclusive than the crashlog. I am not sure the repository (repository.zachmorris) is the problem, but I would also not rule it out. At some point (before clean flashing) I removed this repository from database and it was not readded later, but the bootloop persisted.

NetDwarf, 2022-09-29:
Okay also tested LE11-20220826 on Odroid-C2 and the problem is exactly the same. I also noticed I can browse the NAS folder structure just fine inside the add source dialog, but if I access the added source it just crashes.

NetDwarf, 2022-10-07:
Okay, another idea was to change server version back and see if it works. It does. So Ubuntu 20.04 does not crash LE11 and Ubuntu 22.04 does. Using Alpine (3.16) also exhibits the same issue with LE11. The samba version that works is 4.13 and the one that crashes LE11 is 4.15. 

I add a debug crashlog later (if I hopefully find the time).

NetDwarf, 2022-10-07:
Okay that's the one according to your instructions @mglae https://pastebin.com/JSXgtjGS

I can't really read these crashlogs, but I have the suspicion that the actual crash is not logged at all.

HiassofT, 2022-10-07:
Please test with the latest nightly, you seem to be running a rather old one from end of August.

We recently updated nettle and as your installation is crashing in there there's some chance the issue might already be fixed in current nightlies

NetDwarf, 2022-10-07:
Yeah I did that because of @mglae's  first hint. The last one I tested was [this](https://test.libreelec.tv/11.0/Amlogic/odroid-c2/LibreELEC-AMLGX.arm-11.0-nightly-20220925-b1d73ad-odroid-c2.img.gz).

Nightly [221006](https://test.libreelec.tv/11.0/Amlogic/odroid-c2/LibreELEC-AMLGX.arm-11.0-nightly-20221006-f15e1de-odroid-c2.img.gz) really works tho. Man, I really got the timing right with my Samba 4.15 update. :weary: 

Thank you @HiassofT for the hint!

closed 2022-10-07 by NetDwarf