LibreELEC/LibreELEC.tv #7030: [BUG][LE11][Docker Addon] Docker addon not working on LE11 (622b4d7)

pretoriano80, 2022-10-21:
### Describe the bug
Docker addon installs just fine,but it's not working when issuing various docker commands via SSH terminal.

### To Reproduce
Steps to reproduce the behavior:
1. Install Docker service addon
2. Restart Libreelec (just in case)
3. Open SSH terminal and issue one of the following commands:
  docker -v
  docker ps
4. Check terminal output

### Informations
 - LE Version: LE11 master-622b4d7
 - Hardware Platform: RPi4 8GB

### Log file
[02_System.log](https://github.com/LibreELEC/LibreELEC.tv/files/9840007/02_System.log)

[09_Journal-cur.log](https://github.com/LibreELEC/LibreELEC.tv/files/9840084/09_Journal-cur.log)

[01_KODI.log](https://github.com/LibreELEC/LibreELEC.tv/files/9840095/01_KODI.log)

pretoriano80, 2022-10-21:
I'm getting the "docker not found" error.

heitbaum, 2022-10-21:
nuc12:~ # ls .kodi/addons/service.system.docker/
addon.xml   bin/        config/     default.py  examples/   lib/        resources/  system.d/
nuc12:~ # ls .kodi/addons/service.system.docker/bin
containerd               docker                   dockerd
containerd-shim          docker-config            runc
containerd-shim-runc-v2  docker-init
ctop                     docker-proxy

pretoriano80, 2022-10-21:
Can't check right now,but i will try to get more infos in a bit.

I wonder if it's related to the new RPi4 kernel ,some missing configuration  or something .

heitbaum, 2022-10-21:
Should be ok? I’m doing a compile now - make sure the compile looks right - I’ll check the addon zip file a bit later too.

pretoriano80, 2022-10-21:
I also tried with another docker version which was self-compiled and confirmed working on both LE10 and  older (September ) builds .

Also tried to completely uninstall,reboot and install again ,still no luck,so i'm a bit puzzled because logs aren't showing a well defined error .

heitbaum, 2022-10-21:
Here is my s/create_addon docker

** I haven’t checked the zip on the download server.
** please check your addons folder - is docker there / is there anything amiss,
** But the newly built add on looks right.

```
LibreELEC.kernel11$ unzip -v target/addons/10.80.8/ARMv8/arm/service.system.docker/service.system.docker-10.80.8.138.zip 
Archive:  target/addons/10.80.8/ARMv8/arm/service.system.docker/service.system.docker-10.80.8.138.zip
 Length   Method    Size  Cmpr    Date    Time   CRC-32   Name
--------  ------  ------- ---- ---------- ----- --------  ----
       0  Stored        0   0% 2022-10-21 19:33 00000000  service.system.docker/
    3018  Defl:N     1159  62% 2022-10-21 19:33 34da3bab  service.system.docker/addon.xml
       0  Stored        0   0% 2022-10-21 19:33 00000000  service.system.docker/bin/
35637504  Defl:N 11375321  68% 2022-10-21 19:33 ae195d7e  service.system.docker/bin/containerd
 7203272  Defl:N  2426763  66% 2022-10-21 19:33 7b03aaf0  service.system.docker/bin/containerd-shim
 9250416  Defl:N  3064681  67% 2022-10-21 19:33 e1aad30e  service.system.docker/bin/containerd-shim-runc-v2
   36244  Defl:N     9194  75% 2022-10-21 19:33 7ab4337f  service.system.docker/bin/ctop
21441564  Defl:N  7109788  67% 2022-10-21 19:33 8d35cae1  service.system.docker/bin/docker
     623  Defl:N      272  56% 2022-10-21 19:33 45c4aa32  service.system.docker/bin/docker-config
  462364  Defl:N   219551  53% 2022-10-21 19:33 eb8a64ac  service.system.docker/bin/docker-init
 1786716  Defl:N   710817  60% 2022-10-21 19:33 f0342bea  service.system.docker/bin/docker-proxy
48441196  Defl:N 15990609  67% 2022-10-21 19:33 4c500379  service.system.docker/bin/dockerd
 8924428  Defl:N  3345658  63% 2022-10-21 19:33 b25c3765  service.system.docker/bin/runc
       0  Stored        0   0% 2022-10-21 19:33 00000000  service.system.docker/config/
     192  Defl:N      142  26% 2022-10-21 19:33 8c89c95c  service.system.docker/config/docker.conf
   16284  Defl:N     1642  90% 2022-10-21 19:33 f7c3102d  service.system.docker/default.py
       0  Stored        0   0% 2022-10-21 19:33 00000000  service.system.docker/examples/
     639  Defl:N      291  55% 2022-10-21 19:33 828d6985  service.system.docker/examples/couchpotato.service
     791  Defl:N      349  56% 2022-10-21 19:33 b85751a6  service.system.docker/examples/mysql.service
     639  Defl:N      290  55% 2022-10-21 19:33 302d30d2  service.system.docker/examples/sabnzbd.service
     639  Defl:N      291  55% 2022-10-21 19:33 a9db93ad  service.system.docker/examples/sickbeard.service
     797  Defl:N      322  60% 2022-10-21 19:33 f0dcf05a  service.system.docker/examples/transmission.service
     239  Defl:N      176  26% 2022-10-21 19:33 d7f7d841  service.system.docker/examples/var-lib-docker.mount
       0  Stored        0   0% 2022-10-21 19:33 00000000  service.system.docker/lib/
    4651  Defl:N     1762  62% 2022-10-21 19:33 24200d97  service.system.docker/lib/dockermon.py
       0  Stored        0   0% 2022-10-21 19:33 00000000  service.system.docker/resources/
   13041  Defl:N    12085   7% 2022-10-21 19:33 04c1e8e5  service.system.docker/resources/fanart.png
   29284  Defl:N    28763   2% 2022-10-21 19:33 375bdb16  service.system.docker/resources/icon.png
       0  Stored        0   0% 2022-10-21 19:33 00000000  service.system.docker/resources/language/
       0  Stored        0   0% 2022-10-21 19:33 00000000  service.system.docker/resources/language/English/
    2205  Defl:N      440  80% 2022-10-21 19:33 dcd850bd  service.system.docker/resources/language/English/strings.po
    1332  Defl:N      340  75% 2022-10-21 19:33 94dba67f  service.system.docker/resources/settings.xml
       0  Stored        0   0% 2022-10-21 19:33 00000000  service.system.docker/system.d/
    1057  Defl:N      438  59% 2022-10-21 19:33 137c0978  service.system.docker/system.d/service.system.docker.service
--------          -------  ---                            -------
133259135         44301144  67%                            34 files
```

pretoriano80, 2022-10-21:
The addon looks fine,all the bins are there.

Here is the systemctl output:

**LibreELEC:~ # systemctl status service.system.docker.service | cat**
```
● service.system.docker.service - Docker Application Container Engine
     Loaded: loaded (/storage/.config/system.d/service.system.docker.service; enabled; preset: disabled)
     Active: active (running) since Fri 2022-10-21 22:40:16 EEST; 2min 3s ago
       Docs: https://docs.docker.com
    Process: 557 ExecStartPre=/storage/.kodi/addons/service.system.docker/bin/docker-config (code=exited, status=0/SUCCESS)
   Main PID: 912 (dockerd)
        CPU: 1.116s
     CGroup: /system.slice/service.system.docker.service
             ├─912 /storage/.kodi/addons/service.system.docker/bin/dockerd --exec-opt native.cgroupdriver=systemd --log-driver=journald --group=root --data-root=/storage/.kodi/userdata/addon_data/service.system.docker/docker --storage-driver=overlay2 --storage-opt overlay2.override_kernel_check=1
             └─920 containerd --config /var/run/docker/containerd/containerd.toml --log-level info

Oct 21 22:40:22 LibreELEC dockerd[912]: time="2022-10-21T22:40:22.534313673+03:00" level=info msg="[graphdriver] trying configured driver: overlay2"
Oct 21 22:40:22 LibreELEC dockerd[912]: time="2022-10-21T22:40:22.606875329+03:00" level=info msg="Loading containers: start."
Oct 21 22:40:22 LibreELEC dockerd[912]: time="2022-10-21T22:40:22.744828917+03:00" level=warning msg="Could not load necessary modules for IPSEC rules: protocol not supported"
Oct 21 22:40:22 LibreELEC dockerd[912]: time="2022-10-21T22:40:22.880731263+03:00" level=info msg="Default bridge (docker0) is assigned with an IP address 172.17.0.0/16. Daemon option --bip can be used to set a preferred IP address"
Oct 21 22:40:22 LibreELEC dockerd[912]: time="2022-10-21T22:40:22.958741419+03:00" level=info msg="Loading containers: done."
Oct 21 22:40:23 LibreELEC dockerd[912]: time="2022-10-21T22:40:23.013740573+03:00" level=warning msg="WARNING: No memory limit support"
Oct 21 22:40:23 LibreELEC dockerd[912]: time="2022-10-21T22:40:23.014086721+03:00" level=warning msg="WARNING: No swap limit support"
Oct 21 22:40:23 LibreELEC dockerd[912]: time="2022-10-21T22:40:23.014265888+03:00" level=info msg="Docker daemon" commit=library-import graphdriver=overlay2 version=library-import
Oct 21 22:40:23 LibreELEC dockerd[912]: time="2022-10-21T22:40:23.017719666+03:00" level=info msg="Daemon has completed initialization"
Oct 21 22:40:23 LibreELEC dockerd[912]: time="2022-10-21T22:40:23.067861301+03:00" level=info msg="API listen on /var/run/docker.sock"
```

pretoriano80, 2022-10-21:
And
```
LibreELEC:~ # docker -v                      
 -sh: docker: not found
```

heitbaum, 2022-10-21:
Path problem?

try running it with the full path.

nuc12:~ #  which docker
/storage/.kodi/addons/service.system.docker/bin/docker

# set | grep docker
# echo $PATH

pretoriano80, 2022-10-21:
Hmm,it is indeed a path issue but i need to find out what caused this.

pretoriano80, 2022-10-21:
@heitbaum an existing /storage/.profile   may cause such an issue .

I created that file previously ,after installing docker-compose (but before  installing docker addon) and added docker-compose path to it.

heitbaum, 2022-10-22:
Yes.

you can check what your login profile is doing - as below 
`strace ash -l /bin/date 2>&1 | grep profile`

as below: /etc/profile, then the files in /etc/profile.d/, which in turn include the addon profile.d, then finally .profile.

```
openat(AT_FDCWD, "/etc/profile", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/etc/profile.d", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 3
newfstatat(AT_FDCWD, "/etc/profile.d/00-addons.conf", {st_mode=S_IFREG|0644, st_size=1335, ...}, 0) = 0
openat(AT_FDCWD, "/etc/profile.d/00-addons.conf", O_RDONLY|O_CLOEXEC) = 3
newfstatat(AT_FDCWD, "/storage/.kodi/addons/virtual.network-tools/profile.d", 0x7ffc8ff0aa20, AT_SYMLINK_NOFOLLOW) = -1 ENOENT (No such file or directory)
...
openat(AT_FDCWD, "/storage/.kodi/addons/service.mariadb/profile.d", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 3
newfstatat(AT_FDCWD, "/storage/.kodi/addons/service.mariadb/profile.d/99-mysql-histfile.profile", {st_mode=S_IFREG|0644, st_size=190, ...}, 0) = 0
openat(AT_FDCWD, "/storage/.kodi/addons/service.mariadb/profile.d/99-mysql-histfile.profile", O_RDONLY|O_CLOEXEC) = 3
newfstatat(AT_FDCWD, "/etc/profile.d/10-locale.conf", {st_mode=S_IFREG|0644, st_size=148, ...}, 0) = 0
openat(AT_FDCWD, "/etc/profile.d/10-locale.conf", O_RDONLY|O_CLOEXEC) = 3
newfstatat(AT_FDCWD, "/etc/profile.d/52-nano.conf", {st_mode=S_IFREG|0644, st_size=127, ...}, 0) = 0
openat(AT_FDCWD, "/etc/profile.d/52-nano.conf", O_RDONLY|O_CLOEXEC) = 3
newfstatat(AT_FDCWD, "/etc/profile.d/90-systemd.conf", {st_mode=S_IFREG|0644, st_size=127, ...}, 0) = 0
openat(AT_FDCWD, "/etc/profile.d/90-systemd.conf", O_RDONLY|O_CLOEXEC) = 3
newfstatat(AT_FDCWD, "/etc/profile.d/98-busybox.conf", {st_mode=S_IFREG|0644, st_size=485, ...}, 0) = 0
openat(AT_FDCWD, "/etc/profile.d/98-busybox.conf", O_RDONLY|O_CLOEXEC) = 3
newfstatat(AT_FDCWD, "/etc/profile.d/99-kodi.conf", {st_mode=S_IFREG|0644, st_size=425, ...}, 0) = 0
openat(AT_FDCWD, "/etc/profile.d/99-kodi.conf", O_RDONLY|O_CLOEXEC) = 3
newfstatat(AT_FDCWD, "/etc/profile.d/99-pulseaudio.conf", {st_mode=S_IFREG|0644, st_size=168, ...}, 0) = 0
openat(AT_FDCWD, "/etc/profile.d/99-pulseaudio.conf", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/storage/.profile", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
```

pretoriano80, 2022-10-22:
This is the script that i use to install docker-compose .Never thought that setting the /storage/.profile would cause such issue .

```
#!/bin/bash

### Create NEW folder to contain the executable ###
 mkdir /storage/.config/bin

### Add folder to PATH ###
touch /storage/.profile
echo "PATH=$HOME/.config/bin:$PATH" >> /storage/.profile

### Install docker-compose ###
curl -L "https://github.com/docker/compose/releases/download/v2.11.2/docker-compose-$(uname -s)-$(uname -m)" -o /storage/.config/bin/docker-compose

### Make it executable
chmod +x /storage/.config/bin/docker-compose

### Reload environment variables ###
source /storage/.profile
```

closed 2022-10-23 by pretoriano80