LibreELEC/LibreELEC.tv #7038: [BUG] External drive not added to SMB.conf correctly

the-jeffski, 2022-10-25:
Having added a SD card I found my customised samba was playing up. The last entry in my config file was going to the SD card rather than the location I specified.  Having done some investigation I determined its the script that auto shares the external drive at fault as it ends up with an incorrectly formatted smb.conf .

My example files: 
```
[Tools]
  path = /storage/tools
  available = yes
  browseable = yes
  public = no 
  writeable = yes
  root preexec = mkdir -p /storage/tools
```

Then when the External is added it ends up as 

```
[Tools]
  path = /storage/tools
  available = yes
  browseable = yes
  public = no 
  writeable = yes
  root preexec = mkdir -p /storage/tools[External]
  path = /media/External
  available = yes
  browseable = yes
  public = no 
  writeable = yes
```

If I add  a blank row to my samba.conf  file   it then ends up 

```
[Tools]
  path = /storage/tools
  available = yes
  browseable = yes
  public = no 
  writeable = yes
  root preexec = mkdir -p /storage/tools

  [External]
  path = /media/External
  available = yes
  browseable = yes
  public = no 
  writeable = yes
```

Which works and the shares are separate.  

Seems like the script needs to at least add in a blank row for safety (but also being picky its not indented the same way).

antonlacon, 2023-03-04:
If you're able to test the linked PR, it would be appreciated. I was unable to recreate the issue and took a guess at what happened for you.

the-jeffski, 2023-03-06:
Took some time to get a device set up for this but once run the extra Line fixed what I was seeing before and didn't get this issue again.

heitbaum, 2023-09-04:
- believe fixed in #7785 
- if still an issue - please reopen.

closed 2023-09-04 by heitbaum