LibreELEC/LibreELEC.tv #7071: [BUG] Kodi frequently unresponsive on Raspberry Pi 3 (possibly due to GPU reset) - SSH remains accessible

andrewgreenwood, 2022-11-01:
### Describe the bug
The Kodi interface frequently is frozen when I turn on the TV (the clock is also frozen, and the time displayed is always an earlier time, indicating that the freeze is not related to the TV being switched on.) This has only been happening since I upgraded to LibreELEC 10.0.2 and seems to be a problem that affects versions after that also (e.g. LE 11 nightlies.)

I've previously been running LibreELEC 9.2.8 for quite a while on a Raspberry Pi 3 and it has generally been very stable - I've not had this problem with it.

### To Reproduce
There doesn't appear to be a specific set of steps to replicate this on-demand.

### Informations
 - LE Version: 10.0.2 or nightly 11 20221010-612a22b
 - Hardware Platform: RPi3

### Log files
https://misc.greenwoodprojects.net/libreelec/2022-10-24/kodi.log
https://misc.greenwoodprojects.net/libreelec/2022-10-24/dmesg.log

### Additional context
I'm able to connect via SSH and I've enabled debug logging and couldn't spot anything obvious around the time of the freezes - Kodi appears to still be doing things in the background, just the UI is completely frozen.

The only add-on I have installed is the tvheadend PVR client, which is solely what this is used for at present.

I've managed to catch it in a frozen state soon afterward so I have a kodi.log file and also the "dmesg" output at that point in time. What I have seen consistently happen with these is that the "dmesg" output repeatedly contains this message every second following the freeze:

`[drm] Resetting GPU.`

HiassofT, 2022-11-01:
Please test with current nightly and provide logs either from "pastekodi" (via ssh) - both LE 10.0.2 and 20221010 nighlties are rather ancient now and pure kodi/dmesg output doesn't provide all info we need

andrewgreenwood, 2022-11-04:
I've installed 20221031-d841dd3 a few days ago and it seemed OK for a few days but it has frozen this morning with the same symptoms (frozen screen with "[drm] Resetting GPU." in dmesg output.) Based on the time displayed on the screen, this froze at 7:50am UTC+0.

The log file is too big for pastekodi to upload (I'd enabled some additional component logging such as Video in case that was useful) but I've sent the output from it into another file so this should have the additional information you need:

https://misc.greenwoodprojects.net/libreelec/2022-11-04/kodilogs.gz

Thanks!

heitbaum, 2023-01-15:
Closing as inactive. Please reopen if still and issue and with updated information (using latest LE11 nightly.)

closed 2023-01-15 by heitbaum