LibreELEC/LibreELEC.tv #7088: No Picture for drm video, eg netflix

e-ru, 2022-11-04:
### Describe the bug
when playing videos for drm protected content like netflix there is no picture anymore (but sound is)
log is polluted with this entry: `CEGLImage::CreateImage - failed to import buffer into EGL image: 0x3003`

### To Reproduce
Steps to reproduce the behavior:
1. use this [nightly version](https://test.libreelec.tv/11.0/Generic/Generic/LibreELEC-Generic.x86_64-11.0-nightly-20221104-534ecbc.img.gz)  of libreelec
2. install inputstream helper in version 0.5.10+matrix.1 and inputstream adaptive in version 20.3.1.1
3. install netflix addon from [this repo](https://github.com/castagnait/repository.castagnait/raw/kodi/repository.castagnait-2.0.0.zip)
4. play random video on netflix
5. see the logs for the error

### Informations
 - LE Version: LibreELEC-Generic.x86_64-11.0-nightly-20221104
 - Hardware Platform: CPU: 11th Gen Intel(R) Core(TM) i5-11400, GPU: AMD Radeon RX 6600 XT (navi23, LLVM 15.0.3, DRM 3.48, 6.0.6)
 - linux version: 6.0.6

### Log file

[pastekodi details](http://ix.io/4eYJ)

### Additional context

i think that it worked with older kernels, but not sure about that.

lrusak, 2022-11-05:
Disable drmprime

lrusak, 2022-11-05:
`0x3003 -> EGL_BAD_ALLOC`, are you out of memory?

e-ru, 2022-11-05:
Hello

Thanks for the reply. drmprime was already disabled. Enabling it didn't solve the issue. Also added 4GB of ram - still error: `CEGLImage::CreateImage - failed to import buffer into EGL image: 0x3003`.
I did another [pastekodi](http://ix.io/4f0q) with drmprime enabled and 8GB of ram in total.

Any other ideas?

Thanks for your time.
Erik

lrusak, 2022-11-05:
I can actually confirm this, I'm not sure why it's not working anymore.

I can also confirm that toggling drmprime doesn't change the code path.

e-ru, 2022-11-05:
hm, at least you can confirm this :sweat_smile: is there anything i can do to support you with this? try something out or some other thing?

knaerzche, 2022-11-07:
This is likley because of

`AddOnLog: inputstream.adaptive: ToSSDVideoFormat: Unknown video format 17`

According to https://chromium.googlesource.com/chromium/cdm/+/refs/heads/main/content_decryption_module.h#243 this is a yuv420p10 format which is currently not handled in inputstream adaptive neither supported by drmprime codec. If it is for netflix you're able to switch to VP9 codec in addon settings, which IIRC will be regular 8 bit yuv420 format and should work.

e-ru, 2022-11-13:
hello @knaerzche 

sadly i cannot find the setting to switch to vp9 codec. neither in the inputstream adaptive (version 20.3.1.1) nor in the netflix (version 1.20.2+matrix.1) pluign. 

Do i miss something here?

regards erik

knaerzche, 2022-11-13:
It's in  Netflix addon settings -> Expert: "Enable VP9-profiles (disable if it causes artifacts)". Also make sure you have AV1 disabled all the time (iirc that is not supported by chromeos-version of the widevine )

e-ru, 2022-11-16:
ok, this was already enabled. i played around with this codec settings by toggling them in different variations with no improvement. i created an [issue](https://github.com/xbmc/inputstream.adaptive/issues/1070) in the inputstream.adaptive project

CastagnaIT, 2022-11-17:
the warning refer to 

> AddOnLog: inputstream.adaptive: ToSSDVideoFormat: Unknown video format 17

should not have influence since that value should be unused in kodi
the n@tfix VP9, FWIK has always been 8bit if it is not changed recently
the VP9 profile 2 (10bit) is not enabled and cannot be not be enabled in the add-on if not forced manually by editing the code

> ok, this was already enabled. i played around with this codec settings by toggling them in different variations with no improvement.

if after disabled the problem presist is an allocation problem
also confirmed from the fact that from the log i dont see any message of picture obtained, like this example:

> debug <general>: CAddonVideoCodec: GetPicture::VC_PICTURE with pts ............

then the problem happens right before decoding, on buffer allocation
as the message EGL_BAD_ALLOC on `eglCreateImageKHR` native call 

could be low GPU memory?

ISAdaptive rely to Kodi core to allocate memory there is nothing here
from my part i cannot help more than this

knaerzche, 2022-11-17:
> since that value should be unused in kodi

Yes and that is exactly what the problem ist: Kodi assumes it is 8 bit format, but it isn't and that's why it fails importing it in EGL.

lrusak, 2022-11-19:
I can get this error with the following stream

```
#KODIPROP:inputstream=inputstream.adaptive
#KODIPROP:inputstream.adaptive.manifest_type=mpd
#KODIPROP:inputstream.adaptive.license_type=com.widevine.alpha
#KODIPROP:inputstream.adaptive.license_key=https://proxy.uat.widevine.com/proxy?video_id=d286538032258a1c&provider=widevine_test||R{SSM}|
#KODIPROP:inputstream.adaptive.license_key=https://proxy.uat.widevine.com/proxy?video_id=0894c7c8719b28a0&provider=widevine_test||R{SSM}|
https://storage.googleapis.com/wvmedia/cenc/h264/tears/tears.mpd
```

log file:
```
2022-11-19 13:24:16.904 T:2835175   debug <general>: EGL Debugging:
                                                   Error: EGL_BAD_ALLOC
                                                   Command: eglCreateImageKHR
                                                   Type: EGL_DEBUG_MSG_CRITICAL_KHR
                                                   Message: dri2_create_image_khr_texture
2022-11-19 13:24:16.904 T:2835175   debug <general>: CEGLImage::CreateImage - attributes:
                                                   EGL_WIDTH: 1920
                                                   EGL_HEIGHT: 856
                                                   EGL_LINUX_DRM_FOURCC_EXT: YU12
                                                   EGL_YUV_COLOR_SPACE_HINT_EXT: EGL_ITU_REC709_EXT
                                                   EGL_SAMPLE_RANGE_HINT_EXT: EGL_YUV_NARROW_RANGE_EXT
                                                   EGL_YUV_CHROMA_VERTICAL_SITING_HINT_EXT: EGL_YUV_CHROMA_SITING_0_EXT
                                                   EGL_YUV_CHROMA_HORIZONTAL_SITING_HINT_EXT: EGL_YUV_CHROMA_SITING_0_EXT
                                                   EGL_DMA_BUF_PLANE0_FD_EXT: 67
                                                   EGL_DMA_BUF_PLANE0_OFFSET_EXT: 0
                                                   EGL_DMA_BUF_PLANE0_PITCH_EXT: 1984
                                                   EGL_DMA_BUF_PLANE1_FD_EXT: 67
                                                   EGL_DMA_BUF_PLANE1_OFFSET_EXT: 1698304
                                                   EGL_DMA_BUF_PLANE1_PITCH_EXT: 1984
                                                   EGL_DMA_BUF_PLANE2_FD_EXT: 67
                                                   EGL_DMA_BUF_PLANE2_OFFSET_EXT: 1699296
                                                   EGL_DMA_BUF_PLANE2_PITCH_EXT: 1984
2022-11-19 13:24:17.111 T:2835175   error <general>: CEGLImage::CreateImage - failed to import buffer into EGL image: EGL_BAD_ALLOC
2022-11-19 13:24:17.113 T:2835296   debug <general>: CVideoBufferDMA::SetDimensions - frame layout id=8 fourcc=YU12
                                                   plane[0]: stride=1984        offset=0
                                                   plane[1]: stride=1984        offset=1698304
                                                   plane[2]: stride=1984        offset=1699296
2022-11-19 13:24:17.113 T:2835296   debug <general>: CAddonVideoCodec: GetPicture::VC_PICTURE with pts 416666 1920x856 (1916x856) 2.2403734 0x7f1099059000:2549440 offset:0,1698304,1699296, stride:1984,1984,1984

```
Some of the dimensions see strange to me.

lrusak, 2022-11-19:
Looks like for me it breaks here: https://gitlab.freedesktop.org/mesa/mesa/-/blob/mesa-22.1.7/src/amd/common/ac_surface.c#L2833
```
2833	   if (offset & ((1 << surf->alignment_log2) - 1) ||
(gdb) print offset
$3 = 365280
(gdb) print surf->alignment_log2 
$6 = 8 '\b'
```

which evaluates to 224

So it seems that the alignment needs to be 256?

https://godbolt.org/z/94T5PzPhb

lrusak, 2022-11-19:
This commit seems to fix the import but the strides are wrong now, though I'm not sure how we can fix that. I'll have to look into inputstream.adaptive :thinking: https://github.com/lrusak/xbmc/commit/3c4d10bab7a3c324e9a7638ba3c753517b82877f

```
2022-11-19 14:32:12.438 T:38034   debug <general>: CVideoBufferDMA::SetDimensions - frame layout id=4 fourcc=YU12
                                                   plane[0]: stride=1984	offset=0
                                                   plane[1]: stride=1984	offset=1698304
                                                   plane[2]: stride=1984	offset=1699328
2022-11-19 14:32:12.438 T:38034   debug <general>: CAddonVideoCodec: GetPicture::VC_PICTURE with pts 10791666 1920x856 (1916x856) 2.2403734 0x7f89100ed000:2549440 offset:0,1698304,1699296, stride:1984,1984,1984
2022-11-19 14:32:12.470 T:37899   debug <general>: CEGLImage::CreateImage - attributes:
                                                   EGL_WIDTH: 1920
                                                   EGL_HEIGHT: 856
                                                   EGL_LINUX_DRM_FOURCC_EXT: YU12
                                                   EGL_YUV_COLOR_SPACE_HINT_EXT: EGL_ITU_REC709_EXT
                                                   EGL_SAMPLE_RANGE_HINT_EXT: EGL_YUV_NARROW_RANGE_EXT
                                                   EGL_YUV_CHROMA_VERTICAL_SITING_HINT_EXT: EGL_YUV_CHROMA_SITING_0_EXT
                                                   EGL_YUV_CHROMA_HORIZONTAL_SITING_HINT_EXT: EGL_YUV_CHROMA_SITING_0_EXT
                                                   EGL_DMA_BUF_PLANE0_FD_EXT: 74
                                                   EGL_DMA_BUF_PLANE0_OFFSET_EXT: 0
                                                   EGL_DMA_BUF_PLANE0_PITCH_EXT: 1984
                                                   EGL_DMA_BUF_PLANE1_FD_EXT: 74
                                                   EGL_DMA_BUF_PLANE1_OFFSET_EXT: 1698304
                                                   EGL_DMA_BUF_PLANE1_PITCH_EXT: 1984
                                                   EGL_DMA_BUF_PLANE2_FD_EXT: 74
                                                   EGL_DMA_BUF_PLANE2_OFFSET_EXT: 1699328
                                                   EGL_DMA_BUF_PLANE2_PITCH_EXT: 1984
```

knaerzche, 2022-11-20:
It seems it's a bug/change/fix in mesa - I'm getting this now also with "regular" non-netflix streams when doing sw decoding (which is always via EGL for RK) when using latest 22.2.4, reverting back to something earlier helps. The dimension/stride calcs seem strange, though.

heitbaum, 2023-09-04:
Closing - as fixed. Please reopen if that is not the case with updated info.

closed 2023-09-04 by heitbaum