LibreELEC/LibreELEC.tv #7095: [BUG] scummvm: Monkey Island - sound garbled, no graphics

chri2, 2022-11-07:
### Describe the bug
On Pine Rock64 using 10.0.3:

scummvm didn't start at all when installed from the repositories. Built it on my notebook using the docker method. 

I tested building the add-ons for the official 10.0.3 by `git checkout 10.0.3; git pull` followed by the build commands and tested building the whole libreelec-10.0 branch and additionally the add-ons.

My build commands were:
```
docker build --pull -t libreelec tools/docker/focal
docker run --rm --log-driver none -v `pwd`:/build -w /build -it -e PROJECT=Rockchip -e DEVICE=RK3328 -e ARCH=arm -e UBOOT_SYSTEM=rock64 libreelec make image
docker run --rm --log-driver none -v `pwd`:/build -w /build -it -e PROJECT=Rockchip -e DEVICE=RK3328 -e ARCH=arm -e UBOOT_SYSTEM=rock64 libreelec scripts/create_addon game.libretro
docker run --rm --log-driver none -v `pwd`:/build -w /build -it -e PROJECT=Rockchip -e DEVICE=RK3328 -e ARCH=arm -e UBOOT_SYSTEM=rock64 libreelec scripts/create_addon game.libretro.scummvm
```

Both ways (building add-ons only and building a whole image and the add-ons from the same sourvce) I had the same problem.

The resulting add-ons `game.libretro` and `game.libretro.scummvm` started and showed a menu that let me add the monkey island directory containing the files `000.LFL 902.LFL 904.LFL DISK02.LEC DISK04.LEC 901.LFL 903.LFL DISK01.LEC DISK03.LEC`.

Starting the game let the sound play, but it has been garbled playing some tunes out of order. The screen stayed dark.

### To Reproduce
1. install scummvm add-on from repository (didn't start)
2. clone repo, checkout libreelec-10.0 or 10.0.3
3. build add-ons game.libretro and game.libretro.scummvm
4. `systemctl stop kodi`, install the resulting zip files, reboot or restart kodi
5. copy monkey island files to rock64
6. start scummvm (should run now ;-) )
7. add directory with monkey island files to the list of games
8. select start game

### Informations
 - LE Version: 10.0.3 and own build from actual libreelec-10.0 branch
 - Hardware Platform: Rock64 (with additional sound-card for optical-spdif)

### Log file
[kodi.log](https://github.com/LibreELEC/LibreELEC.tv/files/9950736/kodi.log)

SupervisedThinking, 2022-11-07:
Did you test other The Secret of Monkey Island [data files](https://wiki.scummvm.org/index.php/The_Secret_of_Monkey_Island#Required_data_files) ? Because you've tested Amiga / Atari versions & maybe others work properly?

SupervisedThinking, 2022-11-07:
Well it's probably related to some mesa quirks since LE 10.x ships with really old versons. Could you test a recent LE 11 nightly? https://test.libreelec.tv/11.0/Rockchip/rock64/

chri2, 2022-11-08:
Forgot to mention it:

[Just tried https://test.libreelec.tv/11.0…103-bb2fca8-rock64.img.gz and ScummVM didn't start.](https://forum.libreelec.tv/thread/26164-scummvm-add-on-could-not-be-loaded-an-unknown-error-has-occurred/?postID=173852#post173852)

Though I didn't build LE11 myself nor the add-on. It's not working for me in other parts anyway, yet.

chri2, 2022-11-08:
I can add another data point: I just started Monkey Island in scummvm on my Librem5 which is aarch64 (running PureOS which is mostly Debian) also. Works just fine, so it's not a problem with the architecture.

SupervisedThinking, 2022-11-09:
If you're able to rebuild the addon you could give this a try:
https://github.com/SupervisedThinking/LibreELEC-RR/blob/master-rr/packages/supervisedthinking/emulation/libretro/scummvm/package.mk#L1-L47

don't forget to add this line https://github.com/LibreELEC/LibreELEC.tv/blob/master/packages/emulation/libretro-scummvm/package.mk#L16

Technically this should build you a v2.6.1 scummvm libretro core & use `${PKG_BUILD}/backends/platform/libretro/aux-data/scummvm.zip` as aux data.

chri2, 2022-11-10:
Thanks for helping! Tried to rebuild the game add-on as described, but couldn't get it to build.

Edited `packages/emulation/libretro-scummvm/package.mk` like this:

```
# SPDX-License-Identifier: GPL-2.0
# Copyright (C) 2018-present Frank Hartung (supervisedthinking (@) gmail.com)

PKG_NAME="scummvm"
PKG_VERSION="7cf738d27095c072c253bef45445e63f708a5441" # v2.6.1
PKG_LICENSE="GPL-3.0-or-later"
PKG_SITE="https://github.com/SupervisedThinking/scummvm"
PKG_URL="https://github.com/SupervisedThinking/scummvm.git"
PKG_DEPENDS_TARGET="toolchain linux glibc"
PKG_LONGDESC="ScummVM is an interpreter for point-and-click adventure games that can be used as a libretro core."
PKG_TOOLCHAIN="make"
PKG_GIT_CLONE_BRANCH="branch-2-6-1-libretro"
PKG_GIT_CLONE_SINGLE="yes"
GET_HANDLER_SUPPORT="git"
PKG_BUILD_FLAGS="-sysroot +speed"

PKG_LIBNAME="scummvm_libretro.so"
PKG_LIBPATH="backends/platform/libretro/build/${PKG_LIBNAME}"
PKG_LIBVAR="SCUMMVM_LIB"

PKG_MAKE_OPTS_TARGET="-C backends/platform/libretro/build GIT_VERSION=${PKG_VERSION:0:7}"

pre_configure_target() {
  if [ "${ARCH}" = "arm" ]; then
    PKG_MAKE_OPTS_TARGET+=" platform=armv"
    # ARM NEON support
    if target_has_feature neon; then
      PKG_MAKE_OPTS_TARGET+="-neon"
    fi
    PKG_MAKE_OPTS_TARGET+="-${TARGET_FLOAT}float-${TARGET_CPU}"
  fi
}

pre_make_target() {
  # Rebuild ScummVM auxiliary data
  if [ -f ${PKG_BUILD}/backends/platform/libretro/aux-data/scummvm.zip ]; then
    echo -e "\n### Rebuild ScummVM auxiliary data ####\n"
    cd ${PKG_BUILD}/backends/platform/libretro/aux-data
    ./bundle_aux_data.bash
  fi

  # Fix build path
  cd ${PKG_BUILD}

  if [ ! "${ARCH}" = "x86_64" ]; then
    CXXFLAGS+=" -DHAVE_POSIX_MEMALIGN=1"
  fi
}
```

And then ran `docker run --rm --log-driver none -v `pwd`:/build -w /build -it -e PROJECT=Rockchip -e DEVICE=RK3328 -e ARCH=arm -e UBOOT_SYSTEM=rock64 libreelec scripts/create_addon game.libretro.scummvm` and ended up with this error:

```
scripts/create_addon game.libretro.scummvm
Traceback (most recent call last):
  File "scripts/genbuildplan.py", line 369, in <module>
    REQUIRED_PKGS = processPackages(args, ALL_PACKAGES)
  File "scripts/genbuildplan.py", line 298, in processPackages
    raise Exception(msg)
Exception: Invalid package reference: dependency libretro-scummvm in package game.libretro.scummvm::PKG_DEPENDS_TARGET is not valid
UNKNOWN BUILD FAILURE OR INABILITY TO GENERATE PLAN
``` 

BTW: Aren't there any Rock64 users out there who could confirm that scummvm works or not?

SupervisedThinking, 2022-11-10:
Change PKG_NAME="scummvm" to PKG_NAME="libretro-scummvm"

chri2, 2022-11-11:
Hm, I'd need the 10.0.3 version of the add-on. LE11 nightly had problems with tvheadend when I tested it last and I can't break working stuff (I'm not the only user ;-) ).

I tried to compile scummvm 2.6.1 like described for 10.0.3 and ended up with this error:

```
Submodule path 'devtools/create_prince': checked out '4e2acb737c936a576dee19927bb0cab5a0ae3b2d'
CLEAN      libretro-scummvm
    *      Removing /build/build.LibreELEC-RK3328.arm-10.0-devel/build/libretro-scummvm-80cb7269a33b233dcea27d8d01df084b0d35c80a ...
UNPACK      libretro-scummvm
    APPLY PATCH (common)      packages/emulation/libretro-scummvm/patches/libretro-scummvm-0001-fix-cross-compile.patch
patching file backends/platform/libretro/build/Makefile
Hunk #1 FAILED at 24.
1 out of 1 hunk FAILED -- saving rejects to file backends/platform/libretro/build/Makefile.rej
*********** FAILED COMMAND ***********
patch -d $(echo "${PKG_BUILD}" | cut -f1 -d\ ) -p1 1>&${VERBOSE_OUT}
**************************************
*********** FAILED COMMAND ***********
${SCRIPTS}/unpack "${PKG_NAME}" "${PARENT_PKG}"
**************************************
FAILURE: scripts/build libretro-scummvm:target has failed!

The following log for this failure is available:
  /build/build.LibreELEC-RK3328.arm-10.0-devel/.threads/logs/193.log

>>> libretro-scummvm:target seq 193 >>>
[193/194] [FAIL] build   libretro-scummvm:target

The following log for this failure is available:
  /build/build.LibreELEC-RK3328.arm-10.0-devel/.threads/logs/193.log

<<< game.libretro.scummvm:target seq 194 <<<
FAILURE [build game.libretro.scummvm:target]: a previous dependency process has already failed!

The following log(s) for already failed dependencies are available:
  build   libretro-scummvm:target => /build/build.LibreELEC-RK3328.arm-10.0-devel/.threads/logs/193.log

FAILURE: scripts/build game.libretro.scummvm:target has failed!

The following log for this failure is available:
  /build/build.LibreELEC-RK3328.arm-10.0-devel/.threads/logs/194.log

>>> game.libretro.scummvm:target seq 194 >>>
[194/194] [FAIL] build   game.libretro.scummvm:target

The following log for this failure is available:
  /build/build.LibreELEC-RK3328.arm-10.0-devel/.threads/logs/194.log

The following log from the failed dependency may be relevant:
  build   libretro-scummvm:target => /build/build.LibreELEC-RK3328.arm-10.0-devel/.threads/logs/193.log

FAILED ADDONS:
    game.libretro.scummvm: /build/build.LibreELEC-RK3328.arm-10.0-devel/.threads/logs/194.log
```

[193.log](https://github.com/LibreELEC/LibreELEC.tv/files/9987904/193.log)

I guess I'd have to look into the patch that gets applied or should I just disable it?

Is there any chance anyway that the newer scummvm will build on 10.0.3?

SupervisedThinking, 2022-11-11:
Well you would have to remove the old patch 😸 Anyway I've updated the libretro-scummvm package in the 10.0 branch https://github.com/SupervisedThinking/LibreELEC-RR/tree/libreelec-10.0_scummvm maybe you can pull this in and try to build?

At least this should build flawless for you.

chri2, 2022-11-12:
Build the game.libretro-19.0.2.3.zip and game.libretro.scummvm-2.1.1.23.1.zip and installed them to my LE 10.0.3. Then unziped scummvm.zip in `.kodi/addons/game.libretro.scummvm/resources/system/`.

scummvm started fine, added Monkey Island, but again there's no picture.

chri2, 2022-11-16:
Any other idea how to proceed with this?

SupervisedThinking, 2022-11-16:
@HiassofT ping - you've sucessfully tested ScummVM on your RPi4?
@chri2 well I guess it's probably an issue with mesa 

```
2022-11-07 11:49:10.459 T:1840    ERROR <general>: CEGLImage::CreateImage - failed to import buffer into EGL image: 0x3003
2022-11-07 11:49:10.459 T:1840    DEBUG <general>: EGL Debugging:
                                                   Error: EGL_BAD_PARAMETER
                                                   Command: eglDestroyImageKHR
                                                   Type: EGL_DEBUG_MSG_ERROR_KHR
                                                   Message: _eglDestroyImageCommon
2022-11-07 11:49:10.475 T:1840    DEBUG <general>: EGL Debugging:
                                                   Error: EGL_BAD_ALLOC
                                                   Command: eglCreateImageKHR
                                                   Type: EGL_DEBUG_MSG_CRITICAL_KHR
                                                   Message: dri2_create_image_khr_texture
2022-11-07 11:49:10.476 T:1840    DEBUG <general>: CEGLImage::CreateImage - attributes:
                                                   EGL_WIDTH: 320
                                                   EGL_HEIGHT: 200
                                                   EGL_LINUX_DRM_FOURCC_EXT: RG16
                                                   EGL_DMA_BUF_PLANE0_FD_EXT: 97
                                                   EGL_DMA_BUF_PLANE0_OFFSET_EXT: 0
                                                   EGL_DMA_BUF_PLANE0_PITCH_EXT: 640
```

panfrost is still under heavy developement and IIRC there had been problems in the past with some SBCs & emulation.

HiassofT, 2022-11-16:
Yes, I successfully tested scummvm on an RPi4 running LE10 with a local build of the latest LE10 retroplayer addons from this PR https://github.com/LibreELEC/LibreELEC.tv/pull/7121 (which still hasn't been merged - hint, hint :-) and the German DOS/VGA version of Monkey Island 1, freshly copied off my original "Zehn Adventrures" LucasArts collection CD
```
LibreELEC:~/games/Monkey Island 1 # ls -l
total 4439
-r-xr-xr-x    1 root     root          8357 Nov 13 00:02 000.lfl
-r-xr-xr-x    1 root     root          2572 Nov 13 00:02 901.lfl
-r-xr-xr-x    1 root     root          4023 Nov 13 00:02 902.lfl
-r-xr-xr-x    1 root     root          2572 Nov 13 00:02 903.lfl
-r-xr-xr-x    1 root     root          4782 Nov 13 00:02 904.lfl
-r-xr-xr-x    1 root     root       1101583 Nov 13 00:02 disk01.lec
-r-xr-xr-x    1 root     root       1118422 Nov 13 00:02 disk02.lec
-r-xr-xr-x    1 root     root       1176753 Nov 13 00:02 disk03.lec
-r-xr-xr-x    1 root     root       1048787 Nov 13 00:02 disk04.lec
-r-xr-xr-x    1 root     root         69637 Nov 13 00:02 monkey.exe
-r-xr-xr-x    1 root     root          1417 Nov 13 00:02 readme.vga
```

I agree that some mesa issue doesn't sound too unlikely but other than that my scummvm build, hardware and graphics drivers are all different and any of that could be responsible for causing the issues mentioned here.

chri2, 2022-11-17:
Damn Rock64. It's perfect for running LE10, but this doesn't seem to work.

Is there different/better experience with RockPro64 (RK3399) around? I do have another LE10 which I rarely use, because it is connected to a projector, which I could test.

SupervisedThinking, 2022-11-17:
I hardly doubt it's a Rock64 specific problem rather than a qurik related to years old mesa drivers.

chri2, 2022-11-17:
Hm, what I do not understand is why it seems to work for others? What hardware platform do you run your version on?
BTW: I could test an up-to-date scummvm from your tree if I'd find out how to compile the missing add-ons it depends on.

SupervisedThinking, 2022-11-19:
Well personally I use a chimera of master LE + Retroarch & so I don't rely on LE10.x & Retroplayer. The reason why it does work on RPi4 but not on Rock64 is mostly likely related to the mesa graphic driver. The RPi drivers are probably more compatible or something. LE10 uses 21.0.3 while master uses 22.2.4 & 22.3.0-rc3 was released this week. The lima driver, used by Mali 400/450, has seen plenty of fixes over almost two years.

On the other hand the Kodi v20 beta1 seems to be broken & needs fixing https://github.com/xbmc/xbmc/issues/22159 which is the reason why LE11 nightlies don't work.

heitbaum, 2023-01-15:
Closing as fixed in LE11 nightlies. Please reopen if still an issue and with updated information (using latest LE11 nightly.)

closed 2023-01-15 by heitbaum