LibreELEC/LibreELEC.tv #7142: [BUG] RPI4 freeze while playing twitch.tv live content, during midroll ad.

perkabalo, 2022-11-17:
### Describe the bug
RPI4 freeze while playing twitch.tv live content, during midroll ad.

### To Reproduce
Steps to reproduce the behavior:
1. Go to https://pwn.sh/tools/getstream.html
2. Share livestream to kodi e.g. with Kore app
3. Wait for midroll ad (!important)
4. System freeze

### Information
 - LE Version: [LibreELEC-RPi4.arm-11.0-nightly-20221116-f28aa43]
 - Hardware Platform: [e.g. RPi4]

Same behavior can be reproduced on latest stable version of LibreELEC.
Seems to be kernel driver related.

### Log file
[journalctl.txt](https://github.com/LibreELEC/LibreELEC.tv/files/10035831/journalctl.txt)
[kodi.old.log](https://github.com/LibreELEC/LibreELEC.tv/files/10035832/kodi.old.log)

### Additional context
Same issue can be reproduced on OSMC.

WallyCZ, 2023-01-08:
I've got some log before it got frozen, maybe somebody will direct us who could look into this.

`Linux LibreELEC 6.1.3 #1 SMP Sun Jan  8 00:18:30 UTC 2023 aarch64 GNU/Linux`

```
Jan 08 16:26:21 LibreELEC kernel: ------------[ cut here ]------------
Jan 08 16:26:21 LibreELEC kernel: WARNING: CPU: 3 PID: 24277 at __vb2_queue_cancel+0x21c/0x290 [videobuf2_common]
Jan 08 16:26:21 LibreELEC kernel: Modules linked in: veth xt_nat xt_tcpudp xt_conntrack xt_MASQUERADE nf_conntrack_netlink nfnetlink xt_addrtype iptable_filter iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip_tables x_tables br_netfilter bridge stp llc rfcomm hci_uart btbcm bluetooth ecdh_generic ecc overlay 8021q tda18218 bcm2835_gpiomem brcmfmac af9033 brcmutil rpivid_hevc(C) snd_bcm2835(C) bcm2835_codec(C) bcm2835_isp(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common dvb_usb_af9035 videodev dvb_usb_v2 rc_imon_pad dvb_core snd_usb_audio imon snd_usbmidi_lib snd_hwdep snd_rawmidi mc nvmem_rmem ntfs3 cfg80211 rfkill pkcs8_key_parser fuse
Jan 08 16:26:21 LibreELEC kernel: CPU: 3 PID: 24277 Comm: VideoPlayer Tainted: G         C         6.1.3 #1
Jan 08 16:26:21 LibreELEC kernel: Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
Jan 08 16:26:21 LibreELEC kernel: pstate: 80000005 (Nzcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Jan 08 16:26:21 LibreELEC kernel: pc : __vb2_queue_cancel+0x21c/0x290 [videobuf2_common]
Jan 08 16:26:21 LibreELEC kernel: lr : __vb2_queue_cancel+0x38/0x290 [videobuf2_common]
Jan 08 16:26:21 LibreELEC kernel: sp : ffffffc00f7bbac0
Jan 08 16:26:21 LibreELEC kernel: x29: ffffffc00f7bbac0 x28: ffffffc000f2fd70 x27: ffffff808ec82000
Jan 08 16:26:21 LibreELEC kernel: x26: 0000000000000000 x25: 0000000000000000 x24: ffffff8076171700
Jan 08 16:26:21 LibreELEC kernel: x23: ffffff804b05d528 x22: ffffff8051bb2028 x21: ffffff8051bb2028
Jan 08 16:26:21 LibreELEC kernel: x20: 0000000000000009 x19: ffffff8051bb2028 x18: ffffffffffffffff
Jan 08 16:26:21 LibreELEC kernel: x17: 000000010f7bc000 x16: 0000000000000000 x15: 0000000000000037
Jan 08 16:26:21 LibreELEC kernel: x14: 0000000000000004 x13: ffffff8050b40028 x12: 0000000000000000
Jan 08 16:26:21 LibreELEC kernel: x11: ffffff80cee523a8 x10: ffffff80cee52238 x9 : ffffffc000e790f0
Jan 08 16:26:21 LibreELEC kernel: x8 : ffffff80cee52260 x7 : 0000000000000000 x6 : 0000000000000000
Jan 08 16:26:21 LibreELEC kernel: x5 : ffffff804b05d81c x4 : 0000000000000000 x3 : ffffff8050b40008
Jan 08 16:26:21 LibreELEC kernel: x2 : 0000000000000000 x1 : 00000000000051a8 x0 : 0000000000000005
Jan 08 16:26:21 LibreELEC kernel: Call trace:
Jan 08 16:26:21 LibreELEC kernel:  __vb2_queue_cancel+0x21c/0x290 [videobuf2_common]
Jan 08 16:26:21 LibreELEC kernel:  vb2_core_streamoff+0x2c/0xd0 [videobuf2_common]
Jan 08 16:26:21 LibreELEC kernel:  vb2_streamoff+0x20/0x70 [videobuf2_v4l2]
Jan 08 16:26:21 LibreELEC kernel:  v4l2_m2m_streamoff+0x80/0x170 [v4l2_mem2mem]
Jan 08 16:26:21 LibreELEC kernel:  v4l2_m2m_ioctl_streamoff+0x20/0x2c [v4l2_mem2mem]
Jan 08 16:26:21 LibreELEC kernel:  v4l_streamoff+0x2c/0x40 [videodev]
Jan 08 16:26:21 LibreELEC kernel:  __video_do_ioctl+0x16c/0x3e0 [videodev]
Jan 08 16:26:21 LibreELEC kernel:  video_usercopy+0x1fc/0x72c [videodev]
Jan 08 16:26:21 LibreELEC kernel:  video_ioctl2+0x20/0x40 [videodev]
Jan 08 16:26:21 LibreELEC kernel:  v4l2_ioctl+0x48/0x6c [videodev]
Jan 08 16:26:21 LibreELEC kernel:  v4l2_compat_ioctl32+0xa0/0xe4 [videodev]
Jan 08 16:26:21 LibreELEC kernel:  __arm64_compat_sys_ioctl+0xd8/0x14c
Jan 08 16:26:21 LibreELEC kernel:  invoke_syscall+0x50/0x120
Jan 08 16:26:21 LibreELEC kernel:  el0_svc_common.constprop.0+0x68/0x124
Jan 08 16:26:21 LibreELEC kernel:  do_el0_svc_compat+0x24/0x4c
Jan 08 16:26:21 LibreELEC kernel:  el0_svc_compat+0x30/0x9c
Jan 08 16:26:21 LibreELEC kernel:  el0t_32_sync_handler+0x98/0x140
Jan 08 16:26:21 LibreELEC kernel:  el0t_32_sync+0x190/0x194
Jan 08 16:26:21 LibreELEC kernel: ---[ end trace 0000000000000000 ]---
Jan 08 16:26:21 LibreELEC kernel: videobuf2_common: driver bug: stop_streaming operation is leaving buf 00000000d8f05508 in active state

```
Whole kernel log until it froze: 
[rpi4_twitch_freeze.txt](https://github.com/LibreELEC/LibreELEC.tv/files/10368788/rpi4_twitch_freeze.txt)

WallyCZ, 2023-02-11:
Another message from this issue (this at least does not crash whole OS, but only break the stream):
`kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1382400 < 3133440)
`

thomasb22, 2023-08-04:
Same issue here with Twitch, I tested with Twitch plugin and SendToKodi, both freeze the Raspberry Pi 4. However with Twitch plugin a rewind can replay the stream sometimes.

heitbaum, 2023-09-04:
Please retest with current le11/12. If still an issue - please reopen issue or create new and reference this one.

thomasb22, 2023-09-05:
There is no new version since my test.

heitbaum, 2023-09-05:
> There is no new version since my test.

Can you please share your kernel crash from dmesg.

thomasb22, 2023-09-06:
Impossible, on crash the Raspberry Pi 4 reboot, so the dmesg is a new. There are no log from previous boot?

heitbaum, 2023-09-06:
This crashed within `/storage/.kodi/addons/inputstream.adaptive/inputstream.adaptive.so.20.3.12`this does not match the @perkabalo reported bug. Please review issues at https://github.com/xbmc/inputstream.adaptive/issues . There is a reported crash that I have been helping with there. But your crash may/may not be related.

heitbaum, 2023-09-06:
Closing issue as not related to original issue.

closed 2023-09-06 by heitbaum