LibreELEC/LibreELEC.tv #7160: [BUG] build warning - busybox 1.35.0

heitbaum, 2022-11-21:
### Describe the bug
Build warning during `make install`
```
LibreELEC.tv$ more build.LibreELEC-Generic.x86_64-11.0-devel/.threads/logs/170.log
INSTALL      busybox (target)
<string>:1: DeprecationWarning: 'crypt' is deprecated and slated for removal in Python 3.13
```

### To Reproduce
`make install`

### Informations
 - LE Version: master @ 21 Nov 2922
 - Hardware Platform: x86_64

### Log file

```
LibreELEC.tv$ more build.LibreELEC-Generic.x86_64-11.0-devel/.threads/logs/170.log
INSTALL      busybox (target)
<string>:1: DeprecationWarning: 'crypt' is deprecated and slated for removal in Python 3.13
```

### Additional context

antonlacon, 2022-11-21:
It's package.mk adding a root user's password in post_install. See `add_user` in config/functions. https://github.com/LibreELEC/LibreELEC.tv/pull/5736 replaced the python there with openssl.

Python docs say hashlib may replace what crypt is being used for instead. Need to look anyway to see about fixing the proftpd(?) addon's use of crypt in making password hashes too.

heitbaum, 2022-11-21:
Thanks for the info - references of the use of crypt in LE codebase.

https://github.com/LibreELEC/LibreELEC.tv/blob/5132be1a1d7667c5d601e8e18cd7ed695e6e9ac2/packages/sysutils/busybox/package.mk#L163

https://github.com/LibreELEC/LibreELEC.tv/blob/5132be1a1d7667c5d601e8e18cd7ed695e6e9ac2/config/functions#L1615

https://github.com/LibreELEC/LibreELEC.tv/blob/be009ddb371472369a0a197275cf66c039a8b5d0/packages/addons/service/proftpd/source/bin/proftpd.start#L102-L113

heitbaum, 2023-01-15:
Resolved in PR

closed 2023-01-15 by heitbaum