LibreELEC/LibreELEC.tv #7165: [BUG] Audio output choice is reset on reboot.

MparkG, 2022-11-24:
I set the Audio device to USB Soundcard from a list of HDMI, Built-in Analogue, USB Soundcard.
I play something, it works.
Next time i start the device, the choice is reset to builtin audio.
This happens on x86 and raspberry.
I continue using raspberry libreelec with version 9.something, which works fine.

is it that the USB-Device pops up too late, and Kodi or some program resets because it hasnt popped up yet?
It is a generic 8 channel usb soundcard. 

Thanks for the libreelec stuff.

lrusak, 2022-11-25:
It's likely your device isn't ready when Kodi starts. You may be able to see this in the Kodi log if it detects a new device being added. You may be able to add a delay by using the autostart.sh script and a simple `sleep 10` or something.

closed 2022-11-25 by lrusak

MparkG, 2023-03-07:
Hi again!
So, i thought id try v11!
Still the same problem, but now a bit more wierd.
I have had it work for the initial turning on sound and some navigational clicks. but after a second kodi swapps the device to the default one and there is no sound at all.
Here is the kodi.log maybe you can find something:
```
2023-03-07 01:16:01.258 T:891      info <general>: GUI format 1920x1080, Display 1920x1080 @ 60.000000 Hz
2023-03-07 01:16:01.301 T:898      info <general>: CAESinkALSA - Unable to open device "surround71" for playback
2023-03-07 01:16:01.303 T:898      info <general>: CAESinkALSA - Unable to open device "surround51" for playback
2023-03-07 01:16:01.303 T:898      info <general>: CAESinkALSA - Unable to open device "surround71" for playback
2023-03-07 01:16:01.305 T:898      info <general>: CAESinkALSA - Unable to open device "surround40" for playback
2023-03-07 01:16:01.305 T:898      info <general>: CAESinkALSA - Unable to open device "surround51" for playback
2023-03-07 01:16:01.305 T:898      info <general>: CAESinkALSA - Unable to open device "surround71" for playback
2023-03-07 01:16:01.398 T:898   warning <general>: Pulseaudio module module-allow-passthrough not loaded - opening PT devices might fail
2023-03-07 01:16:01.399 T:898      info <general>: Found 2 Lists of Devices
2023-03-07 01:16:01.399 T:898      info <general>: Enumerated ALSA devices:
2023-03-07 01:16:01.399 T:898      info <general>:     Device 1
2023-03-07 01:16:01.399 T:898      info <general>:         m_deviceName      : @
2023-03-07 01:16:01.399 T:898      info <general>:         m_displayName     : Default (vc4-hdmi MAI PCM i2s-hifi-0) ()
2023-03-07 01:16:01.399 T:898      info <general>:         m_displayNameExtra: 
2023-03-07 01:16:01.399 T:898      info <general>:         m_deviceType      : AE_DEVTYPE_PCM
2023-03-07 01:16:01.399 T:898      info <general>:         m_channels        : FL, FR, LFE, UNKNOWN1, FC, BC, BL, BR, BLOC, BROC, FLOC, FROC
2023-03-07 01:16:01.399 T:898      info <general>:         m_sampleRates     : 32000,44100,48000
2023-03-07 01:16:01.399 T:898      info <general>:         m_dataFormats     : AE_FMT_S24NE3,AE_FMT_S24NE4,AE_FMT_S32NE,AE_FMT_S16NE,AE_FMT_S16LE,AE_FMT_S16BE
2023-03-07 01:16:01.399 T:898      info <general>:         m_streamTypes     : No passthrough capabilities
2023-03-07 01:16:01.399 T:898      info <general>:     Device 2
2023-03-07 01:16:01.400 T:898      info <general>:         m_deviceName      : sysdefault
2023-03-07 01:16:01.400 T:898      info <general>:         m_displayName     : vc4-hdmi ()
2023-03-07 01:16:01.400 T:898      info <general>:         m_displayNameExtra: MAI PCM i2s-hifi-0
2023-03-07 01:16:01.400 T:898      info <general>:         m_deviceType      : AE_DEVTYPE_PCM
2023-03-07 01:16:01.400 T:898      info <general>:         m_channels        : FL, FR, LFE, UNKNOWN1, FC, BC, BL, BR, BLOC, BROC, FLOC, FROC
2023-03-07 01:16:01.400 T:898      info <general>:         m_sampleRates     : 32000,44100,48000
2023-03-07 01:16:01.400 T:898      info <general>:         m_dataFormats     : AE_FMT_S24NE3,AE_FMT_S24NE4,AE_FMT_S32NE,AE_FMT_S16NE,AE_FMT_S16LE,AE_FMT_S16BE
2023-03-07 01:16:01.400 T:898      info <general>:         m_streamTypes     : No passthrough capabilities
2023-03-07 01:16:01.400 T:898      info <general>:     Device 3
2023-03-07 01:16:01.400 T:898      info <general>:         m_deviceName      : sysdefault:CARD=vc4hdmi
2023-03-07 01:16:01.400 T:898      info <general>:         m_displayName     : vc4-hdmi (vc4hdmi)
2023-03-07 01:16:01.400 T:898      info <general>:         m_displayNameExtra: MAI PCM i2s-hifi-0
2023-03-07 01:16:01.400 T:898      info <general>:         m_deviceType      : AE_DEVTYPE_PCM
2023-03-07 01:16:01.400 T:898      info <general>:         m_channels        : FL, FR, LFE, UNKNOWN1, FC, BC, BL, BR, BLOC, BROC, FLOC, FROC
2023-03-07 01:16:01.400 T:898      info <general>:         m_sampleRates     : 32000,44100,48000
2023-03-07 01:16:01.400 T:898      info <general>:         m_dataFormats     : AE_FMT_S24NE3,AE_FMT_S24NE4,AE_FMT_S32NE,AE_FMT_S16NE,AE_FMT_S16LE,AE_FMT_S16BE
2023-03-07 01:16:01.400 T:898      info <general>:         m_streamTypes     : No passthrough capabilities
2023-03-07 01:16:01.400 T:898      info <general>:     Device 4
2023-03-07 01:16:01.400 T:898      info <general>:         m_deviceName      : hdmi:CARD=vc4hdmi,DEV=0
2023-03-07 01:16:01.400 T:898      info <general>:         m_displayName     : vc4-hdmi (vc4hdmi)
2023-03-07 01:16:01.400 T:898      info <general>:         m_displayNameExtra: ACR Acer 1080p PJ on HDMI
2023-03-07 01:16:01.400 T:898      info <general>:         m_deviceType      : AE_DEVTYPE_HDMI
2023-03-07 01:16:01.400 T:898      info <general>:         m_channels        : FL, FR
2023-03-07 01:16:01.400 T:898      info <general>:         m_sampleRates     : 32000,44100,48000
2023-03-07 01:16:01.400 T:898      info <general>:         m_dataFormats     : AE_FMT_RAW,AE_FMT_S24NE3,AE_FMT_S24NE4,AE_FMT_S32NE,AE_FMT_S16NE,AE_FMT_S16LE,AE_FMT_S16BE,AE_FMT_U8,AE_FMT_RAW
2023-03-07 01:16:01.400 T:898      info <general>:         m_streamTypes     : STREAM_TYPE_AC3,STREAM_TYPE_DTSHD,STREAM_TYPE_DTSHD_MA,STREAM_TYPE_DTSHD_CORE,STREAM_TYPE_DTS_1024,STREAM_TYPE_DTS_2048,STREAM_TYPE_DTS_512,STREAM_TYPE_EAC3,STREAM_TYPE_TRUEHD
2023-03-07 01:16:01.400 T:898      info <general>:     Device 5
2023-03-07 01:16:01.400 T:898      info <general>:         m_deviceName      : @:CARD=ICUSBAUDIO7D,DEV=0
[... cut ...]
2023-03-07 01:16:09.733 T:899   warning <general>: CAESinkAlsa::InitializeHW - Using default alsa values - set failed
2023-03-07 01:16:09.738 T:899     error <general>: CActiveAESink::OpenSink - no sink was returned
2023-03-07 01:16:09.738 T:898     error <general>: ActiveAE::InitSink - returned error

```

Should i go to the xbmc issues with this? this behaviour has existed on raspberry pi 3 and on a x64 intel celeron mini media pc.

Also, the HDMI CEC in this version doesnt turn off the tv anymore.

MparkG, 2023-03-07:
hah lol,
```
T:917      info <general>: service.bluetooth-audio: starting add-on
2023-03-07 01:16:09.432 T:917      info <general>: service.bluetooth-audio: setting default audio device "ALSA:@:CARD=Device,DEV=0" on start
```

i removed addon bt audio device changer (from team libreelec :P)
now it works fine!