LibreELEC/LibreELEC.tv #7166: [BUG] Invalid-key while connecting to wifi

daufinsyd, 2022-11-24:
### Describe the bug
Hello :)
Since upgrading to LireElec 10 (10.0.3) the wifi is really unstable.
Every now and then, LibreElec will forget the wifi password and won't accept to reconnect because of some "invalid-key" error even tho the password

Other times it needs to try a few times before the connection actually is established.

### To Reproduce
Steps to reproduce the behavior:
1. Boot the LibreElec as alway
2. (optional) if some wifi were previously configured and working it happens that libreelec won't connect to the wifi and will ask for it's key
3. Connect to some wifi (2,4 or 5GHz)
5. Connecting fails because of invalid key.

### Informations
 - LE Version: 10.0.3
 - Hardware Platform: RPi4

### Log file

### Additional context
What I tried :
 - rebooting it many times
 - tried wih different wifi band and AP
 - event AP from my phone right next to the raspberry don't work (sometimes yes, some times no)
 - raspberry is in its official case near the tv but far enough from interferences
 - removing the case doesn't help

I have two raspberries 4 with LibreElec, but having issues, so I don't think it's an hardware issue.
I also got an Odroid c2 which work flawlessly with the same wifi as the 2 other.

Even with LibreElec 9 there was wifi issues (failing to connect, without reason or because of input/output error) (for the two raspberries).

CSchlipp, 2022-12-16:
Same issue here, seems to have gotten worse with versions >nightly20221128

heitbaum, 2022-12-16:
> Same issue here, seems to have gotten worse with versions >nightly20221128

Are you having the issue with LE10 or LE1 nightlies?

CSchlipp, 2022-12-16:
> Are you having the issue with LE10 or LE1 nightlies?

LE11

heitbaum, 2022-12-16:
> > Are you having the issue with LE10 or LE1 nightlies?
> 
> LE11

Hard to compare LE10 and LE11 (we shouldn’t) 
- le10 runs wpasupplicant
- le11 runs iwd

For le11 please check with > 20221213 - as this is the new kernel (check in dmesg for kernel 6.1) and includes the Kodi rc1. We will need the logs etc …

heitbaum, 2023-01-15:
Inactive issue. Please reopen if still an issue and with updated information (using latest LE11 nightly.)

vilhelmgray, 2023-01-15:
I encountered this issue today with the latest [LE11 nightly (f4f7d8e)](https://test.libreelec.tv/11.0/Allwinner/pine-h64-model-b/LibreELEC-H6.arm-11.0-nightly-20230115-f4f7d8e-pine-h64-model-b.img.gz). I ran it on a **Pine H64 Model B**, so it seems that this issue is affecting more hardware platforms than just the **Raspberry Pi 4**.

If you reopen this issue, I'm willing to provide any debug logs you need to troubleshoot this; just guide me through what I should do.

heitbaum, 2023-01-15:
Updated as LE11 issue (which uses iwd)

JGouyau, 2023-02-08:
On last Nightly build, stop and restart wireless connection in parameter allow connecting to Wi-Fi. If that can help someone

Sebazzz, 2023-03-12:
The solution [from the LibreElec forum appears to works](https://forum.libreelec.tv/thread/26761-invalid-wifi-key/?postID=177925#post177925):

> To create the modprobe configuration file:
>
> ```
>    echo 'option iwlwifi amsdu_size=3' > /storage/.config/modprobe.d/iwlwifi.conf
> ```
>
> Then reboot.

vilhelmgray, 2023-03-12:
Why does this `amsdu_size=3` option seem to fix this issue? Is it because the device needs monitor mode as described in the (iwlwifi wiki](https://wireless.wiki.kernel.org/en/users/drivers/iwlwifi#about_the_monitorsniffer_mode)? Apparently this option will put pressure on the memory subsystem, but I suppose that can't be avoided if it's necessary for correct device operation.

Sebazzz, 2023-03-12:
I'm not sure - but I'm pretty sure it works. When I ran into the issue after upgrading to LE11, I rebooted, wiped the wifi settings, etc without result. Eventually I found that thread, applied the setting, and now it works and it keeps working even after reboots.

vilhelmgray, 2023-03-27:
I tried the `amsdu_size=3` modprobe configuration on **LibreELEC nightly-20230327-9da7818**, but it did not resolve the issue for my Pine H64 Model B device. So it seems like that's unfortunately not the solution for this issue. :slightly_frowning_face:

Sebazzz, 2023-03-28:
I tried to figure out troubleshooting steps and perhaps I'm adding more noise to to this issue than actual value, so take the below with a huge grain of salt.

---

In the ArchLinux wiki I read that iwd can sometimes come online too early, so I tried to add a systemd unit that simply calls /usr/bin/sleep. This didn't have any effect.

---

You can also run iwd in debugging mode, so that's what I tried next.

For this I ran these commands:

```sh
systemctl stop iwd.service
STATE_DIRECTORY=/tmp/iwtest
export STATE_DIRECTORY
mkdir -p $STATE_DIRECTORY
/usr/lib/iwd -d
```

Then in another SSH session, to attempt to connect to a wireless network:

```
iwctl station wlan0 scan
iwctl station wlan0 get-networks # This should list your networks
iwctl station wlan0 connect Skynet
```

[image]

Then I can see in the other SSH session:

```
build/build.LibreELEC-RPi4.arm-11.0.1/build/iwd-2.3/src/station.c:station_autoconnect_next() autoconnect: Trying SSID: Skynet Xtra
/build/build.LibreELEC-RPi4.arm-11.0.1/build/iwd-2.3/src/station.c:station_autoconnect_next() autoconnect: 'fc:ee:e6:00:83:d2' freq: 2457, rank: 591, strength: -4800
/build/build.LibreELEC-RPi4.arm-11.0.1/build/iwd-2.3/src/station.c:station_autoconnect_next() autoconnect: network_autoconnect: No such file or directory (-2)
/build/build.LibreELEC-RPi4.arm-11.0.1/build/iwd-2.3/src/wiphy.c:wiphy_select_akm() Network is WPA3-Personal...
/build/build.LibreELEC-RPi4.arm-11.0.1/build/iwd-2.3/src/station.c:station_autoconnect_next() autoconnect: Trying SSID: Skynet
/build/build.LibreELEC-RPi4.arm-11.0.1/build/iwd-2.3/src/station.c:station_autoconnect_next() autoconnect: No suitable BSSes found
/build/build.LibreELEC-RPi4.arm-11.0.1/build/iwd-2.3/src/wiphy.c:wiphy_select_akm() Network is WPA3-Personal...
/build/build.LibreELEC-RPi4.arm-11.0.1/build/iwd-2.3/src/station.c:station_autoconnect_next() autoconnect: Trying SSID: D23
/build/build.LibreELEC-RPi4.arm-11.0.1/build/iwd-2.3/src/station.c:station_autoconnect_next() autoconnect: '76:ac:b9:22:01:ee' freq: 2472, rank: 591, strength: -6100
/build/build.LibreELEC-RPi4.arm-11.0.1/build/iwd-2.3/src/station.c:station_autoconnect_next() autoconnect: network_autoconnect: No such file or directory (-2)
/build/build.LibreELEC-RPi4.arm-11.0.1/build/iwd-2.3/src/wiphy.c:wiphy_select_akm() Network is WPA3-Personal...
/build/build.LibreELEC-RPi4.arm-11.0.1/build/iwd-2.3/src/station.c:station_autoconnect_next() autoconnect: Trying SSID: Blackhole
/build/build.LibreELEC-RPi4.arm-11.0.1/build/iwd-2.3/src/station.c:station_autoconnect_next() autoconnect: 'fe:ee:e6:00:83:d2' freq: 2457, rank: 591, strength: -4700
/build/build.LibreELEC-RPi4.arm-11.0.1/build/iwd-2.3/src/station.c:station_autoconnect_next() autoconnect: network_autoconnect: No such file or directory (-2)
/build/build.LibreELEC-RPi4.arm-11.0.1/build/iwd-2.3/src/station.c:station_autoconnect_next() autoconnect: Trying SSID: RaspHTPC-WIFI
/build/build.LibreELEC-RPi4.arm-11.0.1/build/iwd-2.3/src/station.c:station_autoconnect_next() autoconnect: '76:ac:b9:32:01:ee' freq: 2472, rank: 591, strength: -6100
/build/build.LibreELEC-RPi4.arm-11.0.1/build/iwd-2.3/src/station.c:station_autoconnect_next() autoconnect: network_autoconnect: No such file or directory (-2)
/build/build.LibreELEC-RPi4.arm-11.0.1/build/iwd-2.3/src/station.c:station_autoconnect_next() autoconnect: Trying SSID: Solar-WiFi223W0101
/build/build.LibreELEC-RPi4.arm-11.0.1/build/iwd-2.3/src/station.c:station_autoconnect_next() autoconnect: '2c:9c:6e:9e:b6:26' freq: 2457, rank: 492, strength: -5900
/build/build.LibreELEC-RPi4.arm-11.0.1/build/iwd-2.3/src/station.c:station_autoconnect_next() autoconnect: network_autoconnect: No such file or directory (-2)
/build/build.LibreELEC-RPi4.arm-11.0.1/build/iwd-2.3/src/station.c:station_autoconnect_next() autoconnect: Trying SSID: ZiggoFE76B37
/build/build.LibreELEC-RPi4.arm-11.0.1/build/iwd-2.3/src/station.c:station_autoconnect_next() autoconnect: 'ac:22:05:4c:83:2a' freq: 5220, rank: 13, strength: -8600
/build/build.LibreELEC-RPi4.arm-11.0.1/build/iwd-2.3/src/station.c:station_autoconnect_next() autoconnect: network_autoconnect: No such file or directory (-2)
```

The "network_autoconnect" happens for any network, but the network I just tried to connect to shows "No suitable BSSes found". I can't determine whether this is of any significance. I also cannot find any other debug tracing or apparent messages of importance.

ajbathe, 2023-03-28:
Are you able to test your case with IWD v2.4 - I could guess that your issue is fixed - at least my "operation failed" issue disappeared when using not v2.3

ajbathe, 2023-03-29:
> Please retest with iwd2.4 - LibreELEC-yyyyy-11.0-nightly-20230329-a0f002b.img.gz (or newer) from https://test.libreelec.tv/11.0

I alread tested iwd 2.4 by using LE11.0 and LE11.0.1 offical and replacing only iwd by system drop-in, nothing else was changed. Autoconnect test (101 boots with 30s wait and asking whether wlan0 is up by a crontab entry showed 97% success rate. Looks good to me. Is there anything special with the nightly?

Sebazzz, 2023-03-29:
Unfortunately still the same issues, same symptoms (invalid key, input/output error).
Tested with:

[image]

--- 

For those who want to try, add "https://test.libreelec.tv/" as alternative update channel:
[image]

And pick a release:
[image]

heitbaum, 2023-03-30:
@Sebazzz good to know. I see a couple "debug" method - the first one looks to get the right information.
- https://bbs.archlinux.org/viewtopic.php?id=268631
- https://lore.kernel.org/all/0e7ab80d46b3e138910af83e5ba138fcb3178a92.camel@gmail.com/T/

For your SSID - "Skynet" can you please share what type is it ? 
It looks like it is 2.4Ghz Channel 10. 
It is not Hidden.
- Is it WPA3-PSK?
Have you tried WPA2-PSK?

Some questions / feedback - complicated / simple / special character PSK?
- is the PSK something like "12BobIsAGreatGuy"

From my use
- I haven't seen the error - for a while ... if at all the same ... 
- I have seen iwd not reconnect on a dropped connection
  - #6643
- I have seen the invalid "error" but not consistently like you are experiencing.

Sebazzz, 2023-03-30:
Skynet is WPA2/WPA3 (whatever that means on Unifi). The password is in range [A-z][0-9] only.

I'll try setting up a separate network with WPA2 only and check what that does.

Met vriendelijke groet,
Sebastiaan Dammann

  *   https://bbs.archlinux.org/viewtopic.php?id=268631<https://emea01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fbbs.archlinux.org%2Fviewtopic.php%3Fid%3D268631&data=05%7C01%7C%7Cea1b93b2bfb44f34fbc808db30c99ad0%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C638157414666721690%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=k3bUd7F1rHIWh%2FvXpHHEcT8zi%2Fam10nDKr9u48dHXo8%3D&reserved=0>
  *   ***@***.***/T/<https://emea01.safelinks.protection.outlook.com/?url=https%3A%2F%2Flore.kernel.org%2Fall%2F0e7ab80d46b3e138910af83e5ba138fcb3178a92.camel%40gmail.com%2FT%2F&data=05%7C01%7C%7Cea1b93b2bfb44f34fbc808db30c99ad0%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C638157414666721690%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=pO%2BNYqXprEJBkIkll4X1JPokpX7EQ5MGh6fscFctONA%3D&reserved=0>

     *   #6643<https://emea01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FLibreELEC%2FLibreELEC.tv%2Fissues%2F6643&data=05%7C01%7C%7Cea1b93b2bfb44f34fbc808db30c99ad0%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C638157414666721690%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=n7iwZn8BIh14Lotgg%2BMtNQDEZVzNL8ewvj%2FvzjTePhg%3D&reserved=0>

ajbathe, 2023-03-30:
does it work when you tell your router not to use WPA3?

vilhelmgray, 2023-03-30:
Here's a brief update from my side: I tried [LibreELEC-H6.arm-11.0-nightly-20230328-a0f002b-pine-h64-model-b.img.gz](https://test.libreelec.tv/11.0/Allwinner/pine-h64-model-b/LibreELEC-H6.arm-11.0-nightly-20230328-a0f002b-pine-h64-model-b.img.gz) but I'm still experiencing the `invalid-key` error message consistently; my network is set for WPA2, frequency available at 2.4Ghz/5Ghz, and the password is a mix of special characters and alphanumerical characters.

vilhelmgray, 2023-03-30:
Sorry, I don't have an ethernet cable so I'm only able to troubleshoot directly on the device. :slightly_frowning_face:

ajbathe, 2023-03-30:
then I am out of any ideas ... a kingdom for a network cable ;-)

chewitt, 2023-03-31:
You can enable hidden files/folders in Kodi settings and then use the Kodi file manager GUI to navigate-to and delete the folder (to see if that helps).

vilhelmgray, 2023-03-31:
I procured an Ethernet cable so I can SSH into the system now. There are no wifi directories/files under `/storage/.cache/connman` (as expected because this is a freshly formatted SD card) so I don't think that's the problem. However, I can now run any sort of commands you need for debug information.

I tried the `iwctl` commands as described [earlier in this thread](https://github.com/LibreELEC/LibreELEC.tv/issues/7166#issuecomment-1487338648). I can see my network when I issue `iwctl station wlan0 get-networks`, but trying to connect results in an `Operation failed` message. However I did get the following debug log print out:

```
build/build.LibreELEC-H6.arm-11.0-devel/build/iwd-2.4/src/agent.c:agent_register() agent register called
/build/build.LibreELEC-H6.arm-11.0-devel/build/iwd-2.4/src/agent.c:agent_register() agent :1.23 path /agent/802
/build/build.LibreELEC-H6.arm-11.0-devel/build/iwd-2.4/src/network.c:network_connect() 
/build/build.LibreELEC-H6.arm-11.0-devel/build/iwd-2.4/src/network.c:network_connect_psk() ask_passphrase: true
/build/build.LibreELEC-H6.arm-11.0-devel/build/iwd-2.4/src/agent.c:agent_request_passphrase() agent 0x14168c8 owner :1.23 path /agent/802
/build/build.LibreELEC-H6.arm-11.0-devel/build/iwd-2.4/src/agent.c:agent_send_next_request() send request to :1.23 /agent/802
/build/build.LibreELEC-H6.arm-11.0-devel/build/iwd-2.4/src/agent.c:agent_receive_reply() agent 0x14168c8 request id 88
/build/build.LibreELEC-H6.arm-11.0-devel/build/iwd-2.4/src/network.c:passphrase_callback() result 0
/build/build.LibreELEC-H6.arm-11.0-devel/build/iwd-2.4/src/network.c:network_generate_sae_pt() Generating PT for Group 19
/build/build.LibreELEC-H6.arm-11.0-devel/build/iwd-2.4/src/network.c:network_generate_sae_pt() Generating PT for Group 20
/build/build.LibreELEC-H6.arm-11.0-devel/build/iwd-2.4/src/netdev.c:netdev_cqm_rssi_update() 
/build/build.LibreELEC-H6.arm-11.0-devel/build/iwd-2.4/src/wiphy.c:wiphy_radio_work_insert() Inserting work item 13
/build/build.LibreELEC-H6.arm-11.0-devel/build/iwd-2.4/src/wiphy.c:wiphy_radio_work_next() Starting work item 13
/build/build.LibreELEC-H6.arm-11.0-devel/build/iwd-2.4/src/station.c:__station_connect_network() connecting to BSS 88:96:4e:37:ed:00
/build/build.LibreELEC-H6.arm-11.0-devel/build/iwd-2.4/src/station.c:station_enter_state() Old State: autoconnect_full, new state: connecting
/build/build.LibreELEC-H6.arm-11.0-devel/build/iwd-2.4/src/scan.c:scan_periodic_stop() Stopping periodic scan for wdev 1
CMD_SET_CQM failed: Operation not supported
/build/build.LibreELEC-H6.arm-11.0-devel/build/iwd-2.4/src/station.c:station_netdev_event() Associating
/build/build.LibreELEC-H6.arm-11.0-devel/build/iwd-2.4/src/netdev.c:netdev_mlme_notify() MLME notification Connect(46)
/build/build.LibreELEC-H6.arm-11.0-devel/build/iwd-2.4/src/netdev.c:netdev_connect_event() 
/build/build.LibreELEC-H6.arm-11.0-devel/build/iwd-2.4/src/netdev.c:netdev_connect_event() aborting and ignore_connect_event not set, proceed
/build/build.LibreELEC-H6.arm-11.0-devel/build/iwd-2.4/src/netdev.c:netdev_connect_event() expect_connect_failure not set, proceed
/build/build.LibreELEC-H6.arm-11.0-devel/build/iwd-2.4/src/wiphy.c:wiphy_radio_work_done() Work item 13 done
/build/build.LibreELEC-H6.arm-11.0-devel/build/iwd-2.4/src/station.c:station_connect_cb() 4, result: 2
/build/build.LibreELEC-H6.arm-11.0-devel/build/iwd-2.4/src/station.c:station_reset_connection_state() 4
/build/build.LibreELEC-H6.arm-11.0-devel/build/iwd-2.4/src/station.c:station_roam_state_clear() 4
/build/build.LibreELEC-H6.arm-11.0-devel/build/iwd-2.4/src/station.c:station_enter_state() Old State: connecting, new state: disconnected
/build/build.LibreELEC-H6.arm-11.0-devel/build/iwd-2.4/src/station.c:station_enter_state() Old State: disconnected, new state: autoconnect_quick
/build/build.LibreELEC-H6.arm-11.0-devel/build/iwd-2.4/src/scan.c:scan_periodic_start() Starting periodic scan for wdev 1
/build/build.LibreELEC-H6.arm-11.0-devel/build/iwd-2.4/src/wiphy.c:wiphy_radio_work_insert() Inserting work item 14
/build/build.LibreELEC-H6.arm-11.0-devel/build/iwd-2.4/src/wiphy.c:wiphy_radio_work_next() Starting work item 14
/build/build.LibreELEC-H6.arm-11.0-devel/build/iwd-2.4/src/netdev.c:netdev_mlme_notify() MLME notification Connect(46)
/build/build.LibreELEC-H6.arm-11.0-devel/build/iwd-2.4/src/netdev.c:netdev_connect_event() 
/build/build.LibreELEC-H6.arm-11.0-devel/build/iwd-2.4/src/netdev.c:netdev_connect_event() aborting and ignore_connect_event not set, proceed
Unexpected connection related event -- is another supplicant running?
```

Perhaps that `CMD_SET_CQM failed: Operation not supported` line is relevant to this issue. I'm not familiar with iwd, so I'm not sure what this means.

heitbaum, 2023-03-31:
A quick google for “CMD_SET_CQM” brings up heaps. But looking at them - 2 things that stand out (and worth testing)
- make sure WPA3 is not an option on the SSID
- Use only 2.4 GHz
- (And just use a simple ascii only pass phrase for the test)

hopefully that should illuminate those “issues” / “others with similar issues”

Sebazzz, 2023-04-07:
> Skynet is WPA2/WPA3 (whatever that means on Unifi). The password is in range [A-z][0-9] only. I'll try setting up a separate network with WPA2 only and check what that does.

Sorry it took a while to get back. I set up a wireless network with WPA2 only and [A-z] range password:

[image]

I still experienced the same issues (invalid-key, I/O error).

I'm not sure what changed with LibreElec 4.0, but that is obviously giving the trouble.

heitbaum, 2023-04-07:
Let’s try the following and see how far we get.

```
iwctl known-networks list
iwctl known-networks <"network name"> forget 
# all on them
iwctl station list
iwctl station wlan0 connect RaspHTPC-WIFI
```

fyi - my SSID is just a lowercase 6 [a-z]

Sebazzz, 2023-04-07:
I feel it is completely unrelated to security settings or WPA type:

- I changed the SSID to "raspie"
- Removed MAC filter
- Removed the password (so it is an open network now)

Just like above when I mentioned that the debug option of IWD fixed the issue (but it didn't), I was able to authenticate one time:

[image]

Then trying again, it failed again:
[image]

Retrying my attempt to get more logging out of the system and running iwd manually yields these logs:

```
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/agent.c:agent_register() agent register called
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/agent.c:agent_register() agent :1.27 path /agent/1058
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/network.c:network_connect()
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/netdev.c:netdev_cqm_rssi_update()
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/wiphy.c:wiphy_radio_work_insert() Inserting work item 12
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/wiphy.c:wiphy_radio_work_next() Starting work item 12
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/station.c:__station_connect_network() connecting to BSS da:21:f9:ff:37:d9
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/station.c:station_enter_state() Old State: autoconnect_full, new state: connecting
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/scan.c:scan_periodic_stop() Stopping periodic scan for wdev 1
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/station.c:station_netdev_event() Associating
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/netdev.c:netdev_mlme_notify() MLME notification Connect(46)
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/netdev.c:netdev_connect_event()
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/netdev.c:netdev_connect_event() aborting and ignore_connect_event not set, proceed
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/netdev.c:netdev_connect_event() expect_connect_failure not set, proceed
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/wiphy.c:wiphy_radio_work_done() Work item 12 done
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/station.c:station_connect_cb() 3, result: 2
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/station.c:station_reset_connection_state() 3
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/station.c:station_roam_state_clear() 3
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/station.c:station_enter_state() Old State: connecting, new state: disconnected
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/station.c:station_enter_state() Old State: disconnected, new state: autoconnect_quick
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/scan.c:scan_periodic_start() Starting periodic scan for wdev 1
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/wiphy.c:wiphy_radio_work_insert() Inserting work item 13
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/wiphy.c:wiphy_radio_work_next() Starting work item 13
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/scan.c:scan_notify() Scan notification Trigger Scan(33)
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/scan.c:scan_request_triggered() Passive scan triggered for wdev 1
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/scan.c:scan_periodic_triggered() Periodic scan triggered for wdev 1
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/agent.c:agent_disconnect() agent :1.27 disconnected
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/agent.c:agent_free() agent free 0x1788118
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/scan.c:scan_notify() Scan notification New Scan Results(34)
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/netdev.c:netdev_link_notify() event 16 on ifindex 3
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/scan.c:get_scan_callback() get_scan_callback
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/scan.c:scan_parse_bss_information_elements() Load: 40/255
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/scan.c:get_scan_callback() get_scan_callback
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/scan.c:scan_parse_bss_information_elements() Load: 40/255
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/scan.c:get_scan_callback() get_scan_callback
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/scan.c:scan_parse_bss_information_elements() Load: 40/255
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/scan.c:get_scan_callback() get_scan_callback
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/scan.c:get_scan_callback() get_scan_callback
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/scan.c:get_scan_callback() get_scan_callback
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/scan.c:scan_parse_bss_information_elements() Load: 42/255
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/scan.c:get_scan_callback() get_scan_callback
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/scan.c:scan_parse_bss_information_elements() Load: 42/255
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/scan.c:get_scan_callback() get_scan_callback
[... cut ...]
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/station.c:station_autoconnect_next() autoconnect: network_autoconnect: No such file or directory (-2)
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/wiphy.c:wiphy_radio_work_done() Work item 13 done
```

Nothing obvious, except perhaps:

```
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/station.c:station_autoconnect_next() autoconnect: Trying SSID: raspi
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/station.c:station_autoconnect_next() autoconnect: No suitable BSSes found
```

Sebazzz, 2023-04-07:
Just to be sure I updated to the latest nightly just now with the same issue.

[image]

vilhelmgray, 2023-04-07:
We know that this issue isn't happening in previous version of LibreELEC, and we know that it is happening in the latest version; it seems the next logical step is to perform a `git bisect` to determine the commit that introduces this issue. Hopefully that will give us an idea of what's causing the error.

Sebazzz, 2023-04-07:
If I read it correctly iwd was introduct in LibreELEC 4, so that seems to be the major change when it comes to networking.

vilhelmgray, 2023-04-07:
Ah, in that case a `git bisect` won't help unfortunately for us. :disappointed:

heitbaum, 2023-04-07:
This is where it is failing
- https://git.kernel.org/pub/scm/network/wireless/iwd.git/tree/src/network.c#n800

can you please set a preshared key for the testing. And share the output from.

`iw dev wlan0 scan ssid raspi`

hopefully that illuminates why the bss “doesn’t match” what it is looking for.

chewitt, 2023-04-07:
LE7.x to LE10.x use wpa_supplicant, and LE11 uses iwd. We know exactly when that change was committed, but that is not that useful because a) we already know the change is responsible, b) we aren't going to revert the change to fix a relatively small number or user issues. The issue needs to be triaged and reported (and then hopefully investigated further and fixed) upstream.

Sebazzz, 2023-04-07:
I changed `raspi` back to a PSK WPA2 network and gave it a password `github1234`:

[image]

```
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/station.c:station_autoconnect_next() autoconnect: Trying SSID: raspi
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/station.c:station_autoconnect_next() autoconnect: 'da:21:f9:ff:37:d9' freq: 2412, rank: 591, strength: -5400
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/station.c:station_autoconnect_next() autoconnect: network_autoconnect: No such file or directory (-2)
```

First lines logged when running `iwd -d`:

```
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/main.c:main() Using configuration directory /etc/iwd
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/storage.c:storage_create_dirs() Using state directory /tmp/iwtest
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/main.c:nl80211_appeared() Found nl80211 interface
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/module.c:iwd_modules_init()
station: Network configuration is disabled.
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/wsc.c:wsc_init()
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/knownnetworks.c:known_network_frequencies_load() No known frequency file found.
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/eap.c:__eap_method_enable()
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/eap-wsc.c:eap_wsc_init()
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/eap-md5.c:eap_md5_init()
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/eap-tls.c:eap_tls_init()
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/eap-ttls.c:eap_ttls_init()
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/eap-mschapv2.c:eap_mschapv2_init()
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/eap-sim.c:eap_sim_init()
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/eap-aka.c:eap_aka_prime_init()
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/eap-aka.c:eap_aka_init()
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/eap-peap.c:eap_peap_init()
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/eap-gtc.c:eap_gtc_init()
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/eap-pwd.c:eap_pwd_init()
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/manager.c:manager_wiphy_dump_callback() New wiphy phy0 added (0)
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/manager.c:manager_wiphy_dump_done()
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/manager.c:manager_filtered_wiphy_dump_done()
Wiphy: 0, Name: phy0
        Permanent Address: dc:a6:32:c8:da:b6
        2.4Ghz Band:
                Bitrates (non-HT):
                         1.0 Mbps
                         2.0 Mbps
                         5.5 Mbps
                        11.0 Mbps
                         6.0 Mbps
                         9.0 Mbps
                        12.0 Mbps
                        18.0 Mbps
                        24.0 Mbps
                        36.0 Mbps
                        48.0 Mbps
                        54.0 Mbps
                HT Capabilities:
                        HT40
                        Short GI for 20Mhz
                        Short GI for 40Mhz
                HT RX MCS indexes:
                        0-7
        5Ghz Band:
                Bitrates (non-HT):
                         6.0 Mbps
                         9.0 Mbps
                        12.0 Mbps
                        18.0 Mbps
                        24.0 Mbps
                        36.0 Mbps
                        48.0 Mbps
                        54.0 Mbps
                HT Capabilities:
                        HT40
                        Short GI for 20Mhz
                        Short GI for 40Mhz
                HT RX MCS indexes:
                        0-7
                VHT Capabilities:
                        Short GI for 80Mhz
                        Max RX MCS: 0-9 for NSS: 1
                        Max TX MCS: 0-9 for NSS: 1
        Ciphers: BIP-CMAC-128 CCMP-128 TKIP
        Supported iftypes: ad-hoc station ap p2p-client p2p-go p2p-device
Wiphy phy0 will only use the default interface
```

Specifically of interest perhaps:

```
station: Network configuration is disabled.
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/wsc.c:wsc_init()
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/knownnetworks.c:known_network_frequencies_load() No known frequency file found.
```

vilhelmgray, 2023-04-07:
@Sebazzz Are you able to connect if you remove the password entirely? If I switch from `WPA2` to just `none`, I still receive the "invalid-key" error message initially, but interestingly it does connect after a reboot but not a power-off event.

Sebazzz, 2023-04-07:
> @Sebazzz Are you able to connect if you remove the password entirely? If I switch from `WPA2` to just `none`, I still receive the "invalid-key" error message initially, but interestingly it does connect after a reboot but not a power-off event.

No and yes. One time I was able to connect. The other attempts failed.

Sebazzz, 2023-04-07:
I think I found a way to get some more debug logs, by doing this:

```
cat <<EOF
[General]
LogLevel=debug
EOF >> /tmp/iwd.conf
systemctl stop iwd.service
mkdir -p /tmp/iwtest
STATE_DIRECTORY=/tmp/iwtest CONFIGURATION_DIRECTORY=/tmp/ /usr/lib/iwd -d
```

These are the logs (I've removed the /build/build.... prefix):

```
/src/agent.c:agent_register() agent register called
/src/agent.c:agent_register() agent :1.72 path /agent/1107
/src/network.c:network_connect()
/src/network.c:network_connect_psk() ask_passphrase: true
/src/agent.c:agent_request_passphrase() agent 0x11f1fe0 owner :1.72 path /agent/1107
/src/agent.c:agent_send_next_request() send request to :1.72 /agent/1107
/src/agent.c:agent_receive_reply() agent 0x11f1fe0 request id 39
/src/network.c:passphrase_callback() result 0
/src/network.c:network_generate_sae_pt() Generating PT for Group 19
/src/network.c:network_generate_sae_pt() Generating PT for Group 20
/src/netdev.c:netdev_cqm_rssi_update()
/src/wiphy.c:wiphy_radio_work_insert() Inserting work item 6
/src/wiphy.c:wiphy_radio_work_next() Starting work item 6
/src/station.c:__station_connect_network() connecting to BSS da:21:f9:ff:37:d9
/src/station.c:station_enter_state() Old State: autoconnect_full, new state: connecting
/src/scan.c:scan_periodic_stop() Stopping periodic scan for wdev 1
/src/station.c:station_netdev_event() Associating
/src/netdev.c:netdev_mlme_notify() MLME notification Connect(46)
/src/netdev.c:netdev_connect_event()
/src/netdev.c:netdev_connect_event() aborting and ignore_connect_event not set, proceed
/src/netdev.c:netdev_connect_event() expect_connect_failure not set, proceed
/src/wiphy.c:wiphy_radio_work_done() Work item 6 done
/src/station.c:station_connect_cb() 3, result: 2
/src/netdev.c:netdev_cqm_rssi_update()
/src/wiphy.c:wiphy_radio_work_insert() Inserting work item 7
/src/wiphy.c:wiphy_radio_work_next() Starting work item 7
/src/station.c:__station_connect_network() connecting to BSS 76:ac:b9:22:01:ee
/src/station.c:station_try_next_bss() Attempting to connect to next BSS 76:ac:b9:22:01:ee
/src/station.c:station_netdev_event() Associating
/src/netdev.c:netdev_mlme_notify() MLME notification Connect(46)
/src/netdev.c:netdev_connect_event()
/src/netdev.c:netdev_connect_event() aborting and ignore_connect_event not set, proceed
/src/netdev.c:netdev_connect_event() expect_connect_failure not set, proceed
/src/wiphy.c:wiphy_radio_work_done() Work item 7 done
/src/station.c:station_connect_cb() 3, result: 2
/src/station.c:station_reset_connection_state() 3
/src/station.c:station_roam_state_clear() 3
/src/station.c:station_enter_state() Old State: connecting, new state: disconnected
/src/station.c:station_enter_state() Old State: disconnected, new state: autoconnect_quick
/src/scan.c:scan_periodic_start() Starting periodic scan for wdev 1
/src/wiphy.c:wiphy_radio_work_insert() Inserting work item 8
/src/wiphy.c:wiphy_radio_work_next() Starting work item 8
/src/scan.c:scan_notify() Scan notification Trigger Scan(33)
/src/scan.c:scan_request_triggered() Passive scan triggered for wdev 1
/src/scan.c:scan_periodic_triggered() Periodic scan triggered for wdev 1
/src/agent.c:agent_disconnect() agent :1.72 disconnected
/src/agent.c:agent_free() agent free 0x11f1fe0
```

heitbaum, 2023-04-07:
So with a WPA2 PSK configured - the bss error is gone. 
Thus the BSS is now compatible with iwd.

What is the output of `iw dev wlan0 scan ssid raspi` and `iwctl known-networks list`

chewitt, 2023-04-07:
For the sake of asking .. what brand(s) of routers are people using?

vilhelmgray, 2023-04-07:
I'm getting this error on both my Pixel 6a hotspot, as well as an Archer A6 MU-MIMO.

heitbaum, 2023-04-07:
> For the sake of asking .. what brand(s) of routers are people using?

I’m running against a Cisco 9130AX and Apple Airport Timemachine

Sebazzz, 2023-04-07:
> So with a WPA2 PSK configured - the bss error is gone. Thus the BSS is now compatible with iwd.
> 
> What is the output of `iw dev wlan0 scan ssid raspi` and `iwctl known-networks list`

[image]

```
src/station.c:station_dbus_scan() Scan called from DBus
src/wiphy.c:wiphy_radio_work_insert() Inserting work item 13
src/wiphy.c:wiphy_radio_work_next() Starting work item 13
src/scan.c:scan_notify() Scan notification Trigger Scan(33)
src/scan.c:scan_request_triggered() Passive scan triggered for wdev 1
src/station.c:station_dbus_scan_triggered() station_scan_triggered: 0
src/station.c:station_dbus_scan_triggered() Scan triggered for wlan0 subset 0
src/agent.c:agent_disconnect() agent :1.92 disconnected
src/agent.c:agent_free() agent free 0x20569c8
src/scan.c:scan_notify() Scan notification New Scan Results(34)
src/netdev.c:netdev_link_notify() event 16 on ifindex 3
src/scan.c:get_scan_callback() get_scan_callback
src/scan.c:scan_parse_bss_information_elements() Load: 50/255
src/scan.c:get_scan_callback() get_scan_callback
src/scan.c:scan_parse_bss_information_elements() Load: 50/255
src/scan.c:get_scan_callback() get_scan_callback
src/scan.c:scan_parse_bss_information_elements() Load: 50/255
src/scan.c:get_scan_done() get_scan_done
src/station.c:station_add_seen_bss() Processing BSS 'd6:21:f9:ff:37:d9' with SSID: D23, freq: 2412, rank: 591, strength: -6000, data_rate: 72.2
src/station.c:station_add_seen_bss() Processing BSS 'd0:21:f9:ff:37:d9' with SSID: Skynet, freq: 2412, rank: 591, strength: -6000, data_rate: 72.2
src/station.c:station_add_seen_bss() Processing BSS 'da:21:f9:ff:37:d9' with SSID: raspi, freq: 2412, rank: 591, strength: -6100, data_rate: 72.2
src/station.c:station_add_seen_bss() Processing BSS 'da:21:f9:ff:37:da' with SSID: Prei23, freq: 5180, rank: 1473, strength: -5600, data_rate: 180.0
src/station.c:station_add_seen_bss() Processing BSS 'd6:21:f9:ff:37:da' with SSID: D23, freq: 5180, rank: 1473, strength: -5600, data_rate: 180.0
src/station.c:station_add_seen_bss() Processing BSS 'd0:21:f9:ff:37:da' with SSID: Skynet, freq: 5180, rank: 1473, strength: -5600, data_rate: 180.0
src/station.c:station_add_seen_bss() Processing BSS 'ae:22:05:4c:82:2f' with SSID: , freq: 5220, rank: 13, strength: -8300, data_rate: 2.0
src/station.c:station_add_seen_bss() BSS has hidden SSID
src/station.c:station_add_seen_bss() Processing BSS 'ac:22:05:4c:83:2a' with SSID: ZiggoFE76B37, freq: 5220, rank: 13, strength: -8400, data_rate: 2.0
src/wiphy.c:wiphy_radio_work_insert() Inserting work item 14
src/wiphy.c:wiphy_radio_work_done() Work item 13 done
src/wiphy.c:wiphy_radio_work_next() Starting work item 14
src/scan.c:scan_notify() Scan notification Trigger Scan(33)
src/scan.c:scan_request_triggered() Passive scan triggered for wdev 1
src/station.c:station_dbus_scan_triggered() station_scan_triggered: 0
src/station.c:station_dbus_scan_triggered() Scan triggered for wlan0 subset 1
src/scan.c:scan_notify() Scan notification New Scan Results(34)
src/netdev.c:netdev_link_notify() event 16 on ifindex 3
src/scan.c:get_scan_callback() get_scan_callback
src/scan.c:scan_parse_bss_information_elements() Load: 50/255
src/scan.c:get_scan_callback() get_scan_callback
src/scan.c:scan_parse_bss_information_elements() Load: 50/255
src/scan.c:get_scan_callback() get_scan_callback
src/scan.c:scan_parse_bss_information_elements() Load: 50/255
src/scan.c:get_scan_callback() get_scan_callback
src/scan.c:scan_parse_bss_information_elements() Load: 45/255
src/scan.c:get_scan_callback() get_scan_callback
src/scan.c:scan_parse_bss_information_elements() Load: 45/255
src/scan.c:get_scan_callback() get_scan_callback
src/scan.c:scan_parse_bss_information_elements() Load: 45/255
src/scan.c:get_scan_callback() get_scan_callback
src/scan.c:get_scan_callback() get_scan_callback
src/scan.c:get_scan_done() get_scan_done
src/station.c:station_add_seen_bss() Processing BSS 'da:21:f9:ff:37:da' with SSID: Prei23, freq: 5180, rank: 1473, strength: -5600, data_rate: 180.0
src/station.c:station_add_seen_bss() Processing BSS 'd6:21:f9:ff:37:da' with SSID: D23, freq: 5180, rank: 1473, strength: -5600, data_rate: 180.0
src/station.c:station_add_seen_bss() Processing BSS 'd0:21:f9:ff:37:da' with SSID: Skynet, freq: 5180, rank: 1473, strength: -5600, data_rate: 180.0
src/station.c:station_add_seen_bss() Processing BSS 'd6:21:f9:ff:37:d9' with SSID: D23, freq: 2412, rank: 591, strength: -6000, data_rate: 72.2
src/station.c:station_add_seen_bss() Processing BSS 'd0:21:f9:ff:37:d9' with SSID: Skynet, freq: 2412, rank: 591, strength: -6000, data_rate: 72.2
src/station.c:station_add_seen_bss() Processing BSS 'da:21:f9:ff:37:d9' with SSID: raspi, freq: 2412, rank: 591, strength: -6100, data_rate: 72.2
src/station.c:station_add_seen_bss() Processing BSS 'ae:22:05:4c:82:2f' with SSID: , freq: 5220, rank: 13, strength: -8400, data_rate: 2.0
src/station.c:station_add_seen_bss() BSS has hidden SSID
src/station.c:station_add_seen_bss() Processing BSS 'ac:22:05:4c:83:2a' with SSID: ZiggoFE76B37, freq: 5220, rank: 13, strength: -8500, data_rate: 2.0
src/wiphy.c:wiphy_radio_work_insert() Inserting work item 15
src/wiphy.c:wiphy_radio_work_done() Work item 14 done
src/wiphy.c:wiphy_radio_work_next() Starting work item 15
[... 52 lines cut ...]
src/network.c:network_generate_sae_pt() Generating PT for Group 19
src/network.c:network_generate_sae_pt() Generating PT for Group 20
src/netdev.c:netdev_cqm_rssi_update()
[... cut ...]
src/station.c:station_reset_connection_state() 3
src/station.c:station_roam_state_clear() 3
src/station.c:station_enter_state() Old State: connecting, new state: disconnected
src/station.c:station_enter_state() Old State: disconnected, new state: autoconnect_quick
src/agent.c:agent_disconnect() agent :1.93 disconnected
src/agent.c:agent_free() agent free 0x20569c8
```

> For the sake of asking .. what brand(s) of routers are people using?

Unifi Dream Machine and Unifi 6 Mesh, but I also can't connect to my iPhone hotspot.

heitbaum, 2023-04-07:
It got this far - which is successful.
`https://git.kernel.org/pub/scm/network/wireless/iwd.git/tree/src/netdev.c#n2818 `
But fails auth shortly there after I think.

I would like to see the known-networks output to make sure it has/hasn’t cached the key.
Also still need to see the iw dev output - what is set in the RSN…

Sebazzz, 2023-04-07:
> I would like to see the known-networks output to make sure it has/hasn’t cached the key.

```
RaspHTPC:~ # iwctl known-networks list
                                 Known Networks
--------------------------------------------------------------------------------
  Name                              Security     Hidden     Last connected
--------------------------------------------------------------------------------
```

(As also shown in my screenshot 😊)

> Also still need to see the iw dev output - what is set in the RSN…

Not sure what you need, the above is all what I can pull out of iwd.

heitbaum, 2023-04-07:
> ```
> RaspHTPC:~ # iwctl known-networks list
>                                  Known Networks
> -------------------------------------------------------------------------------- [...]

This is good. No cached password.

Sebazzz, 2023-04-07:
Assuming you meant `IWD_GENL_DEBUG`:

```
src/agent.c:agent_register() agent register called
src/agent.c:agent_register() agent :1.98 path /agent/1175
src/network.c:network_connect()
src/network.c:network_connect_psk() ask_passphrase: true
src/agent.c:agent_request_passphrase() agent 0x1c75a30 owner :1.98 path /agent/1175
src/agent.c:agent_send_next_request() send request to :1.98 /agent/1175
src/agent.c:agent_receive_reply() agent 0x1c75a30 request id 38
src/network.c:passphrase_callback() result 0
src/network.c:network_generate_sae_pt() Generating PT for Group 19
src/network.c:network_generate_sae_pt() Generating PT for Group 20
src/netdev.c:netdev_cqm_rssi_update()
src/wiphy.c:wiphy_radio_work_insert() Inserting work item 6
src/wiphy.c:wiphy_radio_work_next() Starting work item 6
src/station.c:__station_connect_network() connecting to BSS da:21:f9:ff:37:d9
src/station.c:station_enter_state() Old State: autoconnect_full, new state: connecting
[GENL] > 30 00 00 00 18 00 05 00 1a 00 00 00 90 04 00 00  0...............
[GENL]   3f 00 00 00 08 00 03 00 03 00 00 00 14 00 5e 80  ?.............^.
[GENL]   08 00 01 00 ba ff ff ff 08 00 02 00 05 00 00 00  ................
[GENL] < 24 00 00 00 02 00 00 01 1a 00 00 00 90 04 00 00  $...............
[GENL]   00 00 00 00 30 00 00 00 18 00 05 00 1a 00 00 00  ....0...........
[GENL]   90 04 00 00                                      ....
[GENL] > bc 00 00 00 18 00 05 00 1b 00 00 00 90 04 00 00  ................
[GENL]   2e 00 00 00 08 00 03 00 03 00 00 00 08 00 26 00  ..............&.
[GENL]   6c 09 00 00 0a 00 06 00 da 21 f9 ff 37 d9 00 00  l........!..7...
[GENL]   09 00 34 00 72 61 73 70 69 00 00 00 08 00 35 00  ..4.raspi.....5.
[GENL]   00 00 00 00 24 00 fe 00 0b 9d a7 c8 bb 1b 0a 29  ....$..........)
[GENL]   7a 6b 82 52 38 4c 79 ec 49 fc 00 27 a6 29 71 68  zk.R8Ly.I..'.)qh
[GENL]   7c 44 0a 55 6b 54 85 13 04 00 46 00 04 00 cc 00  |D.UkT....F.....
[GENL]   08 00 49 00 04 ac 0f 00 08 00 4a 00 04 ac 0f 00  ..I.......J.....
[GENL]   08 00 4c 00 02 ac 0f 00 08 00 4b 00 02 00 00 00  ..L.......K.....
[GENL]   04 00 44 00 26 00 2a 00 30 14 01 00 00 0f ac 04  ..D.&.*.0.......
[GENL]   01 00 00 0f ac 04 01 00 00 0f ac 02 00 00 7f 0a  ................
[GENL]   00 00 08 00 00 00 00 00 00 01 00 00              ............
[GENL] < 24 00 00 00 02 00 00 01 1b 00 00 00 90 04 00 00  $...............
[GENL]   00 00 00 00 bc 00 00 00 18 00 05 00 1b 00 00 00  ................
[GENL]   90 04 00 00                                      ....
/build/build.LibreELEC-RPi4.arm-11.0-devel/build/iwd-2.4/src/station.c:station_netdev_event() Associating
[GENL] < 38 00 00 00 18 00 00 00 00 00 00 00 00 00 00 00  8...............
[GENL]   2e 01 00 00 08 00 01 00 00 00 00 00 08 00 03 00  ................
[GENL]   03 00 00 00 0a 00 06 00 00 00 00 00 00 00 00 00  ................
[GENL]   06 00 48 00 10 00 00 00                          ..H.....
src/netdev.c:netdev_mlme_notify() MLME notification Connect(46)
src/netdev.c:netdev_connect_event()
src/netdev.c:netdev_connect_event() aborting and ignore_connect_event not set, proceed
src/netdev.c:netdev_connect_event() expect_connect_failure not set, proceed
src/wiphy.c:wiphy_radio_work_done() Work item 6 done
src/station.c:station_connect_cb() 3, result: 2
src/station.c:station_reset_connection_state() 3
src/station.c:station_roam_state_clear() 3
src/station.c:station_enter_state() Old State: connecting, new state: disconnected
src/station.c:station_enter_state() Old State: disconnected, new state: autoconnect_quick
src/agent.c:agent_disconnect() agent :1.98 disconnected
src/agent.c:agent_free() agent free 0x1c75a30
```

heitbaum, 2023-04-07:
What’s the output of
`iw dev wlan0 scan ssid raspi`

Sebazzz, 2023-04-07:
This is the output:

```
RaspHTPC:~ # iw dev wlan0 scan ssid raspi
BSS da:21:f9:ff:37:d9(on wlan0)
        last seen: 4164.111s [boottime]
        TSF: 0 usec (0d, 00:00:00)
        freq: 2412
        beacon interval: 100 TUs
        capability: ESS Privacy ShortPreamble ShortSlotTime RadioMeasure (0x1431)
        signal: -56.00 dBm
        last seen: 0 ms ago
        SSID: raspi
        Supported rates: 1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0
        DS Parameter set: channel 1
        TIM: DTIM Count 0 DTIM Period 1 Bitmap Control 0x0 Bitmap[0] 0x0
        Country: NL     Environment: Indoor/Outdoor
                Channels [1 - 13] @ 20 dBm
        TPC report: TX power: 17 dBm
        ERP: <no flags>
        Extended supported rates: 24.0 36.0 48.0 54.0
        RSN:     * Version: 1
                 * Group cipher: CCMP
                 * Pairwise ciphers: CCMP
                 * Authentication suites: PSK
                 * Capabilities: 1-PTKSA-RC 1-GTKSA-RC (0x0000)
        BSS Load:
                 * station count: 0
                 * channel utilisation: 40/255
                 * available admission capacity: 31250 [*32us]
        RM enabled capabilities:
                Capabilities: 0x73 0xd0 0x00 0x00 0x0c
                        Link Measurement
                        Neighbor Report
                        Beacon Passive Measurement
                        Beacon Active Measurement
                        Beacon Table Measurement
                        LCI Measurement
                        Transmit Stream/Category Measurement
                        Triggered Transmit Stream/Category
                        FTM Range Report
                        Civic Location Measurement
                Nonoperating Channel Max Measurement Duration: 0
                Measurement Pilot Capability: 4
        HT capabilities:
                Capabilities: 0x9ad
                        RX LDPC
                        HT20
                        SM Power Save disabled
                        RX HT20 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 7935 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: No restriction (0x00)
                HT TX/RX MCS rate indexes supported: 0-15
        HT operation:
                 * primary channel: 1
                 * secondary channel offset: no secondary
                 * STA channel width: 20 MHz
                 * RIFS: 0
                 * HT protection: no
                 * non-GF present: 1
                 * OBSS non-GF present: 0
                 * dual beacon: 0
                 * dual CTS protection: 0
                 * STBC beacon: 0
                 * L-SIG TXOP Prot: 0
                 * PCO active: 0
[... 54 lines cut ...]
                        OM Control UL MU Data Disable RX
                HE PHY Capabilities: (0x006008881d41810c110800):
                        LDPC Coding in Payload
                        HE SU PPDU with 1x HE-LTF and 0.8us GI
                        STBC Rx <= 80MHz
                        DCM Max Constellation Rx: 1
                        SU Beamformer
                        SU Beamformee
                        Beamformee STS <= 80Mhz: 7
                        Sounding Dimensions <= 80Mhz: 1
                        Ng = 16 SU Feedback
                        Codebook Size SU Feedback
                        PPE Threshold Present
                        HE SU PPDU & HE PPDU 4x HE-LTF 0.8us GI
                        Max NC: 1
                        HE ER SU PPDU 4x HE-LTF 0.8us GI
                        HE ER SU PPDU 1x HE-LTF 0.8us GI
                        RX 1024-QAM
                PPE Threshold 0xfa 0xff 0xfa 0xff 0x19 0x1c 0xc7 0x71
        WMM:     * Parameter version 1
                 * BE: CW 15-1023, AIFSN 3
                 * BK: CW 15-1023, AIFSN 7
                 * VI: CW 7-15, AIFSN 2, TXOP 3008 usec
                 * VO: CW 3-7, AIFSN 2, TXOP 1504 usec
```

heitbaum, 2023-04-07:
Well we are getting close:
`src/station.c:station_connect_cb() 3, result: 2`
https://git.kernel.org/pub/scm/network/wireless/iwd.git/tree/src/station.c#n3124

so iwd is failing with NETDEV_RESULT_ASSOCIATION_FAILED / not authentication.

so probably time to check the access points debug logs and find out why.

heitbaum, 2023-04-07:
Only RSN difference is that my AP offer 4- instead of 1-PTKSA-RC 1-GTKSA-RC https://www.cwnp.com/uploads/802-11_rsn_ft.pdf

on a side note - my supported rates are (stops devices negotiating low rates and causing issue) - don’t think that is the problem. 
Supported rates: 11.0 6.0 9.0 12.0 18.0 24.0 36.0 48.0 
        Extended supported rates: 54.0* 

nothing else jumps out. I will look at mine side by side. But almost sleep time.

Sebazzz, 2023-04-07:
While searching on how I can get more information out of my Unifi devices I let the LibreElec running, and when I wanted to retry connecting, I was able to connect. 

This is what a successful attempt looks like:

```
src/netdev.c:netdev_mlme_notify() MLME notification Connect(46)
src/netdev.c:netdev_connect_event()
src/netdev.c:netdev_connect_event() aborting and ignore_connect_event not set, proceed
src/netdev.c:netdev_connect_event() expect_connect_failure not set, proceed
src/netdev.c:parse_request_ies()
src/netdev.c:netdev_connect_event() Request / Response IEs parsed
src/station.c:station_handshake_event() Setting keys
src/netdev.c:netdev_connect_ok()
src/station.c:station_connect_cb() 3, result: 0
src/station.c:station_connect_ok()
src/station.c:station_enter_state() Old State: connecting, new state: connected
src/wiphy.c:wiphy_radio_work_done() Work item 9 done
```

I was able to disconnect and reconnect without issue. However, when terminated and restarted IWD I have the same issue again.

Just another datapoint, maybe it helps.

heitbaum, 2023-04-07:
This getting somewhere.

try `iwctl known-networks list` during the working and non-Working scenarios. It should show the know network and allow the connect - but as you say - on restart, it will not have the known-network (pretty sure)

Sebazzz, 2023-04-07:
> so iwd is failing with NETDEV_RESULT_ASSOCIATION_FAILED / not authentication.
> 
> so probably time to check the access points debug logs and find out why.

First I though unifi wasn't logging, but it turns out `iwd` isn't even getting to the point of talking to the access point!

This is my iPhone connecting to `raspi` network and I deliberately enter a wrong password:

```
Apr  7 18:20:17 WirelessAttic daemon.info hostapd[12028]: ath2: STA 46:93:8f:a3:e4:57 IEEE 802.11: authenticated
Apr  7 18:20:17 WirelessAttic kern.err kernel: [158863.042890] wlan: [0:I:ANY] [UNSPECIFIED] vap-2(ath2): [46:93:8f:a3:e4:57]recv auth frame with algorithm 0 seq 1
Apr  7 18:20:17 WirelessAttic kern.err kernel: [158863.042890]
Apr  7 18:20:17 WirelessAttic kern.err kernel: [158863.042890]
Apr  7 18:20:17 WirelessAttic kern.warn kernel: [158863.048852] ieee80211_update_node_ecm_flags: node with aid 1 and mac 46:93:8f:a3:e4:57 has been tagged [ dvlan ecm accelerated ]
Apr  7 18:20:17 WirelessAttic kern.err kernel: [158863.055440] wlan: [0:I:ANY] [UNSPECIFIED] vap-2(ath2): [46:93:8f:a3:e4:57]station associated at aid 1: short preamble, short slot time, QoSHE20 cap 0x1431
Apr  7 18:20:17 WirelessAttic kern.err kernel: [158863.055440]
Apr  7 18:20:17 WirelessAttic kern.err kernel: [158863.055440]
Apr  7 18:20:17 WirelessAttic daemon.info hostapd[12028]: ath2: STA 46:93:8f:a3:e4:57 IEEE 802.11: associated (aid 1)
Apr  7 18:20:17 WirelessAttic daemon.info hostapd[12028]: ath2: STA 46:93:8f:a3:e4:57 WPA: sending 1/4 msg of 4-Way Handshake
Apr  7 18:20:17 WirelessAttic kern.warn kernel: [158863.076303] [ath2] ieee80211_set_node_vid node 46:93:8f:a3:e4:57 lan[0] dvlan stas 0
Apr  7 18:20:17 WirelessAttic daemon.info hostapd[12028]: ath2: STA 46:93:8f:a3:e4:57 WPA: received EAPOL-Key frame (2/4 Pairwise)
Apr  7 18:20:17 WirelessAttic daemon.info hostapd[12028]: ath2: STA 46:93:8f:a3:e4:57 WPA: sending 1/4 msg of 4-Way Handshake
Apr  7 18:20:17 WirelessAttic daemon.info hostapd[12028]: ath2: STA 46:93:8f:a3:e4:57 WPA: received EAPOL-Key frame (2/4 Pairwise)
Apr  7 18:20:17 WirelessAttic user.info libubnt[12020]: wevent[12020]: wevent.ubnt_custom_event(): EVENT_STA_JOIN ath2: 46:93:8f:a3:e4:57 / 1
Apr  7 18:20:18 WirelessAttic daemon.info hostapd[12028]: ath2: STA 46:93:8f:a3:e4:57 WPA: sending 1/4 msg of 4-Way Handshake
Apr  7 18:20:18 WirelessAttic daemon.info hostapd[12028]: ath2: STA 46:93:8f:a3:e4:57 WPA: received EAPOL-Key frame (2/4 Pairwise)
Apr  7 18:20:19 WirelessAttic daemon.info hostapd[12028]: ath2: STA 46:93:8f:a3:e4:57 WPA: sending 1/4 msg of 4-Way Handshake
Apr  7 18:20:19 WirelessAttic daemon.info hostapd[12028]: ath2: STA 46:93:8f:a3:e4:57 WPA: received EAPOL-Key frame (2/4 Pairwise)
Apr  7 18:20:20 WirelessAttic daemon.notice /sbin/udhcpc[12023]: ifconfig: SIOCSIFNETMASK: Address not available
Apr  7 18:20:20 WirelessAttic user.notice syswrapper: ipready.br0.9 = 0.0.0.0
Apr  7 18:20:20 WirelessAttic daemon.notice /sbin/udhcpc[12023]: syswrapper: ipready.br0.9 = 0.0.0.0
Apr  7 18:20:20 WirelessAttic daemon.info hostapd[12028]: ath2: STA 46:93:8f:a3:e4:57 IEEE 802.11: disassociated
Apr  7 18:20:20 WirelessAttic user.info libubnt[12020]: wevent[12020]: wevent.ubnt_custom_event(): EVENT_STA_LEAVE ath2: 46:93:8f:a3:e4:57 / 1
```

This is my iPhone connecting to `raspi`, entering the correct password:

```
Apr  7 18:20:44 WirelessAttic daemon.info hostapd[12028]: ath2: STA 46:93:8f:a3:e4:57 IEEE 802.11: authenticated
Apr  7 18:20:44 WirelessAttic kern.err kernel: [158889.598563] wlan: [0:I:ANY] [UNSPECIFIED] vap-2(ath2): [46:93:8f:a3:e4:57]recv auth frame with algorithm 0 seq 1
Apr  7 18:20:44 WirelessAttic kern.err kernel: [158889.598563]
Apr  7 18:20:44 WirelessAttic kern.err kernel: [158889.598563]
Apr  7 18:20:44 WirelessAttic kern.warn kernel: [158889.606655] ieee80211_update_node_ecm_flags: node with aid 1 and mac 46:93:8f:a3:e4:57 has been tagged [ dvlan ecm accelerated ]
Apr  7 18:20:44 WirelessAttic kern.err kernel: [158889.611153] wlan: [0:I:ANY] [UNSPECIFIED] vap-2(ath2): [46:93:8f:a3:e4:57]station associated at aid 1: short preamble, short slot time, QoSHE20 cap 0x1431
Apr  7 18:20:44 WirelessAttic kern.err kernel: [158889.611153]
Apr  7 18:20:44 WirelessAttic kern.err kernel: [158889.611153]
Apr  7 18:20:44 WirelessAttic daemon.info hostapd[12028]: ath2: STA 46:93:8f:a3:e4:57 IEEE 802.11: associated (aid 1)
Apr  7 18:20:44 WirelessAttic daemon.info hostapd[12028]: ath2: STA 46:93:8f:a3:e4:57 WPA: sending 1/4 msg of 4-Way Handshake
Apr  7 18:20:44 WirelessAttic daemon.info hostapd[12028]: ath2: STA 46:93:8f:a3:e4:57 WPA: received EAPOL-Key frame (2/4 Pairwise)
Apr  7 18:20:44 WirelessAttic daemon.info hostapd[12028]: ath2: STA 46:93:8f:a3:e4:57 WPA: sending 3/4 msg of 4-Way Handshake
Apr  7 18:20:44 WirelessAttic kern.warn kernel: [158889.630610] [ath2] ieee80211_set_node_vid node 46:93:8f:a3:e4:57 lan[0] dvlan stas 0
Apr  7 18:20:44 WirelessAttic daemon.info hostapd[12028]: ath2: STA 46:93:8f:a3:e4:57 WPA: received EAPOL-Key frame (4/4 Pairwise)
Apr  7 18:20:44 WirelessAttic daemon.info hostapd[12028]: ath2: STA 46:93:8f:a3:e4:57 WPA: pairwise key handshake completed (RSN)
Apr  7 18:20:44 WirelessAttic user.info libubnt[12020]: wevent[12020]: wevent.ubnt_custom_event(): EVENT_STA_JOIN ath2: 46:93:8f:a3:e4:57 / 1
Apr  7 18:20:44 WirelessAttic user.info libubnt[12020]: wevent[12020]: wevent.ubnt_custom_event(): EVENT_STA_IP ath2: 46:93:8f:a3:e4:57 / 192.168.0.134
[... cut ...]
Apr  7 18:20:47 WirelessAttic daemon.info stahtd: stahtd[12021]: [STA-TRACKER].stahtd_dump_event(): {"message_type":"STA_ASSOC_TRACKER","mac":"46:93:8f:a3:e4:57","vap":"ath2","event_type":"success","assoc_status":"0","traffic_delta":"3140000","dns_responses":"14","dns_timeouts":"0","ip_delta":"1180000","ip_assign_type":"dhcp","assoc_delta":"30000","auth_delta":"10000","event_id":"4","auth_ts":"158889.593902","arp_reply_gw_seen":"yes","dns_resp_seen":"yes","sta_dc_reason":"sta left","avg_rssi":"-64"}
```

However, when I try to connect with LibreELEC/iwd, nothing is logged. That probably means it isn't even talking to the access point. Things are already failing somewhere earlier on.

mglae, 2023-04-07:
@vilhelmgray @Sebazzz @chewitt The Invalid-key was already reported for LE 10.0 with wpa_supplicant. Please read OP.

Sebazzz, 2023-04-07:
> 

But:

> Same issue here, seems to have gotten worse with versions

vilhelmgray, 2023-04-07:
I tried LE 10.0.4 and LE 10.0.2. I did not get an "invalid-key" error, but instead would regularly experience "connect-fail" messages. Sometimes it would connect successfully, but other times it would crash Kodi entirely. It felt very unstable overall.

LE 10.0.2 is when support for my device (Pine H64 Model B) was introduced, so I'm not able to test any further back than that. @Sebazzz are you able to test your device earlier than that?

Sebazzz, 2023-04-07:
> I tried LE 10.0.4 and LE 10.0.2. I did not get an "invalid-key" error, but instead would regularly experience "connect-fail" messages.

I rarely had connection issues with my RPi 3B+ and RPi 4 with LibreELEC 10.x. 

I had some performance issues, particularly on the 3B+, but I more blamed it on that I only had a single access point one floor below.

LibreELEC 11 is completely unusable for me.

rdorsch, 2023-04-07:
I think I see the same issue with LE11, Wetek Play 2 (Amlogic) hardware, and D-Link DIR-505 Router with OpenWrt firmware:

https://forum.libreelec.tv/thread/26928-wetek-play-2-no-wifi-with-box-image/

vilhelmgray, 2023-04-07:
It's possible my connection issues in LE 10 series are caused by an unrelated issue that happens to be resolved in LE 11, and unfortunately a new issue in LE 11 is the cause of the new "invalid-key" error messages I see.

[116 more replies]

closed 2024-03-16 by heitbaum