LibreELEC/LibreELEC.tv #7192: [BUG] |Random restarts over and over

xen0m0rph2020, 2022-12-01:
### Describe the bug
So a change was added to LE after 15th October that has broken TVMBHelper plugin. On all nightlies after 15th Oct Kodi would just restart 2 to 3 times when idle or when moving around home screen when you use a skin that uses the tvmbhelper add on.
Bugs were raised on skin github but it impacts all skins that use it and so it must be a LE/Kodi issue. FOr reference see the bug and logs here: https://github.com/jurialmunkey/skin.arctic.horizon.2/issues/676 

### To Reproduce
Steps to reproduce the behavior:
Start LE and use a skin that uses TVMBHelper like Artic Horizon 2

### Informations
 - LE Version: [e.g. 9.2.1] Nightly after 15/10/22
 - Hardware Platform: [e.g. RPi3]: RPi 4b

### Log file
[kodi_crashlog_20221130200446.log](https://github.com/LibreELEC/LibreELEC.tv/files/10136050/kodi_crashlog_20221130200446.log)

### Additional context

xen0m0rph2020, 2022-12-04:
The developer of the plugin cannot reproduce the bugs on Linux or Windows.
He thinks its most likely a change made to Python. Can anyone help? Work with the developer?
This is broken for me and I cannot run any skin that uses this plugin on my Pi.

saxoloman, 2022-12-17:
Is this still an issue after RC1 ?
I didnt update in ages

antonlacon, 2023-01-15:
This is still an issue? If I followed the upstream issue report then #7340 should have resolved it.

heitbaum, 2023-01-15:
Closing as Inactive issue. Please reopen if still an issue and with updated information (using latest LE11 nightly.)

closed 2023-01-15 by heitbaum