LibreELEC/LibreELEC.tv #7277: [BUG]RPI4 HDMI/Display is broken/no signal after nightly 12th dec.

the-dreamer, 2022-12-20:
### Describe the bug
Latest master is given a no signal on hdmi. In log file the edid file cannot loaded. if Edid file is disabled, no edid can be fetched from display. 

latest working nightly is from 12th Dec. So far the latest image with kernel 6.0.8. 

I assume there are some changes which either cannot load edid files anymore or the file location is expected somewhere else or edid is broken in 6.1 kernel. 

but that is only my first assumption. 

### To Reproduce
Steps to reproduce the behavior:
load latest nightly

### Informations

### Log file

Dec 08 23:01:13 LibreELEC kernel: Console: switching to colour frame buffer device 160x45
Dec 08 23:01:13 LibreELEC kernel:         [00] ZERO 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Dec 08 23:01:13 LibreELEC kernel:         [00] ZERO 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Dec 08 23:01:13 LibreELEC kernel:         [00] ZERO 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Dec 08 23:01:13 LibreELEC kernel:         [00] ZERO 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Dec 08 23:01:13 LibreELEC kernel:         [00] ZERO 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Dec 08 23:01:13 LibreELEC kernel:         [00] ZERO 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Dec 08 23:01:13 LibreELEC kernel:         [00] ZERO 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Dec 08 23:01:13 LibreELEC kernel:         [00] ZERO 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Dec 08 23:01:13 LibreELEC kernel: EDID block 0 is all zeroes
Dec 08 23:01:13 LibreELEC kernel: [drm:edid_load] *ERROR* Requesting EDID firmware "edid/edid.dat" failed (err=-2)
Dec 08 23:01:13 LibreELEC kernel: platform HDMI-A-1: Direct firmware load for edid/edid.dat failed with error -2
Dec 08 23:01:13 LibreELEC kernel: [drm] Initialized vc4 0.0.0 20140616 for gpu on minor 0

### Additional context

HiassofT, 2022-12-21:
Please provide a full pastekodi log as mentioned in the wiki - snippets are useless

the-dreamer, 2022-12-21:
I will do, hopefully later today or tomorrow. 

after some more tests: 

The Snippet is indeed useless. 

my last full functional system was a build end of september. HDMI Video and Audio. 
Last working Display was as mention in this ticket the nightly 12th dec. HDMI Audio is not working anymore. I don't know when Audio was broken. i have not tried all builds from september to 12th december.

Maybe as first hint for me: would do a clean install help? Is there something, that has been changed? For Example A config file for example. Last year the config.txt and distconfig.txt was introduced and broke also my system.

the-dreamer, 2022-12-24:
little late but now the logs: 

nightly 12th December: HDMI Picture yes / HDMI Audio NO: http://ix.io/4jqs

nightly 23rd December: No HDMI Singal at all, CEC Is still working: http://ix.io/4jqu

nightly 30th September, at that time around i built my images: Working HDMI as expected: http://ix.io/4jqv

HiassofT, 2022-12-27:
The log shows you didn't configure an initrd with a cpio of the edid(s) so the edid isn't available when the driver is loaded - it was pure luck that it seems to have worked before.

You can use the internal "create-edid-cpio" script to create edid.cpio in /flash, then add "initramfs edid.cpio" to the end of config.txt

closed 2022-12-27 by HiassofT

the-dreamer, 2022-12-28:
Thanks for that advice. I fixed that. Unfortunately the issue persists. Again: 12th signal yes/no audio and now 27th still with no signal at all (showed on avr and TV do not show any resolution): 

12th December: http://ix.io/4jPu
27th December: http://ix.io/4jPw

Maybe there is still something wrong on my site.

the-dreamer, 2022-12-31:
Very strange: It looks like, the edid file is somehow making troubles. I connected the rpi4 to another TV and fetch a new edid file and now everything is working okay. 

I am asking me, why my working edid file was working for years and then suddenly it is not good anymore and why the system is not compaining it.

HiassofT, 2022-12-31:
It's really hard to tell, especially since you also added an explicit video mode (1280x720M@50e, not sure if you did that on purpose to fix some issue)

In general I recommend using only the getedid script, that takes care about setting everything up properly

the-dreamer, 2022-12-31:
Originally i had 1280x720@50D in the boot param. But changed nothing. 

The reason is to force the boot splash to use 720p. otherwise it would output 1080p or anything other which is not supported on the pioneer plasma. 1080i not not supported anymore for some versions ago. Kodi itself is still supporting the 1080i.  

the edid file is used for many years and was used originally from my x86 Libreelec System before i had a new tv for that. 

The pioneer itself has a broken edid (crc) and is only valid together with the avr. But the AVR is adding also some 1080p resolutions. 

the edid file on rpi was necessary since the config.txt is not working anymore for hdmi configuration.

the-dreamer, 2023-01-02:
Maybe i can get some light in the dark. The issue is in general solved for me. But maybe i can understand the root issue: 

Edid-decode Pioneer <- Onkyo ARV <- RPI4: 
edid-decode (hex):

00 ff ff ff ff ff ff 00 3d cb 42 0a 00 00 00 00
00 14 01 03 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 10 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 10 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 fc 00 54
58 2d 53 52 35 30 38 0a 20 20 20 20 00 00 00 fd
00 17 f0 0f 7e 11 00 0a 20 20 20 20 20 20 01 6c

02 03 42 40 53 84 03 02 0e 0f 23 24 10 05 93 12
11 1d 1e 25 26 01 1f 14 38 09 7f 07 0f 7f 07 17
07 50 3f 06 c0 4d 02 00 57 06 00 5f 7e 01 67 54
00 83 4f 00 00 6c 03 0c 00 30 00 80 21 c0 ff 08
80 33 00 00 00 10 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 10 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 10 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 d8

----------------

Block 0, Base EDID:
  EDID Structure Version & Revision: 1.3
  Vendor & Product Identification:
    Manufacturer: ONK
    Model: 2626
    Made in: 2010
  Basic Display Parameters & Features:
    Analog display
    Signal Level Standard: 0.700 : 0.300 : 1.000 V p-p
    Blank level equals black level
    Image size is variable
    Gamma: 1.00
    Monochrome or grayscale display
  Color Characteristics:
    Red  : 0.0000, 0.0000
    Green: 0.0000, 0.0000
    Blue : 0.0000, 0.0000
    White: 0.0000, 0.0000
  Established Timings I & II: none
  Standard Timings:
  Detailed Timing Descriptors:
    Dummy Descriptor:
    Dummy Descriptor:
    Display Product Name: 'TX-SR508'
    Display Range Limits:
      Monitor ranges (GTF): 23-240 Hz V, 15-126 kHz H, max dotclock 170 MHz
  Extension blocks: 1
Checksum: 0x6c

----------------

Block 1, CTA-861 Extension Block:
  Revision: 3
  Basic audio support
  Native detailed modes: 0
  Video Data Block:
    VIC   4:  1280x720    60.000000 Hz  16:9     45.000 kHz     74.250000 MHz (native)
    VIC   3:   720x480    59.940060 Hz  16:9     31.469 kHz     27.000000 MHz
    VIC   2:   720x480    59.940060 Hz   4:3     31.469 kHz     27.000000 MHz
    VIC  14:  1440x480    59.940060 Hz   4:3     31.469 kHz     54.000000 MHz
    VIC  15:  1440x480    59.940060 Hz  16:9     31.469 kHz     54.000000 MHz
    VIC  35:  2880x480    59.940060 Hz   4:3     31.469 kHz    108.000000 MHz
    VIC  36:  2880x480    59.940060 Hz  16:9     31.469 kHz    108.000000 MHz
    VIC  16:  1920x1080   60.000000 Hz  16:9     67.500 kHz    148.500000 MHz
    VIC   5:  1920x1080i  60.000000 Hz  16:9     33.750 kHz     74.250000 MHz
[... 177 lines cut ...]
      Audio Format Code dependent value: 0x01
    MAT (MLP):
      Max channels: 8
      Supported sample rates (kHz): 192 96 48
      Supports only Dolby TrueHD
  Speaker Allocation Data Block:
    FL/FR - Front Left/Right
    LFE1 - Low Frequency Effects 1
    FC - Front Center
    BL/BR - Back Left/Right
    RLC/RRC - Rear Left/Right of Center (Deprecated)
  Vendor-Specific Data Block (HDMI), OUI 00-0C-03:
    Source physical address: 3.0.0.0
    Supports_AI
  Detailed Timing Descriptors:
    DTD 3:   720x480    59.940060 Hz   3:2     31.469 kHz     27.000000 MHz (708 mm x 398 mm)
                 Hfront   16 Hsync  62 Hback   60 Hpol N
                 Vfront    9 Vsync   6 Vback   30 Vpol N
    DTD 4:   720x576    50.000000 Hz   5:4     31.250 kHz     27.000000 MHz (708 mm x 398 mm)
                 Hfront   12 Hsync  64 Hback   68 Hpol N
                 Vfront    5 Vsync   5 Vback   39 Vpol N
    DTD 5:  1280x720    60.000000 Hz  16:9     45.000 kHz     74.250000 MHz (708 mm x 398 mm)
                 Hfront  110 Hsync  40 Hback  220 Hpol P
                 Vfront    5 Vsync   5 Vback   20 Vpol P
Checksum: 0x82  Unused space in Extension Block: 11 bytes

HiassofT, 2023-01-04:
Testing with the broken edid indicates the linux DRM layer trips over the missing "Digital display" bit in the EDID - according to the EDID the TV is an Analog, Monochrome or grayscale display - and thus no RGB or YCbCr modes are available and the display stays blank