LibreELEC/LibreELEC.tv #7278: [BUG] Ryzen 5600g using ff-hevc (sw)

atoulmin, 2022-12-21:
Hi,

I’ve built a new Ryzen 5600g system with the Asus b550 mitx motherboard.  Using the nightly build of LibreELEC it seems to be not using hardware decoding.  I’m aware of the issues with the current kernel and amd. However this is regardless of which build I use.

I’ve pasted my log here.
[http://ix.io/4jcl](url)

SupervisedThinking, 2022-12-21:
What's the output `vainfo` if you run it in a console?

SupervisedThinking, 2022-12-21:
Well no log into your system per console & run `vainfo` which should give you some result like this:

```
phoenix:~ # vainfo
Trying display: x11
libva info: VA-API version 1.16.0
libva info: Trying to open /usr/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_16
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.16 (libva 2.16.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 22.6.4 (2461190509)
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileNone                   :	VAEntrypointStats
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointFEI
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointFEI
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointFEI
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointFEI
```

Maybe you could also test some of those video samples? https://test-videos.co.uk/bigbuckbunny/mp4-h265

atoulmin, 2022-12-21:
I will do that tonight when I get home from work. It’s interesting that switching the display with the “\” key and back fixes the colour output. 

I will try again tonight.

atoulmin, 2022-12-22:
Ok so this is vainfo after playing the video from the start normally

LibreELEC:~ # vainfo
Trying display: drm
libva info: VA-API version 1.16.0
libva info: Trying to open /usr/lib/dri/radeonsi_drv_video.so
libva info: Found init function __vaDriverInit_1_16
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.16 (libva 2.16.0)
vainfo: Driver version: Mesa Gallium driver 22.2.2 for AMD Radeon Graphics (renoir, LLVM 15.0.4, DRM 3.48, 6.0.6)
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
      VAProfileNone                   : VAEntrypointVideoProc
LibreELEC:~

smp79, 2022-12-22:
https://gitlab.freedesktop.org/mesa/mesa/-/issues/7126

atoulmin, 2022-12-22:
https://salsa.debian.org/xorg-team/lib/mesa/-/commit/78fe00e7470aab38ab214161e28557ef2ff4d9ff

Possible fix?

atoulmin, 2022-12-23:
I just confirmed by putting in a Radeon 6400 that it has the same issue. So this seems to affect APU and GPU

closed 2022-12-27 by mglae