LibreELEC/LibreELEC.tv #7299: [BUG] Display corruption AMD

atoulmin, 2022-12-26:
Hi,

The current nightly builds appear to have display corruption on AMD GPU / APU. 

I have managed to track down the build that introduced the bug “corruption”. 

The nightly build that was compiled on the 2022-12-01 introduces the corruption. 

It does not occur on the build compiled on the 2022-11-30. I’m not sure what changed here. However whatever it was introduced the display corruption. In the menu system and playing videos. 

[image]

SupervisedThinking, 2022-12-26:
Well we need more detailed information e.g. which nightly did you use? GBM or X11 builds? What's the exact Mesa version? It's probably related to Mesa 22.3.y

atoulmin, 2022-12-26:
This is the nightly build.  

https://test.libreelec.tv/11.0/Generic/Generic/LibreELEC-Generic.x86_64-11.0-nightly-20221201-732acb2.img.gz

It also happens on every build since then in LE 11 nightly. 

When I get back home I’ll get the rest of the information 

[LE 11](https://test.libreelec.tv/11.0/Generic/Generic/LibreELEC-Generic.x86_64-11.0-nightly-20221201-732acb2.img.gz)

atoulmin, 2022-12-26:
Ok,

so this is the information from the build with no corruption

LibreELEC (community): nightly-20221130-5c149b0 (Generic.×86_64)
LibreELEC:~ # vainfo
Trying display: drm
libva info: VA-API version 1.16.0 libva info: Trying to open /us/lib/dri/radeonsi_dr_video.so libva info: Found init function __vaDriverInit_1
16
libva info: va_openDriver () returns 0 vainfo: VA-API version: 1.16 (libva 2.16.0)
vainfo: Driver version: Mesa Gallium driver 22.2.4 for AMD Radeon Graphics (renoir, LLVM 15.0.5, DRM 3.48, 6.0.9) vainfo: Supported profile and entrypoints
VAProfileMPEG2Simple
VAEntrypointVLD
VAProfileMPEG2Main
VAEntrypointVLD
VAProfileJPEGBaseline
VAEntrypointVLD
VAProfileVP9Profileo
VAEntrypointVLD
VAProfileVP9Profile2
VAEntrypointVLD
VAProfileNone
VAEntrypointVideoProc

This is the build with corruption

LibreELEC (community): nightly-20221201-732acb2 (Generic.×86_64)
LibreELEC:~ # vainfo
Trying display: drm
libva info: VA-API version 1.16.0 libva info: Trying to open /us/lib/dri/radeonsi_dr_video.so libva info: Found init function __vaDriverInit_1
16
libva info: va_openDriver () returns 0 vainfo: VA-API version: 1.16 (libva 2.16.0)
vainfo: Driver version: Mesa Gallium driver 22.3.0 for AMD Radeon Graphics (renoir, LLVM 15.0.6, DRM 3.48, 6.0.10) vainfo: Supported profile and entrypoints
VAProfileMPEG2Simple
VAEntrypointVLD
VAProfileMPEG2Main
VAEntrypointVLD
VAProfileJPEGBaseline
VAEntrypointVLD
VAProfileVP9Profileo
VAEntrypointVLD
VAProfileVP9Profile2
VAEntrypointVLD
VAProfileNone
LibreELEC:~ # I
VAEntrypointVideoProc

atoulmin, 2022-12-26:
Just tried the latest build with the Radeon 6400.  Still get display corruption.

LibreELEC (community): nightly-20221225-99de708 (Generic.x86_64)
LibreELEC:~ # vainfo
Trying display: drm
libva info: VA-API version 1.16.0
libva info: Trying to open /usr/lib/dri/radeonsi_drv_video.so
libva info: Found init function __vaDriverInit_1_16
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.16 (libva 2.16.0)
vainfo: Driver version: Mesa Gallium driver 22.3.1 for AMD Radeon RX 6400 (navi24, LLVM 15.0.6, DRM 3.49, 6.1.0)
vainfo: Supported profile and entrypoints
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
      VAProfileNone                   : VAEntrypointVideoProc
LibreELEC:~ #

SupervisedThinking, 2022-12-26:
Could you give the latest "legacy" nightly a try? https://test.libreelec.tv/11.0/Generic/Generic-legacy/ you should be able to seamlessly swap between them.

If both fail we new it's not limited to OpenGL or GLES

The missing decoding profiles should be fixed in another PR which adds the decoders.

atoulmin, 2022-12-26:
I tried the latest legacy, no video corruption.  Also noticed it doesn't detect HDR in display capabilities. 

LibreELEC (community): nightly-20221225-99de708 (Generic-legacy.x86_64)
LibreELEC:~ # vainfo
Trying display: x11
libva info: VA-API version 1.16.0
libva info: Trying to open /usr/lib/dri/radeonsi_drv_video.so
libva info: Found init function __vaDriverInit_1_16
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.16 (libva 2.16.0)
vainfo: Driver version: Mesa Gallium driver 22.3.1 for AMD Radeon RX 6400 (navi24, LLVM 15.0.6, DRM 3.49, 6.1.0)
vainfo: Supported profile and entrypoints
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
      VAProfileNone                   : VAEntrypointVideoProc

SupervisedThinking, 2022-12-26:
HDR won't work with X11, it's a GBM only feature but at least we can narrow it down to the builds which use GLES.

atoulmin, 2022-12-26:
Ok I assumed that was the case.  More than happy to try different builds or anything else to help find the issue.

SupervisedThinking, 2022-12-27:
Can you take some screenshots of the issue? `CTRL+S` should take some & store them in `/storage/screenshots`

I guess if this wasn't already fixed in [staging/22.3](https://github.com/Mesa3D/mesa/commits/staging/22.3) we need to open an issue at the mesa bug tracker.

atoulmin, 2022-12-27:
Sure, I’ll take some this morning. I’ll also do a video.

atoulmin, 2022-12-27:
Well one good thing, the latest PR fixed the HW acceleration. 

I tried multiple times but Ctrl+s does not capture the glitch. The screenshot looks fine. 

I took a video and photo with my phone. 

One other bug I’ve found too. Kodi will not output HDR colourspace correctly when u start a movie. You have to keep it playing return to the menu then back to video and it’s correct. One bug at a time. 

Here is the photo and video

[image]

atoulmin, 2022-12-27:
https://youtu.be/FG0bxNfu6gU

atoulmin, 2022-12-28:
https://youtu.be/kBzyGiSPqiE

second video. YouTube removed first one?

atoulmin, 2022-12-28:
Also Iookong at actually playback. It appears to be dropping frames.

atoulmin, 2022-12-29:
I just got myself an Intel ARC 750. So I now have an Intel 12th gen Nuc, Amd 5600g, Radeon 6400 and Arc 750 for testing.

heitbaum, 2022-12-30:
- Mesa update in #7312 - latest bug fix release - only seems to be minors

atoulmin, 2022-12-30:
Yeah. Will try it once merged and report back.

atoulmin, 2022-12-30:
Just tested latest build with Mesa 22.3.2. Still have the same issue with graphic corruption.

heitbaum, 2022-12-30:
> Just tested latest build with Mesa 22.3.2. Still have the same issue with graphic corruption.

Will need to report it to the mesa team - and hopefully a fix available.

>>
If you find any issues, please report them here:
https://gitlab.freedesktop.org/mesa/mesa/-/issues/new

The next bugfix release is due in two weeks, on January 11th.
<<

atoulmin, 2022-12-30:
Created issue as requested

https://gitlab.freedesktop.org/mesa/mesa/-/issues/7977

atoulmin, 2022-12-31:
https://gitlab.freedesktop.org/mesa/mesa/-/issues/7977#note_1700594

[M Briar](https://gitlab.freedesktop.org/mbriar)
[@mbriar](https://gitlab.freedesktop.org/mbriar)
· [25 minutes ago](https://gitlab.freedesktop.org/mesa/mesa/-/issues/7977#note_1700594)

Can you try mesa_glthread=false?

lrusak, 2022-12-31:
Ssh in and run the following commands
```
systemctl stop kodi
echo "mesa_glthread=false" >> /run/libreelec/kodi.conf
systemctl start kodi
```

atoulmin, 2022-12-31:
Yep tried it with no success, Kodi launched and graphic problem still happening

mglae, 2022-12-31:
> echo "mesa_glthread=false" >> /run/libreelec/kodi.conf

Does not work (is overwritten by kodi-config)

Use `/storage/.config/kodi.conf`

atoulmin, 2022-12-31:
> > echo "mesa_glthread=false" >> /run/libreelec/kodi.conf
> 
> Does not work (is overwritten by kodi-config)
> 
> Use `/storage/.config/kodi.conf`

I’ll just edit that file and add the line in to test I’m guessing.

chewitt, 2022-12-31:
You need to create the file, it doesn't exist by default.

atoulmin, 2023-01-01:
> You need to create the file, it doesn't exist by default.

Ok so I did as suggested and this fixed the problem.  So disabling GLTHREAD fixes the display corruption.

Only have one issue left which i will create a new bug.  But well done this fixes it.

heitbaum, 2023-02-09:
@atoulmin - need to test with mesa 22.3.5 - https://gitlab.freedesktop.org/mesa/mesa/-/blob/22.3/docs/relnotes/22.3.5.rst

Marek Olšák (6):

Revert "radeonsi: enable glthread by default"
radeonsi: fix RB+ blending with sRGB formats
radeonsi/gfx11: unset SAMPLE_MASK_TRACKER_WATERMARK to fix hangs
amd: split GFX1103 into GFX1103_R1 and GFX1103_R2
amd: fix tile_swizzle on gfx11 - should be shifted by 10 bits, not 8
amd: fix typo in shadowed uconfig registers on gfx11

atoulmin, 2023-02-09:
Happy to try it, however we still have this issue.

https://github.com/LibreELEC/LibreELEC.tv/issues/7324

atoulmin, 2023-02-10:
Confirm latest mesa, fixes problem.

closed 2023-02-10 by heitbaum