LibreELEC/LibreELEC.tv #7300: [BUG] raspberry pi irrecord does not work with dtoverlay=gpio-ir

thawn, 2022-12-27:
### Describe the bug
There is a known incompatibility between lirc and gpio-ir that prevents irrecord from working with dtoverlay=gpio-ir

There is a patch and a thorough description for how to install it available:
https://gist.github.com/billpatrianakos/cb72e984d4730043fe79cbe5fc8f7941

unfortunately, i cannot install this patch myself, because apt does not work

also, this bug has been fixed in the lirc version on raspberry pi os since October 2021

### To Reproduce
* connect an ir receiver via gpio to a raspi
* enable dtoverlay=gpio-ir
* try to record remote commands via irrecord

### Informations
 - LE Version: 9.2.8 and 10.0.3
 - Hardware Platform: RPi4

HiassofT, 2022-12-27:
Please test with the latest LE11 nightly from here https://test.libreelec.tv/11.0/RPi/RPi4/ (currently 20221225), LE11 contains the fix since April https://test.libreelec.tv/11.0/RPi/RPi4/

BTW: I strongly recommend to use kernel IR decoders instead of lirc as they are much more reliable - see here https://wiki.libreelec.tv/configuration/ir-remotes

thawn, 2022-12-28:
@HiassofT thanks a lot for the quick reply
I can confirm that `irrecord` works with the latest nightly of LE11 (20221227)

Also, thanks for the link to the wiki. After reading the whole post (I had found the page before, but failed with the predefinde keymaps), I realized that the kernel IR decoders actually work for my remote. In the end, I discarded `irrecord` and used `ir-keytable -t` to create my own keymap as described on the wiki.

closed 2022-12-28 by thawn