LibreELEC/LibreELEC.tv #7305: [BUG] RPi - kodi crashes if xbox360 (compatible) USB controller is connected.

misanthropos, 2022-12-28:
### 
Kodi will crash while (last entry log lines) 
```
2022-12-28 12:34:06.412 T:2358     info <general>: AddOnLog: peripheral.joystick: Initialized joystick 0: "Microsoft X-Box 360 p
ad", axes: 8, hats: 0, buttons: 11
2022-12-28 12:34:06.412 T:2358     info <general>: Loading controller layout: /usr/share/kodi/addons/game.controller.default/res
ources/layout.xml
```

### To Reproduce
Steps to reproduce the behavior:
1. Have a the game controller connected at boot or connect one while running Kodi.
2. Instant crash

### Informations
 - LE Version: 3d590bfba47c974677a49486d0eba802f9a7f072 and prior
 - Hardware Platform: RPi 4B
 - Game controller (USB): Gamesir-T4w 1.39 (Bus 001 Device 004: ID 045e:028e Microsoft Corp. Xbox360 Controller)

### Log file
http://ix.io/4jLO

### Additional context
Maybe this should be filed to the Kodi devolpers?

HiassofT, 2022-12-29:
Remove peripheral.joystick in /storage/.kodi/addons/ - it seems that addon was rolled out via our addon repo by mistake end of August and this is now causing issues (peripheral.joystick is always included in the LE image)
```
rm -rf /storage/.kodi/addons/peripheral.joystick/
```

misanthropos, 2022-12-29:
That was the problem ineed! Thank you for solving that issue.

closed 2022-12-29 by misanthropos