LibreELEC/LibreELEC.tv #7323: Nvidia driver not loading on latest nightly

RenEH01, 2022-12-31:
Libreelec hangs on Libreelec logo not loading the graphics driver on 12/30/2022 nightly and 12/29/2022 nightly. 

Share folders are still present and OS appears to be running fine otherwise. 

Graphics card is gt1030.

SupervisedThinking, 2022-12-31:
Which nightly did you use? Generic or Generic-Legacy?

RenEH01, 2022-12-31:
Generic, is this card now considered legacy?

SupervisedThinking, 2022-12-31:
You have to use those images for NVIDIA GPUS https://test.libreelec.tv/11.0/Generic/Generic-legacy/ since they use X11 instead of GBM

RenEH01, 2022-12-31:
Doesn't the newer Nvidia drivers support GBM as of 495.44?

SupervisedThinking, 2023-01-01:
It does support GBM+Wayland but it doesn't work without a displayserver, NVIDIA users have to stick to Generic-Legacy for now.

RenEH01, 2023-01-01:
> It does support GBM+Wayland but it doesn't work without a displayserver, NVIDIA users have to stick to Generic-Legacy for now.

Thanks for the info!

closed 2023-01-02 by heitbaum