LibreELEC/LibreELEC.tv #7324: [BUG] AMD HDR Colourspace

atoulmin, 2023-01-01:
I will upload a log later today, but I thought I would start the thread.

I've noticed that in the current nightlies that if you start a HDR file with an AMD APU / GPU that it will start playing but not in the correct cloudscape.  It will play in SDR709.  You then have to just return to the Kodi menu and go back to Fullscreen and the output will then be correct BT2020.

This does not occur using Intel.  I'm not sure if this is a Kodi issue or a driver issue.

This happens every time  you play a file with HDR.  Playing a SDR file works correctly.

atoulmin, 2023-01-01:
LOG

https://paste.kodi.tv/tikisuwido.kodi

atoulmin, 2023-01-01:
I wonder if this is related?

https://github.com/xbmc/xbmc/pull/21845

atoulmin, 2023-01-08:
Any thoughts?

heitbaum, 2023-09-04:
Believe that this is fixed. If not please reopen with updated info.

closed 2023-09-04 by heitbaum