LibreELEC/LibreELEC.tv #7327: 10.0.2 nvidia hevc sw

liujixu1984, 2023-01-01:
My version is 10.0.2, and the video card is nvidia 1080. Why can't I hard understand the Hevc code? Screen display sw
[image]
[image]

liujixu1984, 2023-01-01:
I have tried version 10.0.3 and several versions of nightly build, but they cannot be started

[image]

atoulmin, 2023-01-01:
I don’t think NVIDIA is supported on generic. You have to use generic legacy. Give that a go.

liujixu1984, 2023-01-01:
> I don’t think NVIDIA is supported on generic. You have to use generic legacy. Give that a go.

I have used 10.0.2 and 9. x versions, and the results are the same. Can you provide an address for me to download and test!

atoulmin, 2023-01-01:
Try this build

https://test.libreelec.tv/11.0/Generic/Generic-legacy/LibreELEC-Generic-legacy.x86_64-11.0-nightly-20221231-b9ea43f.img.gz

liujixu1984, 2023-01-01:
> 试试这个版本
> 
> https://test.libreelec.tv/11.0/Generic/Generic-legacy/LibreELEC-Generic-legacy.x86_64-11.0-nightly-20221231-b9ea43f.img.gz

My machine cannot be started in this version. See Figure
[image]

lrusak, 2023-01-01:
Please use the forum for support. This is a bug tracker.

https://forum.libreelec.tv/

closed 2023-01-01 by lrusak