LibreELEC/LibreELEC.tv #7332: [BUG] Busybox - ls command broken

KOPRajs, 2023-01-02:
### Describe the bug

On recent nightly build I cannot list my emulators ROM folder:
`LibreELEC:~ # ls /media/DATA/emulators/
ls: /media/DATA/emulators/Earthworm Jim (USA).md: Value too large for defined data type
ls: /media/DATA/emulators/Aladdin (USA).md: Value too large for defined data type
...`

Some of the files get listed and some throw the above error.

### To Reproduce
Steps to reproduce the behavior:
1. Go to '...'
2. Play '....'
3. See error

### Informations
 - LE Version: 11 (nightly 2023-01-01)
 - Hardware Platform: RK3399

### Log file

### Additional context

chewitt, 2023-01-02:
For background https://bugs.busybox.net/show_bug.cgi?id=11651

KOPRajs, 2023-01-02:
I'm not sure if this is a separate issue or if it is related to the busybox issue, but it seems the file listing is broken in Kodi as well. 

This is definitely not a correct ascending file order:
[image]

I've tried to switch between ascending and descending order and the result is different every time (somehow random). I've also tried to view the same folder in Kodi under Windows and it works as expected there so it seems to be a LibreELEC specific problem. It happens in basically every folder I've tried containing ROM files in the same naming scheme.

KOPRajs, 2023-01-03:
I've added more info to the original bug report, see the _Additional context_ section.

I've also done some more testing and I've come to a conclusion that the sorting problem in Kodi is a separate issue:
1. The Busybox issue affects specific inodes and is consistent (the same inode is always affected), while the sorting issue is not consistent and the results are different everytime.
2. I've been able to reproduce the sorting bug on a different system (and on different filesystem - ext4 vs. vfat) which doesn't seem to be affected by the Busybox bug at all. At least the same folder with the broken sorting in Kodi can be listed in terminal (`ls`) without issues and is sorted as expected.

I've reported the sorting issue to Kodi:
https://github.com/xbmc/xbmc/issues/22373

bambury, 2023-08-02:
Subscribing, I'm having this issue, too. Still present on 11.0.3.

`RPi4.arm-11.0.3`
`BusyBox v1.32.1 (2023-07-15 14:20:54 UTC) multi-call binary.`

heitbaum, 2023-09-04:
@bambury - please confirm if still an issue with LE12 nightlies (not expected as the user environment is now 64bit)
@KOPRajs - please confirm if still an issue with LE12 nightlies. (User is still 32bit, but much of the system has been updated - as most libraries are now current and are expected not to have major changes before LE12 - be good to ensure this bug is fixed.)

heitbaum, 2023-09-04:
> @heitbaum I tried installing the latest nightly, but I got an error because of different archs (RPi4.arm vs. RPi4.aarch64). In light of what you wrote, is it safe to ignore and override?

Yes it will work.
Caveat - Your addons will be broken until they are updated though.
Rollback to LE11 is not easy.

Suggest using a new SD card and test first before taking the plunge. LE12 is still in the development stages.

heitbaum, 2023-10-16:
Closing as “fixed” in LE12 with the package updates and 64bit userland. Please reopen or create a new issue - if this still is an issue with LE12 nightlies.

closed 2023-10-16 by heitbaum