LibreELEC/LibreELEC.tv #7353: Raspberry Pi's Incapable of WPA3 Attempt to Connect Via WPA3 and Fail if Network in Mixed Mode.

Steve-Cooley, 2023-01-07:
### Describe the bug
Here are some links that are related: 

https://github.com/raspberrypi/linux/issues/3619
https://forums.raspberrypi.com/viewtopic.php?t=318958

And a link to an issue that sounds similar, but I think is unrelated:
https://github.com/LibreELEC/LibreELEC.tv/issues/7166

Raspberry Pi 3 (and possibly other RPIs) attempt to use WPA3, even though WPA3 is not supported by these devices (I'm not sure if this lack of support is due to hardware or software).  This can be mitigated by either disabling PFM(MFP?) in the AP, or by completely disabling WPA3 (which also disables PFM). The problems with this mitigation are:
1. If PFM is disabled, but WPA3 is enabled, then this allows the raspberry pi running Libreelec to connect, however some other WPA3 capable devices will not be able to connect to the network. (I think that technically WPA3 requires PFM, even though some routers allow PFM to be disabled while using WPA3)
2. If WPA3 is completely disabled, then everything works, but WPA2 is insecure and the network is vulnerable to attack.

I've read elsewhere that there are significant changes to the network stack in Libreelec 11, so this might not affect that version. I have not tried Libreelec 11.

I had hoped that this bug would have been fixed in LE10.0.3, but it apparently has not, which is why I'm speaking up now.

### To Reproduce
Steps to reproduce the behavior (in 10.0.2 and 10.0.3)
1. Run a wireless network in mixed WPA2/WPA3 mode with Management Frame Protection enabled.
3. Attempt to connect to network with correct credentials
4. Connection attempt fails with "Invalid Key".   (This is where the error occurs)
5. Disable WPA3 and Management Frame Protection completely in network
6. Attempt to connect to network with correct credentials
7. Connection completes successfully (This is how home network admin can "fix" the problem by compromising the security of their network)

### Information
 - LE Version: This affects both 10.0.3 and 10.0.2. I think that older versions worked fine (It's been a long time)
 - Rpi3 (and possibly other rpi versions)

HiassofT, 2023-01-07:
Yes, this is well known - not much we can do about  that, it needs to be fixed by RPi folks (wifi firmware and probably kernel changes)

closed 2023-01-07 by HiassofT

Steve-Cooley, 2023-01-07:
Is there a possible software fix that will prevent attempts at WPA3 connections on these devices?

HiassofT, 2023-01-07:
sorry, no idea. Please post your question on the RPi and LE forums

Steve-Cooley, 2023-01-08:
I don't mean to pester you. :)

...But I think there's a way that LE devs could fix this, and not have to wait for changes from upstream.  That's what I meant by "software fix".  Some simple logic like this:

```
If raspberry pi:
  Disable WPA3
```
Done.

This would allow the device to join a network that has WPA3/2 mixed mode.   I could be wrong about this, but I think this would work.  I don't know where this logic would need to be run or anything, if it would be trivial, or a major pain.  Maybe if a dev doesn't want to do it, but doesn't mind pointing me in the right direction... I can take a whack at it. No promises though, as I've never collaborated on any project.  

If you just happen to know that everything I'm saying is wrong... please let me know that too, so I can just drop the issue.

Thanks.

HiassofT, 2023-01-08:
PRs with solutions are always welcome, then we can have a look if the solution is feasible to be included in LE.

BTW: I'm not running WPA3 here so I can't help with development