LibreELEC/LibreELEC.tv #7366: [BUG] snapclient addon breaks kodi audio [RPi4]

Uatschitchun, 2023-01-10:
### Describe the bug
Did an upgrade on my RPi4 from LE 9 to 10. 
Prior to this I had snapclient running, including dual audio (snapclient & kodi simultaneously). 

With LE 10 kodi has no audio device (greyed out) when installing and enabling snapclient addon. 
After rebooting, snapclient works fine, but kodi has no audio. It's not to be possible to get it going like with LE 9 on same RPi4. Logs show 'device busy'. Disabling snapclient addon, kodi has sound again after reboot. Starting snapclient from terminal then also gives 'device busy'. 

So both seems to be blocking each other. 

The 'bin/snapclient.start' even isn't working, as there's no '/proc/asound/ALSA'. 

The way it is now, snapclient addon isn't usable for RPI4 with LE 10. I also checked LE 11 with same results. 

I tried lots and lots of things and found a _hacky_ solution, I'll post in a reply. 

### To Reproduce
Steps to reproduce the behavior:
1. Fresh LE install 
2. Install snapclient addon 
3. Reboot 
4. No UI sounds and system-audio is greyed out, whereas snapclient is working as it should 

### Informations
 - LE Version: 10.0.3 & 11 nightly from today 
 - Hardware Platform: RPI4 

### Log file
I'll provide logs, if needed, but it essentially is a blocking issue. Snapclient blocks kodi's audio and vice versa

Uatschitchun, 2023-01-10:
### `Hacky` Solution

* Install snapclient addon, but do not enable it or disable it when already installed
* Set `dtparam=audio=on` in **`/flash/config.txt`**. Remount RW before: `mount -o remount,rw /flash`
* Add this as **`.config/autostart.sh`**: (Borrowed and adapted from: https://github.com/LibreELEC/LibreELEC.tv/issues/6309#issuecomment-1071239808)
```
#!/bin/sh
pactl load-module module-null-sink sink_name="just_a_sink_1"
pactl suspend-sink just_a_sink_1
pactl load-module module-udev-detect
pactl set-default-sink alsa_output.0.hdmi-stereo
```
* Add **`.config/system.d/snapclient.service`**:
```
[Unit]
Description=Snapclient
After=kodi.service network-online.target sound.target
Requires=kodi.service network-online.target sound.target

[Service]
ExecStart=/storage/.kodi/addons/service.snapclient/bin/snapclient --player pulse
Restart=always

[Install]
WantedBy=kodi.target
```
_You may add needed options for snapclient after `--player pulse`, if needed_

### That's all. ;)

After that I've got:

**`# aplay -l`**
```
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
```
**`# aplay -L`**
```
null
    Discard all samples (playback) or generate zero samples (capture)
default
    Default Audio Device
sysdefault
    Default Audio Device
default:CARD=vc4hdmi0
    vc4-hdmi-0, MAI PCM i2s-hifi-0
    Default Audio Device
sysdefault:CARD=vc4hdmi0
    vc4-hdmi-0, MAI PCM i2s-hifi-0
    Default Audio Device
hdmi:CARD=vc4hdmi0,DEV=0
    vc4-hdmi-0, MAI PCM i2s-hifi-0
    HDMI Audio Output
default:CARD=vc4hdmi1
    vc4-hdmi-1, MAI PCM i2s-hifi-0
    Default Audio Device
sysdefault:CARD=vc4hdmi1
    vc4-hdmi-1, MAI PCM i2s-hifi-0
    Default Audio Device
hdmi:CARD=vc4hdmi1,DEV=0
    vc4-hdmi-1, MAI PCM i2s-hifi-0
    HDMI Audio Output
default:CARD=Headphones
    bcm2835 Headphones, bcm2835 Headphones
    Default Audio Device
sysdefault:CARD=Headphones
    bcm2835 Headphones, bcm2835 Headphones
    Default Audio Device
```
**`# pactl list sinks short`**
```
1       just_a_sink_1   module-null-sink.c      s16le 2ch 44100Hz       SUSPENDED
2       alsa_output.2.stereo-fallback   module-alsa-card.c      s16le 2ch 44100Hz       SUSPENDED
3       alsa_output.0.hdmi-stereo       module-alsa-card.c      s16le 2ch 44100Hz       RUNNING
```
_This is snapclient playing_

**`# pactl list modules short`**
```
0       module-native-protocol-unix
1       module-stream-restore
2       module-device-restore
3       module-card-restore
4       module-default-device-restore
5       module-always-sink
7       module-suspend-on-idle
8       module-dbus-protocol
9       module-bluetooth-policy
10      module-bluetooth-discover
11      module-bluez5-discover
12      module-zeroconf-publish
13      module-switch-on-connect
14      module-null-sink        sink_name=just_a_sink_1
15      module-udev-detect
16      module-alsa-card        device_id="2" name="2" card_name="alsa_card.2" namereg_fail=false tsched=yes fixed_latency_range=no ignore_dB=no deferred_volume=yes use_ucm=yes avoid_resampling=no card_properties="module-udev-detect.discovered=1"
17      module-alsa-card        device_id="0" name="0" card_name="alsa_card.0" namereg_fail=false tsched=yes fixed_latency_range=no ignore_dB=no deferred_volume=yes use_ucm=yes avoid_resampling=no card_properties="module-udev-detect.discovered=1"
```

Maci74, 2023-03-31:
hi friend, i followed your instructions step by step (i'm a newbie) but i must not have understood something, can you help me please?

Giving the commands you said I get this output

# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: PCH [HDA Intel PCH], device 3: HDMI 0 [SAMSUNG]
   Subdevices: 0/1
   Subdevice #0: Subdevice #0
card 0: PCH [HDA Intel PCH], device 7: HDMI 1 [HDMI 1]
   Subdevices: 1/1
   Subdevice #0: Subdevice #0
card 0: PCH [HDA Intel PCH], device 8: HDMI 2 [HDMI 2]
   Subdevices: 1/1
   Subdevice #0: Subdevice #0
card 0: PCH [HDA Intel PCH], device 9: HDMI 3 [HDMI 3]
   Subdevices: 1/1
   Subdevice #0: Subdevice #0
card 1: Audio [KT USB Audio], device 0: USB Audio [USB Audio]
   Subdevices: 1/1
   Subdevice #0: Subdevice #0

# aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
hdmi:CARD=PCH,DEV=0
    HDA Intel PCH, SAMSUNG
    HDMI Audio Output
hdmi:CARD=PCH,DEV=1
    HDA Intel PCH, HDMI 1
    HDMI Audio Output
hdmi:CARD=PCH,DEV=2
    HDA Intel PCH, HDMI 2
    HDMI Audio Output
hdmi:CARD=PCH,DEV=3
    HDA Intel PCH, HDMI 3
    HDMI Audio Output
default:CARD=Audio
    KT USB Audio, USB Audio
    Default Audio Device
sysdefault:CARD=Audio
    KT USB Audio, USB Audio
    Default Audio Device
front:CARD=Audio,DEV=0
    KT USB Audio, USB Audio
    Front output / input...

# pactl list sinks short
1	just_a_sink_1	module-null-sink.c	s16le 2ch 44100Hz	SUSPENDED
2	alsa_output.usb-KTMicro_KT_USB_Audio_2020-02-20-0000-0000-0000--00.analog-stereo	module-alsa-card.c	s16le 2ch 44100Hz	SUSPENDED
3	alsa_output.pci-0000_00_1f.3.hdmi-stereo	module-alsa-card.c	s16le 2ch 44100Hz	RUNNING
4	Snapserver	module-pipe-sink.c	s16le 2ch 48000Hz	SUSPENDED
5	librespot	module-null-sink.c	s16le 2ch 44100Hz	RUNNING

thank you very much and sorry for the english

Uatschitchun, 2023-04-02:
@Maci74 What's the question?

closed 2023-04-02 by lrusak

Uatschitchun, 2023-04-02:
Why closing? Without the workaround, snapclient addon isn't working!

lrusak, 2023-04-03:
> Why closing? Without the workaround, snapclient addon isn't working! 

Dual audio isn't possible on any platform without using pulseaudio. Dual audio on the past was a feature quirk of the raspberry Pi firmware. Since switching to use VC4 this isn't possible anymore.

Uatschitchun, 2023-04-03:
Ok, understand that closing it is the right way to do, if it won't/can't get fixed.
But would it be possible to add this workaround to the wiki or somewhere else to be found by users wanting to use snapclient?