LibreELEC/LibreELEC.tv #7368: [BUG] Missing info on dropped/skipped frames

cunlem, 2023-01-11:
### Describe the bug
`drop` and `skip` frame counters in [Player Debug Info](https://kodi.wiki/view/Player_process_info#Player_Debug_Info) are not working

### To Reproduce
Steps to reproduce the behavior:
1. Play a video
2. Open Player Debug Info (`Ctrl`+`Shift`+`O`)
3.  Observe the counters for dropped and skipped frames. They will stay at 0 no matter how much device is struggling with video playback

### Informations
 - LE Version: 10.0.2
 - Hardware Platform: RPi4

### Additional context
My guess is it's something that used to work in older non-KMS LE versions

chewitt, 2023-01-12:
I'm watching an interlaced H264 TVHeadend stream where the skip counter is climbing; the source is 3,500Km away so there are always some playback errors although most aren't visible. I don't see errors with local media files, but then I self-rip so media quality is good, and there has been a lot of human investment from RPi devs in hardware decoding over time so more media plays fine without errors. I'd start with updating to a current image; current LE10 nightlies will be the same as 10.0.4 which will ship soon.

cunlem, 2023-01-13:
Hello, thanks for the reply. Perhaps it behaves differently when the reason for skipping frames is network condition versus decoding bottleneck? In order to experience what I described you need to give Pi a high resolution, high bitrate content in format not hardware accelerated by it. Here's an short excerpt from VC-1 encoded _Vertigo_ Blu-ray: [VC-1 1080p24 @ 30 Mb/s](https://drive.google.com/file/d/1gF2LQN_SOFjT0r0Ic1UH1YbBS5aLSAud/view). At stock Pi clocks (1500 MHz) it looks like slideshow yet there is no indication of frame skip in the counter. I was able to make it watchable with 2100 MHz CPU overclock using my eyes to judge. I wanted to use the counter to verify the effect but it looks like it's broken for this case.

heitbaum, 2023-09-04:
Please retest with current LE11/LE12. If still an issue - please reopen issue or create a new issue referencing this one.

closed 2023-09-04 by heitbaum