LibreELEC/LibreELEC.tv #7384: Unable to build

klattimer, 2023-01-15:
I hit a dead stop at build stage 18 with this. 

```
Trying to build on ubuntu 20.04 LTS and at stage 18 of the build I get the following error.

CMake Error at cmake/StdFilesystem.cmake:22 (message):
  Toolchain doesn't support std::filesystem with nor without -lstdc++fs
Call Stack (most recent call first):
  cmake/StandardSettings.cmake:57 (include)
  CMakeLists.txt:70 (include)
```

heitbaum, 2023-01-15:
> I hit a dead stop at build stage 18 with this.
>
> ```
> Trying to build on ubuntu 20.04 LTS and at stage 18 of the build I get the following error.
>
> CMake Error at cmake/StdFilesystem.cmake:22 [...]
@klattimer - there not enough information to diagnose - can you:

Please share the command you ran to do the build.

Please share the full log for 18
- `LibreELEC.tv$ more build.LibreELEC-Generic.x86_64-11.0-devel/.threads/logs/`

klattimer, 2023-01-15:
Sure. `PROJECT=RPi ARCH=arm DEVICE=RPi4 make image`

klattimer, 2023-01-15:
Sorry 14 - it was 18 on LibreELEC-rr 

```
    TOOLCHAIN      cmake-make
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 7.5.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/karl/Projects/LibreELEC.tv/build.LibreELEC-RPi4.arm-11.0-devel/build/ccache-4.7.4/.x86_64-linux-gnu/libreelec-local-cc
-- The ASM_MASM compiler identification is unknown
-- Found assembler: ml
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/karl/Projects/LibreELEC.tv/build.LibreELEC-RPi4.arm-11.0-devel/build/ccache-4.7.4/.x86_64-linux-gnu/libreelec-local-cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/karl/Projects/LibreELEC.tv/build.LibreELEC-RPi4.arm-11.0-devel/build/ccache-4.7.4/.x86_64-linux-gnu/libreelec-local-cxx - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Ccache version: 4.7.4
-- Ccache dev mode: OFF
-- Performing Test std_atomic_without_libatomic
-- Performing Test std_atomic_without_libatomic - Success
-- Performing Test std_filesystem_without_libfs
-- Performing Test std_filesystem_without_libfs - Failed
-- Performing Test std_filesystem_with_libfs
-- Performing Test std_filesystem_with_libfs - Failed
CMake Error at cmake/StdFilesystem.cmake:22 (message):
  Toolchain doesn't support std::filesystem with nor without -lstdc++fs
Call Stack (most recent call first):
  cmake/StandardSettings.cmake:57 (include)
  CMakeLists.txt:70 (include)

-- Configuring incomplete, errors occurred!
See also "/home/karl/Projects/LibreELEC.tv/build.LibreELEC-RPi4.arm-11.0-devel/build/ccache-4.7.4/.x86_64-linux-gnu/CMakeFiles/CMakeOutput.log".
See also "/home/karl/Projects/LibreELEC.tv/build.LibreELEC-RPi4.arm-11.0-devel/build/ccache-4.7.4/.x86_64-linux-gnu/CMakeFiles/CMakeError.log".
FAILURE: scripts/build ccache:host during configure_host (package.mk)
*********** FAILED COMMAND ***********
cmake -DCMAKE_TOOLCHAIN_FILE=cmake-ccache.conf -DCMAKE_INSTALL_PREFIX=${TOOLCHAIN} -DENABLE_DOCUMENTATION=OFF -DREDIS_STORAGE_BACKEND=OFF -DZSTD_FROM_INTERNET=OFF -DENABLE_TESTING=OFF ..
**************************************
FAILURE: scripts/build ccache:host has failed!
```

heitbaum, 2023-01-15:
> Sorry 14 - it was 18 on LibreELEC-rr
>
> ```
>     TOOLCHAIN      cmake-make
> -- The C compiler identification is GNU 8.4.0
> -- The CXX compiler identification is GNU 7.5.0
> -- The ASM compiler [...]

looks like a local ccache host build issue. 

please try by setting `LOCAL_CCACHE=""`

`LOCAL_CCACHE="" PROJECT=RPi ARCH=arm DEVICE=RPi4 make image`

let us know

klattimer, 2023-01-15:
Looks the same, do I need to make clean?

```
BUILD      ccache (host)
    TOOLCHAIN      cmake-make
-- Ccache version: 4.7.4
-- Ccache dev mode: OFF
CMake Error at cmake/StdFilesystem.cmake:22 (message):
  Toolchain doesn't support std::filesystem with nor without -lstdc++fs
Call Stack (most recent call first):
  cmake/StandardSettings.cmake:57 (include)
  CMakeLists.txt:70 (include)

-- Configuring incomplete, errors occurred!
See also "/home/karl/Projects/LibreELEC.tv/build.LibreELEC-RPi4.arm-11.0-devel/build/ccache-4.7.4/.x86_64-linux-gnu/CMakeFiles/CMakeOutput.log".
See also "/home/karl/Projects/LibreELEC.tv/build.LibreELEC-RPi4.arm-11.0-devel/build/ccache-4.7.4/.x86_64-linux-gnu/CMakeFiles/CMakeError.log".
FAILURE: scripts/build ccache:host during configure_host (package.mk)
*********** FAILED COMMAND ***********
cmake -DCMAKE_TOOLCHAIN_FILE=cmake-ccache.conf -DCMAKE_INSTALL_PREFIX=${TOOLCHAIN} -DENABLE_DOCUMENTATION=OFF -DREDIS_STORAGE_BACKEND=OFF -DZSTD_FROM_INTERNET=OFF -DENABLE_TESTING=OFF ..
**************************************
FAILURE: scripts/build ccache:host has failed!

The following log for this failure is available:
  /home/karl/Projects/LibreELEC.tv/build.LibreELEC-RPi4.arm-11.0-devel/.threads/logs/14.log

>>> ccache:host seq 14 >>>
[014/279] [FAIL] build   ccache:host

The following log for this failure is available:
  /home/karl/Projects/LibreELEC.tv/build.LibreELEC-RPi4.arm-11.0-devel/.threads/logs/14.log

Parallel build failure - see log for details. Time of failure: Sun Jan 15 10:04:03 UTC 2023
Makefile:10: recipe for target 'image' failed
make: *** [image] Error 1
```

heitbaum, 2023-01-15:
I didn’t think you had to

PROJECT=RPi ARCH=arm DEVICE=RPi4 scripts/clean ccache
LOCAL_CCACHE="" PROJECT=RPi ARCH=arm DEVICE=RPi4 make image

Or then

LOCAL_CCACHE="" PROJECT=RPi ARCH=arm DEVICE=RPi4 make clean
LOCAL_CCACHE="" PROJECT=RPi ARCH=arm DEVICE=RPi4 make image

and if that doesn’t work 

try `sudo apt remove ccache`
PROJECT=RPi ARCH=arm DEVICE=RPi4 make image

mglae, 2023-01-15:
```
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 7.5.0
```
Don't mix compiler versions. Furthermore ccache requires GCC >= 8.

klattimer, 2023-01-15:
I had my alternatives set wrong! Thank you!

closed 2023-01-15 by klattimer