LibreELEC/LibreELEC.tv #7437: [BUG] pvr.vdr.vnsi needs a “Generic-legacy” version

heitbaum, 2023-01-28:
### Describe the bug

https://forum.libreelec.tv/thread/26546-le-nightly-legacy-missing-libglesv2-so-2/

### Informations
 - LE Version: LE11
 - Hardware Platform: Generic-legacy

### Log file

### Additional context

heitbaum, 2023-01-28:
Tried a few things - latest as below. 
The build of the addon in GHA Generic-legacy works fine. Issue is the the addon itself “self identifies” as `pvr.vdr.vnsi` - to allow a second build of `pvr.vdr.vnsi` as `legacy-pvr.vdr.vnsi` would require code changes ….

```
$ git diff
diff --git a/config/functions b/config/functions
index 365d5639ea..49f837e26c 100644
--- a/config/functions
+++ b/config/functions
@@ -1269,7 +1269,10 @@ source_package() {
     PKG_LONGDESC="${PKG_LONGDESC:-${PKG_NAME} (autogenerated)}"
 
     if [ "$PKG_IS_ADDON" = "yes" -o "$PKG_IS_ADDON" = "embedded" ] ; then
-      [ -z $PKG_SECTION ] && PKG_ADDON_ID="$PKG_NAME" || PKG_ADDON_ID="${PKG_SECTION//\//.}.$PKG_NAME"
+      if [ -z $PKG_ADDON_ID ]; then
+        # Allow for PKG_ADDON_ID tombe overridden to handle Genetic-legacy packages
+        [ -z $PKG_SECTION ] && PKG_ADDON_ID="$PKG_NAME" || PKG_ADDON_ID="${PKG_SECTION//\//.}.$PKG_NAME"
+      fi
       [ "$PKG_ADDON_IS_STANDALONE" != "yes" ] && PKG_NEED_UNPACK="${PKG_NEED_UNPACK} $(get_pkg_directory $MEDIACENTER)"
     fi
 
diff --git a/packages/mediacenter/kodi-binary-addons/legacy-pvr.vdr.vnsi/package.mk b/packages/mediacenter/kodi-binary-addons/leg
acy-pvr.vdr.vnsi/package.mk
index c3691699f6..df05993175 100644
--- a/packages/mediacenter/kodi-binary-addons/legacy-pvr.vdr.vnsi/package.mk
+++ b/packages/mediacenter/kodi-binary-addons/legacy-pvr.vdr.vnsi/package.mk
@@ -8,6 +8,7 @@ PKG_ARCH="x86_64"
 PKG_LONGDESC="pvr.vdr.vnsi for Generic-legacy"
 PKG_URL=""
 PKG_DEPENDS_UNPACK+=" pvr.vdr.vnsi"
+PKG_ADDON_ID="pvr.vdr.vnsi"
 
 unpack() {
   mkdir -p ${PKG_BUILD}
```

CvH, 2023-01-28:
just to clarify, a complete legacy repo is not gone happen (too much overhead)

> would require code changes

doesn't looks too scary

heitbaum, 2023-02-09:
Generic-legacy repo built.

closed 2023-02-09 by heitbaum