LibreELEC/LibreELEC.tv #7550: [BUG] Generic x86_64 Intel 4650U Graphics issues

leakypixel, 2023-02-27:
### Describe the bug
I'm using an HTPC with an i7 4650U, and get some weird hashing/banding pattern once the system has booted. I see the same issue with the 10 stable build (LibreELEC-Generic.x86_64-10.0.4.img.gz) and both 11 beta builds (LibreELEC-Generic-legacy.x86_64-10.95.1.img.gz and LibreELEC-Generic.x86_64-10.95.1.img.gz).

### To Reproduce
Steps to reproduce the behavior:
1. Install any of the above images of LibreELEC on an i7-4650U based system
2. Boot as default 
3. See visual issues (see image below)

Image showing issue: https://forum.libreelec.tv/core/attachment/9183-signal-2023-02-24-143832-jpeg/

### Informations
 - LE Version: 10.0.4, 10.95.1 legacy, 10.95.1
 - Hardware Platform: x86_64

### Additional context
In order to debug, I tried a clean install of Ubuntu 22.04 and saw the same issue despite drivers being listed correctly (i915) through lspci. After reading the intel documentation at [https://dgpu-docs.intel.com/in…ntu/ubuntu-jammy-arc.html](https://dgpu-docs.intel.com/installation-guides/ubuntu/ubuntu-jammy-arc.html) I followed the instructions. Once all the required software was installed (but NOT the DKMS modules), I still saw the same issue. After installing the DKMS modules however, the issue was resolved.

Thinking then that I had a solution, I returned to the LibreELEC 10 stable build and used the docker addon via ssh to build the firmware files for the running kernel (which completely successfully) using the instructions here: [https://github.com/intel/media…er/doc/intel-gpu-dkms.rst](https://github.com/intel/media-delivery/blob/master/doc/intel-gpu-dkms.rst)

I then copied the generated firmware files to the /storage/.config/firmware directory and rebooted, expecting that the appropriate firmware would now be used and my graphics would be functional again - unfortunately, this was not the case.

So, my question is: what am I missing here? I don't mind getting my hands dirty and doing a custom build if required, but it feels like there's a way to do this that I'm just not getting.

Apologies if this is an obvious question - I've tried reading the LibreELEC docs to find a solution, but perhaps due to my inexperience with this type of system I can't find anything helpful.

Any help would be greatly appreciated!

chewitt, 2023-02-27:
What happens if you add `pci=realloc=off` to kernel boot params?

heitbaum, 2023-02-27:
Hi @leakypixel

- when you performed the fix to this on Ubuntu 22.04. What driver is loaded. Please shared your dmesg and vainfo.
- what files did you place in /storage/.config/firmware?
- logs from LE would help confirm i915/i965

I don’t have / haven’t tried a gen4 cpu; some hints
- https://wiki.libreelec.tv/hardware/intel-x86-64-generic
- https://forum.libreelec.tv/thread/26149-how-to-overide-mesa-driver-uhd-i965-le-10-02/

leakypixel, 2023-02-27:
Adding `pci=realloc=off`  to `/flash/syslinux.cfg` seems not to make a difference - the logs in both cases appear identical.

Files placed in /storage/.config/firmware were generated firmware bin files, such as `dg2_guc_70.2.2.bin`. If it's helpful, I can post the full (532 item) list.

These are the `journalctl -u kodi` logs from the standard 10.0.4 image (with the kernel param, but the same without):

```Feb 27 11:03:11 LibreELEC systemd[1]: Starting Kodi Media Center...
Feb 27 11:03:11 LibreELEC systemd[1]: Started Kodi Media Center.
Feb 27 11:03:12 LibreELEC kodi.sh[1076]: libva info: VA-API version 1.11.0
Feb 27 11:03:12 LibreELEC kodi.sh[1076]: libva info: Trying to open /usr/lib/dri/iHD_drv_video.so
Feb 27 11:03:12 LibreELEC kodi.sh[1076]: libva info: Found init function __vaDriverInit_1_11
Feb 27 11:03:12 LibreELEC kodi.sh[1076]: libva error: /usr/lib/dri/iHD_drv_video.so init failed
Feb 27 11:03:12 LibreELEC kodi.sh[1076]: libva info: va_openDriver() returns 1
Feb 27 11:03:12 LibreELEC kodi.sh[1076]: libva info: Trying to open /usr/lib/dri/i965_drv_video.so
Feb 27 11:03:12 LibreELEC kodi.sh[1076]: libva info: Found init function __vaDriverInit_1_11
Feb 27 11:03:12 LibreELEC kodi.sh[1076]: libva info: va_openDriver() returns 0
```

`vainfo`  from 10.95.1 legacy and 10.0.4:

```
LibreELEC (official): 10.95.1 (Generic-legacy.x86_64)
LibreELEC:~ # vainfo
Trying display: x11
libva info: VA-API version 1.17.0
libva info: Trying to open /usr/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_17
libva error: /usr/lib/dri/iHD_drv_video.so init failed
libva info: va_openDriver() returns 1
vaInitialize failed with error code 1 (operation failed),exit
```

```
LibreELEC (official): 10.0.4 (Generic.x86_64)
LibreELEC:~ # vainfo
libva info: VA-API version 1.11.0
libva info: Trying to open /usr/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_11
libva error: /usr/lib/dri/iHD_drv_video.so init failed
libva info: va_openDriver() returns 1
libva info: Trying to open /usr/lib/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_1_11
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.11 (libva 2.11.1)
vainfo: Driver version: Intel i965 driver for Intel(R) Haswell Mobile - 2.4.1
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264MultiviewHigh      :	VAEntrypointVLD
      VAProfileH264MultiviewHigh      :	VAEntrypointEncSlice
      VAProfileH264StereoHigh         :	VAEntrypointVLD
      VAProfileH264StereoHigh         :	VAEntrypointEncSlice
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileJPEGBaseline           :	VAEntrypointVLD
```

Appears to me that the i965 driver is being used as a fallback when iHD is failing to load - should I try forcing this to i915 somehow (and if so, how)?

leakypixel, 2023-02-27:
@heitbaum I'll have to reinstall 22.04 to get the full logs, which I'll have to do this evening and get back to you - but I did confirm at the time that it was using the i915 rather than i965 driver.

heitbaum, 2023-02-27:
The HuC and GuC drivers are not available for GPUs before Skylake. There would be a dmesg showing HuC/GuC in the dmesg (if using i915.enable_guc=3 in the APPEND options of syslinux.cfg)

for my nuc12 dmesg shows i915…. I will have to check the output from my nuc6 - but it is skylake anyhow - but not online at the moment.

```
nuc12:~ # lspci | grep VGA
00:02.0 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)

nuc12:~ # dmesg | grep i915
[    0.000000] Command line: BOOT_IMAGE=/KERNEL boot=UUID=779D-1827 disk=LABEL=Storage i915.enable_guc=3 ssh mitigations=off
[    0.065540] Kernel command line: root=/dev/ram0 rdinit=/init usbcore.autosuspend=-1 BOOT_IMAGE=/KERNEL boot=UUID=779D-1827 disk=LABEL=Storage i915.enable_guc=3 ssh mitigations=off
[    0.065567] Setting dangerous option i915.enable_guc - tainting kernel
[    1.102823] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    1.102826] i915 0000:00:02.0: vgaarb: deactivate vga console
[    1.102860] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    1.103247] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    1.104453] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adlp_dmc_ver2_16.bin (v2.16)
[    1.196744] i915 0000:00:02.0: [drm] GuC firmware i915/adlp_guc_70.bin version 70.5.1
[    1.196747] i915 0000:00:02.0: [drm] HuC firmware i915/tgl_huc.bin version 7.9.3
[    1.210420] i915 0000:00:02.0: [drm] HuC authenticated
[    1.211019] i915 0000:00:02.0: [drm] GuC submission enabled
[    1.211020] i915 0000:00:02.0: [drm] GuC SLPC enabled
[    1.211556] i915 0000:00:02.0: [drm] GuC RC: enabled
[    1.212855] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    1.241734] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[    1.276316] fbcon: i915drmfb (fb0) is primary device
[    1.360723] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
```

leakypixel, 2023-02-27:
> The HuC and GuC drivers are not available for GPUs before Skylake. There would be a dmesg showing HuC/GuC in the dmesg (if using i915.enable_guc=3 in the APPEND options of syslinux.cfg)
> 
> for my nuc12 dmesg shows i915…. I will have to check the output from my nuc6 - but it is skylake anyhow - but not online at the moment.
> 
> ```
> nuc12:~ # lspci | grep VGA
> 00:02.0 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
> 
> nuc12:~ # dmesg | grep i915
> [    0.000000] Command line: BOOT_IMAGE=/KERNEL boot=UUID=779D-1827 disk=LABEL=Storage i915.enable_guc=3 ssh mitigations=off
> [    0.065540] Kernel command line: root=/dev/ram0 rdinit=/init usbcore.autosuspend=-1 BOOT_IMAGE=/KERNEL boot=UUID=779D-1827 disk=LABEL=Storage i915.enable_guc=3 ssh mitigations=off
> [    0.065567] Setting dangerous option i915.enable_guc - tainting kernel
> [    1.102823] i915 0000:00:02.0: [drm] VT-d active for gfx access
> [    1.102826] i915 0000:00:02.0: vgaarb: deactivate vga console
> [    1.102860] i915 0000:00:02.0: [drm] Using Transparent Hugepages
> [    1.103247] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
> [    1.104453] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adlp_dmc_ver2_16.bin (v2.16)
> [    1.196744] i915 0000:00:02.0: [drm] GuC firmware i915/adlp_guc_70.bin version 70.5.1
> [    1.196747] i915 0000:00:02.0: [drm] HuC firmware i915/tgl_huc.bin version 7.9.3
> [    1.210420] i915 0000:00:02.0: [drm] HuC authenticated
> [    1.211019] i915 0000:00:02.0: [drm] GuC submission enabled
> [    1.211020] i915 0000:00:02.0: [drm] GuC SLPC enabled
> [    1.211556] i915 0000:00:02.0: [drm] GuC RC: enabled
> [    1.212855] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
> [    1.241734] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
> [    1.276316] fbcon: i915drmfb (fb0) is primary device
> [    1.360723] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
> 
> nuc12:~ # lspci -vvvv -s 00:02.0  | grep Kern
>         Kernel driver in use: i915
> ```

Oops, my bad - I'll do the 22.04 install again and check, I was sure it showed i915 but you're obviously correct so I can't see how that would have been the case.

eldersixpack13, 2023-02-27:
> The HuC and GuC drivers are not available for GPUs before Skylake. There would be a dmesg showing HuC/GuC in the dmesg (if using i915.enable_guc=3 in the APPEND options of syslinux.cfg)
>
> for my [...]

Don't know if it is correct to add my observations here (Running yesterdays nightly on an NUC_8i3BEH, Coffee Lake)

for i915.enable_guc=2  
I get with "dmesg|grep -iE 'drm|guc|huc|vgem' ":

[    0.000000] Command line: BOOT_IMAGE=/KERNEL boot=UUID=0C49-2862 disk=UUID=25ded9ac-f3b7-4349-a3af-a2c49961ed13 i915.enable_guc=2 quiet
[    0.025579] Kernel command line: root=/dev/ram0 rdinit=/init usbcore.autosuspend=-1 BOOT_IMAGE=/KERNEL boot=UUID=0C49-2862 disk=UUID=25ded9ac-f3b7-4349-a3af-a2c49961ed13 i915.enable_guc=2 quiet
[    0.025625] Setting dangerous option i915.enable_guc - tainting kernel
[    0.488285] ACPI: bus type drm_connector registered
[    0.488298] [drm] radeon kernel modesetting enabled.
[    0.488331] [drm] amdgpu kernel modesetting enabled.
[    0.489090] i915 0000:00:02.0: [drm] Found 128MB of eDRAM
[    0.491202] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/kbl_dmc_ver1_04.bin (v1.4)
[    0.762421] i915 0000:00:02.0: [drm] failed to retrieve link info, disabling eDP
[    0.765218] i915 0000:00:02.0: [drm] [ENCODER:106:DDI C/PHY C] is disabled/in DSI mode with an ungated DDI clock, gate it
[    0.769125] i915 0000:00:02.0: [drm] GuC firmware i915/kbl_guc_70.1.1.bin version 70.1.1
[    0.769130] i915 0000:00:02.0: [drm] HuC firmware i915/kbl_huc_4.0.0.bin version 4.0.0
[    0.791804] i915 0000:00:02.0: [drm] HuC authenticated
[    0.791807] i915 0000:00:02.0: [drm] GuC submission disabled
[    0.791809] i915 0000:00:02.0: [drm] GuC SLPC disabled
[    0.958766] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[    1.046395] fbcon: i915drmfb (fb0) is primary device
[    1.201510] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
[    2.181516] systemd[1]: Starting modprobe@drm.service...
[    2.200741] systemd[1]: modprobe@drm.service: Deactivated successfully.
[    2.200918] systemd[1]: Finished modprobe@drm.service.

and for i915.enable_guc=3

[    0.000000] Command line: BOOT_IMAGE=/KERNEL boot=UUID=0C49-2862 disk=UUID=25ded9ac-f3b7-4349-a3af-a2c49961ed13 i915.enable_guc=3 quiet
[    0.025498] Kernel command line: root=/dev/ram0 rdinit=/init usbcore.autosuspend=-1 BOOT_IMAGE=/KERNEL boot=UUID=0C49-2862 disk=UUID=25ded9ac-f3b7-4349-a3af-a2c49961ed13 i915.enable_guc=3 quiet
[    0.025557] Setting dangerous option i915.enable_guc - tainting kernel
[    0.484737] ACPI: bus type drm_connector registered
[    0.484747] [drm] radeon kernel modesetting enabled.
[    0.484780] [drm] amdgpu kernel modesetting enabled.

**[    0.485178] i915 0000:00:02.0: [drm] Incompatible option enable_guc=3 - GuC submission is N/A**

[    0.485630] i915 0000:00:02.0: [drm] Found 128MB of eDRAM
[    0.487410] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/kbl_dmc_ver1_04.bin (v1.4)
[    0.758759] i915 0000:00:02.0: [drm] failed to retrieve link info, disabling eDP
[    0.765465] i915 0000:00:02.0: [drm] GuC firmware i915/kbl_guc_70.1.1.bin version 70.1.1
[    0.765470] i915 0000:00:02.0: [drm] HuC firmware i915/kbl_huc_4.0.0.bin version 4.0.0
[    0.788753] i915 0000:00:02.0: [drm] HuC authenticated
[    0.788757] i915 0000:00:02.0: [drm] GuC submission disabled
[    0.788759] i915 0000:00:02.0: [drm] GuC SLPC disabled
[    0.966599] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[    1.055708] fbcon: i915drmfb (fb0) is primary device
[    1.243856] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
[    2.025664] systemd[1]: Starting modprobe@drm.service...
[    2.044062] systemd[1]: modprobe@drm.service: Deactivated successfully.
[    2.044224] systemd[1]: Finished modprobe@drm.service.

heitbaum, 2023-02-27:
@eldersixpack13 
- https://wiki.archlinux.org/title/intel_graphics#Enable_GuC_/_HuC_firmware_loading 
- i915.enable_guc=3 is only supported on Alder Lake-P

heitbaum, 2024-03-11:
Is this issue still occurring with latest LE12 Generic/Generic-legacy 20240311-86dba4b images?

heitbaum, 2024-03-16:
Closing as inactive. Happy to open again - but will need to be tested with LE12 and debug logs, ideally a reproducible case. Using the forum for support may get more answers as others may have seen the same issue.

closed 2024-03-16 by heitbaum