LibreELEC/LibreELEC.tv #7557: [BUG] No audio option for Hifiberry AMP 3 selectable

afdev82, 2023-03-01:
### Describe the bug
I have followed the [instructions](https://www.hifiberry.com/docs/software/libreelec-installation-and-configuration/) to add the Hifiberry AMP 3 overlay to get the audio output option in the list to be selected, but I don't see any new option for that.
The overlay was added in the kernel v5.15.60 of the RaspberryPi (I'm using the RaspberryPi 2).

### To Reproduce
Steps to reproduce the behavior:
1. Follow [these instructions](https://www.hifiberry.com/docs/software/libreelec-installation-and-configuration/) and as overlay use `hifiberry-amp3` 
1. Go to Settings/System/Audio
2. In the Audio Output device I don't see anything related to the AMP 3, I see 4 entry for ALSA (vc4-hdmi) and one for PULSE (bluetooth audio)

### Informations
 - LE Version: [e.g. 10.95.1]
 - Hardware Platform: [e.g. RPi2]

Thank you for your support!

afdev82, 2023-03-02:
Sorry for the noise and thank you for your reply.
I've thought this was a bug, since the overlay file it's in the image.
I will ask that in the forum.

HiassofT, 2023-03-02:
Actually this seems to be a bug, LE kernel is missing the codec driver needed for that soundcard.

Can you please test if this build works?
https://www.horus.com/~hias/tmp/libreelec/LibreELEC-RPi2.arm-11.0-devel-20230302101145-ebc99ca.tar

afdev82, 2023-03-02:
> Actually this seems to be a bug, LE kernel is missing the codec driver needed for that soundcard.
> 
> Can you please test if this build works? https://www.horus.com/~hias/tmp/libreelec/LibreELEC-RPi2.arm-11.0-devel-20230302101145-ebc99ca.tar

Yes, for sure.
Usually I just flash an image in the SD card, what should I do with the link provided in this case?

chewitt, 2023-03-02:
Download to `/storage/.update` and reboot to start the update

afdev82, 2023-03-02:
Thank you very much, it worked!

HiassofT, 2023-03-02:
Thanks a lot for reporting back, I'll include the changes in the next RPi kernel update. In the meanwhile just use the image above (other than the kernel/driver change it's identical to what we plan to release as LE 11.0.0 official soon)

PR #7558 (2023-03-05): linux (RPi): update to 6.1.14
This also adds the missing codec driver for the Hifiberry AMP 3

Runtime tested on RPi3B+ and RPi4

Fixes: #7557
closed 2023-03-05 by CvH