LibreELEC/LibreELEC.tv #7589: libfuse3.so.3 missing in LE11

chris42, 2023-03-11:
Just updated on LE11, now my sshfs mounts are gone. Searched the logs and found:
```
Mar 11 09:11:49 mymc systemd[1]: Starting sshfs.nas.service...
Mar 11 09:11:49 mymc sshfs[781]: /storage/.kodi/addons/virtual.network-tools/bin/sshfs: error while loading shared libraries: libfuse3.so.3: cannot open shared object file: No such file or directory
Mar 11 09:11:49 mymc systemd[1]: sshfs.nas.service: Control process exited, code=exited, status=127/n/a
Mar 11 09:11:49 mymc systemd[1]: sshfs.nas.service: Failed with result 'exit-code'.
Mar 11 09:11:49 mymc systemd[1]: Failed to start sshfs.nas.service.
```

I have system-tools and network-tools installed (also reinstalled them already, no change). Would have expected a "libfuse3.so.3" library in /usr/lib, however there is nothing to be found regarding fuse.
The sshfs command I am doing is: 

```
/storage/.kodi/addons/virtual.network-tools/bin/sshfs -o ro,reconnect,ServerAliveInterval=5,ServerAliveCountMax=3,IdentityFile=/storage/.ssh/id_ed25519_nas,port=22 mymc@nas:/ /storage/nas
```

Something might be misconfigured in the addon package or install?

heitbaum, 2023-03-12:
> /storage/.kodi/addons/virtual.network-tools/

Have you restarted since adding the addon? I’m pretty sure the libraries should be there, but are not activated until LibreELEC is restarted.

```
# find /storage/.kodi/addons/virtual.network-tools/ | grep libfuse
/storage/.kodi/addons/virtual.network-tools/lib/libfuse3.so
/storage/.kodi/addons/virtual.network-tools/lib/libfuse.so
/storage/.kodi/addons/virtual.network-tools/lib/libfuse.so.2
/storage/.kodi/addons/virtual.network-tools/lib/libfuse3.so.3.12.0
/storage/.kodi/addons/virtual.network-tools/lib/libfuse.so.2.9.9
/storage/.kodi/addons/virtual.network-tools/lib/libfuse3.so.3
```

chris42, 2023-03-12:
I am pretty sure I rebooted in between, as I saw the error for the nas service on the boot screen. Will have time later today, then uninstall network tools, reboot, install, reboot, report back.

chris42, 2023-03-12:
So tested this quickly, but still getting the error, that the libraries cannot be found. The libraries are on the system, however they are not being used/found by sshfs.

```
# find /storage/.kodi/addons/virtual.network-tools/ | grep libfuse
/storage/.kodi/addons/virtual.network-tools/lib/libfuse3.so
/storage/.kodi/addons/virtual.network-tools/lib/libfuse3.so.3.12.0
/storage/.kodi/addons/virtual.network-tools/lib/libfuse3.so.3
/storage/.kodi/addons/virtual.network-tools/lib/libfuse.so.2.9.9
/storage/.kodi/addons/virtual.network-tools/lib/libfuse.so
/storage/.kodi/addons/virtual.network-tools/lib/libfuse.so.2
```

Maybe sshfs needs additional library paths or linking to /usr/lib is needed?

heitbaum, 2023-03-12:
LibreELEC:~ # sshfs
missing host
see `sshfs -h' for usage
LibreELEC:~ # ldd  /storage/.kodi/addons/virtual.network-tools/bin/sshfs
        /usr/lib/libarmmem-v7l.so (0xf7eb0000)
        libfuse3.so.3 => /storage/.kodi/addons/virtual.network-tools/lib/libfuse3.so.3 (0xf7e86000)
        libglib-2.0.so.0 => /usr/lib/libglib-2.0.so.0 (0xf7d13000)
        libc.so.6 => /usr/lib/libc.so.6 (0xf7ba0000)
        /lib/ld-linux-armhf.so.3 => /usr/lib/ld-linux-armhf.so.3 (0xf7eda000)

chewitt, 2023-03-12:
Make sure you delete the cached packages in `/storage/.kodi/addons/packages` else Kodi will reinstall from the cache and not from the online version (in case that's an issue).

heitbaum, 2023-03-12:
@chris42 - just tested here and I have it “not erroring” - maybe via the systemd script it is… but from the prompt it seems to find the libraries.

mglae, 2023-03-12:
LE10 is including libfuse in /usr/lib of the image because of ntfs-3g. It can ever be found in the default lib path.

LE11 dropped ntfs-3g and libfuse from the image. The library path to the addon has to be added to environment before used.

@chris42 The usual work around is using the profile to update the paths:

```
ExecStart=/bin/sh -c ". /etc/profile;exec sshfs -o ro,reconnect,ServerAliveInterval=5,ServerAliveCountMax=3,IdentityFile=/storage/.ssh/id_ed25519_nas,port=22 mymc@nas:/ /storage/nas"
```

chris42, 2023-03-12:
Yepp, using the profile does work. Thanks for the hint and sorry for the confusion.

closed 2023-03-12 by chris42

reitzig, 2023-07-27:
I've been using a `.mount` file with

```
[Mount]
What=/storage/.kodi/addons/virtual.network-tools/bin/sshfs#user@host:/some/dir
Type=fuse.sshfs
Where=/var/media/host/dir
Options=ro,allow_other,reconnect,ServerAliveInterval=15,IdentityFile=/storage/.ssh/id_rsa
```

This broke after upgrading von LE9 (I know) to LE11 with a similar error:

```
/storage/.kodi/addons/virtual.network-tools/bin/sshfs: error while loading shared libraries: libfuse3.so.3: cannot open shared object file: No such file or directory
```

However, just running

```
/storage/.kodi/addons/virtual.network-tools/bin/sshfs -o ro,allow_other,reconnect,ServerAliveInterval=15,IdentityFile=/storage/.ssh/id_rsa user@host:/some/dir /var/media/host/dir
```

in an SSH shell works.
I _think_ that's the same issue as above, and I can confirm that renaming the file to `.service` and replacing the `[Mount]` section with

```
[Service]
Type=forking
ExecStart=/bin/bash -c ". /etc/profile;exec /storage/.kodi/addons/virtual.network-tools/bin/sshfs -o ro,allow_other,reconnect,ServerAliveInterval=15,IdentityFile=/storage/.ssh/id_rsa user@host:/some/dir /var/media/host/dir"
```

works. Thanks for the hints!

Looking forward to a fix that allows me to switch back to `.mount`, which seems cleaner!

heitbaum, 2023-07-27:
@reitzig The permanent fix will come along with #7602