LibreELEC/LibreELEC.tv #7615: [BUG] Fakeroot 1.30.1 has not been released

mattrude, 2023-03-15:
### Describe the bug
fakeroot version 1.30.1 has not been released and so can not be downloaded.

See: http://ftp.us.debian.org/debian/pool/main/f/fakeroot/

### To Reproduce
Steps to reproduce the behavior:
1. build from source via the make image command
2. See errors in log file

### Information
 - LE Version: 11.0.0.-devel
 - Hardware Platform: Source Code (All platforms)

### Log file
```
--2023-03-15 09:31:04--  http://ftp.debian.org/debian/pool/main/f/fakeroot/fakeroot_1.30.1.orig.tar.gz
Resolving ftp.debian.org (ftp.debian.org)... 2a04:4e42:44::644, 199.232.30.132
Connecting to ftp.debian.org (ftp.debian.org)|2a04:4e42:44::644|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2023-03-15 09:31:04 ERROR 404: Not Found.

--2023-03-15 09:31:04--  http://sources.libreelec.tv/mirror/fakeroot/fakeroot-1.30.1.tar.gz
Resolving sources.libreelec.tv (sources.libreelec.tv)... 138.68.75.163
Connecting to sources.libreelec.tv (sources.libreelec.tv)|138.68.75.163|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://sources.libreelec.tv/mirror/fakeroot/fakeroot-1.30.1.tar.gz [following]
--2023-03-15 09:31:05--  https://sources.libreelec.tv/mirror/fakeroot/fakeroot-1.30.1.tar.gz
Connecting to sources.libreelec.tv (sources.libreelec.tv)|138.68.75.163|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2023-03-15 09:31:05 ERROR 404: Not Found.

--2023-03-15 09:31:05--  http://ftp.debian.org/debian/pool/main/f/fakeroot/fakeroot_1.30.1.orig.tar.gz
Resolving ftp.debian.org (ftp.debian.org)... 2a04:4e42:44::644, 199.232.30.132
Connecting to ftp.debian.org (ftp.debian.org)|2a04:4e42:44::644|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2023-03-15 09:31:05 ERROR 404: Not Found.

--2023-03-15 09:31:05--  http://sources.libreelec.tv/mirror/fakeroot/fakeroot-1.30.1.tar.gz
Resolving sources.libreelec.tv (sources.libreelec.tv)... 138.68.75.163
Connecting to sources.libreelec.tv (sources.libreelec.tv)|138.68.75.163|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://sources.libreelec.tv/mirror/fakeroot/fakeroot-1.30.1.tar.gz [following]
--2023-03-15 09:31:06--  https://sources.libreelec.tv/mirror/fakeroot/fakeroot-1.30.1.tar.gz
Connecting to sources.libreelec.tv (sources.libreelec.tv)|138.68.75.163|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2023-03-15 09:31:06 ERROR 404: Not Found

```

### Additional context

```
diff --git a/packages/devel/fakeroot/package.mk b/packages/devel/fakeroot/package.mk
index 86481f6832..e354a3c0ac 100644
--- a/packages/devel/fakeroot/package.mk
+++ b/packages/devel/fakeroot/package.mk
@@ -3,8 +3,8 @@
 # Copyright (C) 2019-present Team LibreELEC (https://libreelec.tv)

 PKG_NAME="fakeroot"
-PKG_VERSION="1.30.1"
-PKG_SHA256="32ebb1f421aca0db7141c32a8c104eb95d2b45c393058b9435fbf903dd2b6a75"
+PKG_VERSION="1.31"
+PKG_SHA256="63886d41e11c56c7170b9d9331cca086421b350d257338ef14daad98f77e202f"
 PKG_LICENSE="GPL3"
 PKG_SITE="https://tracker.debian.org/pkg/fakeroot"
 PKG_URL="http://ftp.debian.org/debian/pool/main/f/fakeroot/${PKG_NAME}_${PKG_VERSION}.orig.tar.gz"
```

jernejsk, 2023-03-15:
I have it locally in sources folder, so at some point it had to be available. Maybe they retracted release? In any case, update is the way forward.

jernejsk, 2023-03-15:
argh, this is LE11 branch. Copy should be uploaded to our mirror.

vpeter4, 2023-03-17:
It is still available from https source: https://deb.debian.org/debian/pool/main/f/fakeroot/fakeroot_1.30.1.orig.tar.gz