LibreELEC/LibreELEC.tv #7672: [BUG] Build failure of inputstream.ffmpegdirect

heitbaum, 2023-03-20:
### Describe the bug

Build failure of inputstream.ffmpegdirect 

### To Reproduce
Build inputstream.ffmpegdirect 20.5.0

### Informations
 - LE Version: LE12
 - Hardware Platform: All

### Log file

```
/build/build.LibreELEC-ARMv7.arm-12.0-devel/build/inputstream.ffmpegdirect-20.5.0-Nexus/src/stream/FFmpegStream.cpp: In member function 'virtual std::string ffmpegdirect::FFmpegStream::GetStreamCodecName(int)':
/build/build.LibreELEC-ARMv7.arm-12.0-devel/build/inputstream.ffmpegdirect-20.5.0-Nexus/src/stream/FFmpegStream.cpp:2073:42: error: invalid conversion from 'const AVCodec*' to 'AVCodec*' [-fpermissive]
 2073 |     AVCodec* codec = avcodec_find_decoder(stream->codec);
      |                      ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~
      |                                          |
      |                                          const AVCodec*
ninja: build stopped: subcommand failed.
FAILURE: scripts/build inputstream.ffmpegdirect:target during make_target (default)
```
### Additional context

[build.LibreELEC-Generic-legacy.x86_64-12.0-devel-inputstream.ffmpegdirect_target-479.log](https://github.com/LibreELEC/LibreELEC.tv/files/11012687/build.LibreELEC-Generic-legacy.x86_64-12.0-devel-inputstream.ffmpegdirect_target-479.log)

HiassofT, 2023-03-20:
inputstream.ffmpegdirect hasn't been update to ffmpeg 5+  so this is an upstream bug - fix for LE will be to update to a fixed upstream version when it's available

lrusak, 2023-03-20:
> inputstream.ffmpegdirect hasn't been update to ffmpeg 5+  so this is an upstream bug - fix for LE will be to update to a fixed upstream version when it's available

In the Omega branch there is ffmpeg 5.x support.

HiassofT, 2023-03-20:
There's no Omega tag yet and even worse the version in addon.xml is still at the current Nexus release (20.5.0). So we can't pull in HEAD of the Omega branch as that'll create a version mess.

As soon as an Omega release gets tagged our update script will pick up the new version and this issue can be fixed.

lrusak, 2023-03-20:
> There's no Omega tag yet and even worse the version in addon.xml is still at the current Nexus release (20.5.0). So we can't pull in HEAD of the Omega branch as that'll create a version mess.
>
>
>
> As [...]

All you have to do is ask. https://github.com/xbmc/inputstream.ffmpegdirect/pull/233

closed 2023-03-21 by CvH

heitbaum, 2023-03-21:
> > There's no Omega tag yet and even worse the version in addon.xml is still at the current Nexus release (20.5.0). So we can't pull in HEAD of the Omega branch as that'll create a version mess.
> > As soon as an Omega release gets tagged our update script will pick up the new version and this issue can be fixed.
> 
> All you have to do is ask. [xbmc/inputstream.ffmpegdirect#233](https://github.com/xbmc/inputstream.ffmpegdirect/pull/233)

Thanks - PR merged 👍

PR #7676 (2023-03-21): kodi-binary-addons: update to latest versions
- inputstream.ffmpegdirect: update 20.5.0-Nexus to 21.0.0-Omega
- pvr.hts: update 20.6.1-Nexus to 21.0.1-Omega
- pvr.iptvsimple: update 21.1.1-Omega to 21.2.0-Omega
- pvr.vdr.vnsi: update 20.4.0-Nexus to 21.0.1-Omega

- Fixes #7672 
  - https://github.com/xbmc/inputstream.ffmpegdirect/pull/233