LibreELEC/LibreELEC.tv #7678: [BUG] When I try to connect to wifi reports "invalid-key" on Asus TinkerBoard (RK3288)

rgabbo, 2023-03-21:
### Describe the bug

When I try to connect to my wifi, it fails with the error “invalid-key”. I'm sure the key is correct.

### To Reproduce

Steps to reproduce the behavior:

1. Go to 'LibreELEC'
2. Go to 'Connections'
3. Try to connect to wifi
4. See the message "invalid-key"

### Informations

- LE Version: 11.0.0
- Hardware Platform: ASUS TinkerBoard

### Log file

Unable to follow the steps cause I can't connect to the internet

### Additional context

Freedom42082, 2023-03-22:
I have the same issue.

Informations

LE Version: 11.0.0
Hardware Platform: Rasberry Pi 4b 8-gig

Sebazzz, 2023-03-23:
Related: #7166

heitbaum, 2023-03-29:
Please retest with iwd2.4 - LibreELEC-yyyyy-11.0-nightly-20230329-a0f002b.img.gz (or newer) from https://test.libreelec.tv/11.0

chewitt, 2023-04-03:
I'm closing this ticket as the main discussion is taking place in #7166 - please follow up there.

closed 2023-04-03 by chewitt