LibreELEC/LibreELEC.tv #7680: [BUG]

Voosjey, 2023-03-22:
I can't watch 3D movies in 2D, if I switch to 2D, both sides of an HSBS movie remain. when I turn off 2D I suddenly get 4 pictures. I have an i5 12400 with a B660 chipset, with Libreelec 11.0.0. In the legacy version it still works, but then I don't have HDR.
[image]

heitbaum, 2023-09-04:
Closing as known. Understaning your situation is the Generic-legacy LE11 is working as expected. And Generic does not. Discussion at https://forum.kodi.tv/showthread.php?tid=372780

closed 2023-09-04 by heitbaum

Voosjey, 2025-06-04:
We now have LE13 in the nightly builds, and you still cannot watch 3D movies in 2D on the Generic x86x64. Do we all now have to convert our 3D movies into 2D so that we can watch them again? It must be possible to split an HSBS movie, especially since the 2D view is still integrated, and when you turn off 2D, it also doubles the screen, and when you then re-enable 2D, you only have the two images again. So there seems to be some function behind the 2D. It just doesn't work properly. Please check again.