LibreELEC/LibreELEC.tv #7698: [BUG] TigerVNC add-on not working with LE 11 Generic-legacy

ksooo, 2023-03-26:
### Describe the bug
As of https://forum.libreelec.tv/thread/26766-le11-and-vnc/ , TigerVNC add-on is supposed to work with LE 11 Generic-legacy. But it doesn't.

### To Reproduce
Steps to reproduce the behavior:
1. Install and enable TigerVNC add-on
2. From any VNC capable client machine, try to connect to your LE box to open a VNC session
3. See error

### Informations
 - LE Version: 11.0.1 Generic-legacy
 - Hardware Platform: x86_64

### Log file
service.tigervnc.service - TigerVNC
   Loaded: loaded (/storage/.config/system.d/service.tigervnc.service; enabled; preset: disabled)
   Active: activating (auto-restart) (Result: exit-code) since Sat 2023-03-25 12:56:56 CET; 1s ago
  Process: 4127 ExecStart=/bin/sh /storage/.kodi/addons/service.tigervnc/bin/tigervnc.start (code=exited, status=127)
  Main PID: 4127 (code=exited, status=127)
    CPU: 33ms
==========
x0vncserver: error while loading shared libraries: libXtst.so.6: cannot open shared object file: No such file or directory
==========
TV-Server:~/.kodi/addons/service.tigervnc/bin # ldd -r x0vncserver
	linux-vdso.so.1 => linux-vdso.so.1 (0x00007ffeaddd5000)
	libXtst.so.6 => not found
	libXdamage.so.1 => /usr/lib/libXdamage.so.1 (0x00007f5a0719f000)
	libXfixes.so.3 => /usr/lib/libXfixes.so.3 (0x00007f5a07197000)
	libX11.so.6 => /usr/lib/libX11.so.6 (0x00007f5a0706e000)
	libXext.so.6 => /usr/lib/libXext.so.6 (0x00007f5a0705b000)
	libz.so.1 => /usr/lib/libz.so.1 (0x00007f5a07040000)
	libgnutls.so.30 => /usr/lib/libgnutls.so.30 (0x00007f5a06e15000)
	libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x00007f5a06bfb000)
	libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x00007f5a06bdb000)
	libc.so.6 => /usr/lib/libc.so.6 (0x00007f5a06a1b000)
	libxcb.so.1 => /usr/lib/libxcb.so.1 (0x00007f5a069f4000)
	libidn2.so.0 => /usr/lib/libidn2.so.0 (0x00007f5a06999000)
	libnettle.so.8 => /usr/lib/libnettle.so.8 (0x00007f5a0694f000)
	libhogweed.so.6 => /usr/lib/libhogweed.so.6 (0x00007f5a06909000)
	libgmp.so.10 => /usr/lib/libgmp.so.10 (0x00007f5a06893000)
	/lib64/ld-linux-x86-64.so.2 => /usr/lib64/ld-linux-x86-64.so.2 (0x00007f5a071a5000)
	libm.so.6 => /usr/lib/libm.so.6 (0x00007f5a067b8000)
undefined symbol: XTestFakeMotionEvent	(./x0vncserver)
undefined symbol: XTestFakeButtonEvent	(./x0vncserver)
undefined symbol: XTestQueryExtension	(./x0vncserver)
undefined symbol: XTestGrabControl	(./x0vncserver)
undefined symbol: XTestFakeKeyEvent	(./x0vncserver)

### Additional context
Seems libXtst artifacts is not in the image

heitbaum, 2023-03-26:
- #7699 should fix the tigervnc missing library

ksooo, 2023-03-26:
Is there an LE11 zip containing the fix available for download, so I can test?

mglae, 2023-03-26:
No. As a immediate work around install Chrome Browser or JRE addon.

PR #7699 (2023-03-27): [le12] tigervnc: copy missing library libXtst to addon directory
- should fix #7698

looks better
```
unzip -v target/addons/11.80.0/Generic-legacy/x86_64/service.tigervnc/service.tigervnc-11.80.0.0.zip 
Archive:  target/addons/11.80.0/Generic-legacy/x86_64/service.tigervnc/service.tigervnc-11.80.0.0.zip
 Length   Method    Size  Cmpr    Date    Time   CRC-32   Name
--------  ------  ------- ---- ---------- ----- --------  ----
       0  Stored        0   0% 2023-03-26 12:46 00000000  service.tigervnc/
     908  Defl:N      462  49% 2023-03-26 12:46 2930faf0  service.tigervnc/addon.xml
        [... cut]
PR #7700 (2023-03-27): [le11] tigervnc: copy missing library libXtst to addon directory
- backport of #7699
- Fixes #7698
closed 2023-03-27 by CvH