LibreELEC/LibreELEC.tv #7731: Can't start LE11 on Orangepi one plus

ComradeSerg, 2023-03-30:
I tried to replace LE10 to LE11 but after sd card creation with "LibreELEC USB-SD Creator" LE11 can't start because of error:

*** Error in mount_flash: mount_common: Could not mount UUID=1803-1707 ***
### Starting debuggin shell for boot step: mount_flash... type exit to quit ###
#

but keyboard doesn't work.

I have OrangePi One Plus (allwinner H6), microSD ADATA 16GB UHS-I (class 10)

jernejsk, 2023-03-31:
Can you try to use Balena Etcher instead of LibreELEC USB-CD Creator and see if that works?

ComradeSerg, 2023-04-15:
Yes, before I tried Win32 with the same outcome, but now I installed Balena Etcher and nothing changed - Could not mount UUID=1803-1707 ***

jernejsk, 2023-04-16:
Hm... I manually mount LibreELEC-H6.arm-11.0.1-orangepi-one-plus.img and checked all those UUIDs. They match, so I'm not sure what could be wrong. I don't have the board, so I can't test it directly.

However, you can try to fix them:
1. execute `blkid` in that debug shell
2. check and update, if necessary, UUID values in extlinux/extlinux.conf

maartenjd, 2023-07-27:
I have the same issue on my orangepi-one-plus. LE10 boots normally. LE11 11.0.1 produces same error. Serial output differs from boot to boot.

**First boot:**
U-Boot SPL 2022.10 (Mar 18 2023 - 17:20:38 +0000)
DRAM: 1024 MiB
Trying to boot from MMC1
NOTICE:  BL31: v2.8(release):11.0.1-dirty
NOTICE:  BL31: Built : 16:10:38, Mar 18 2023
NOTICE:  BL31: Detected Allwinner H6 SoC (1728)
NOTICE:  BL31: Found U-Boot DTB at 0xa06f740, model: OrangePi One Plus
ERROR:   PMIC: Found unknown PMIC 00
ERROR:   RSB: read command: 0x103
ERROR:   RSB: write command: 0x3
ERROR:   RSB: read command: 0x3
ERROR:   RSB: write command: 0x3
ERROR:   RSB: read command: 0x3
ERROR:   RSB: write command: 0x3
ERROR:   RSB: read command: 0x3
ERROR:   RSB: write command: 0x3

U-Boot 2022.10 (Mar 18 2023 - 17:20:38 +0000) Allwinner Technology

CPU:   Allwinner H6 (SUN50I)
Model: OrangePi One Plus
DRAM:  1 GiB
Core:  48 devices, 15 uclasses, devicetree: separate
WDT:   Not starting watchdog@7020400
MMC:   mmc@4020000: 0
Loading Environment from FAT... Unable to read "uboot.env" from mmc0:1... 
In:    serial@5000000
Out:   serial@5000000
Err:   serial@5000000
Net:   No ethernet found.
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found /extlinux/extlinux.conf
Retrieving file: /extlinux/extlinux.conf
Ignoring unknown command: disk=UUID=e74196e7-acc6-47fe-9d35-e547de3eae47
Ignoring unknown command: console=tty1
1:	LibreELEC
Retrieving file: /KERNEL
append: boot=UUID=1803-1707 
Retrieving file: /sun50i-h6-orangepi-one-plus.dtb
Moving Image from 0x40080000 to 0x40200000, end=417c0000
   Flattened Device Tree blob at 4fa00000
   Booting using the fdt blob at 0x4fa00000
   Loading Device Tree to 0000000049ff4000, end 0000000049fff03c ... OK

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.1.19 (docker@244ea9df888c) (aarch64-none-elf-gcc-12.2.0 (GCC) 12.2.0, GNU ld (GNU Binutils) 2.39) #1 SMP PREEMPT Sat Mar 18 16:53:01 UTC 2023
...
[    0.000000] Kernel command line: boot=UUID=1803-1707 
[    0.000000] Unknown kernel command line parameters "boot=UUID=1803-1707", will be passed to user space.

*** Error in mount_flash: mount_common: Could not mount UUID=1803-1707 ***
Starting debugging shell for boot step: mount_flash... type  exit  to quit

**Hot reboot from debugging shell:**
reboot

[  744.543306] reboot: Restarting system

U-Boot SPL 2022.10 (Mar 18 2023 - 17:20:38 +0000)
DRAM: 1024 MiB
Trying to boot from MMC1
NOTICE:  BL31: v2.8(release):11.0.1-dirty
[... 82 lines cut ...]
187 bytes read in 6 ms (30.3 KiB/s)
1:	LibreELEC
Retrieving file: /KERNEL
23007240 bytes read in 1923 ms (11.4 MiB/s)
append: boot=UUID=0701-0042 disk=UUID=6675ba48-2766-4def-9e11-3d988e312d16 debug console=ttyS0,115200 console=tty1
Retrieving file: /sun50i-h6-orangepi-one-plus.dtb
31227 bytes read in 8 ms (3.7 MiB/s)
Moving Image from 0x40080000 to 0x40200000, end=41860000
   Flattened Device Tree blob at 4fa00000
   Booting using the fdt blob at 0x4fa00000
   Loading Device Tree to 0000000049ff5000, end 0000000049fff9fa ... OK

Starting kernel ...

[?r??????] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.10.161 (docker@2c3ea15273f4) (aarch64-none-linux-gnu-gcc.real (GNU Toolchain for the A-profile Architecture 10.2-2020.11 (arm-10.16)) 10.2.1 20201103, GNU ld (GNU Toolchain for the A-profile Architecture 10.2-2020.11 (arm-10.16)) 2.35.1.20201028) 
...
[    2.336113] Run /init as init process
[    2.339799]   with arguments:
[    2.342779]     /init
[    2.345063]   with environment:
[    2.348205]     HOME=/
[    2.350573]     TERM=linux
[    2.353290]     boot=UUID=0701-0042
[    2.356780]     disk=UUID=6675ba48-2766-4def-9e11-3d988e312d16

heitbaum, 2023-12-02:
@ComradeSerg / @maartenjd - is this still occurring on LE12 nightlies?

maartenjd, 2023-12-15:
Hello [heitbaum](https://github.com/heitbaum).

Erros(s) still occur. All with the same SD-card but previously the same error also with other SD-cards. See below. Is it U-boot related? Suggestions for any further troubleshooting?

Tested it with LibreELEC-H6.**aarch64**-12.0-nightly-20231214-8a29159-orangepi-one-plus and it seems to boot fine now (removed `quiet` from cmd line in extlinux.conf). See [bootlog](https://gist.github.com/maartenjd/493e49f45d65b0aa6f90c6a973097b8e#file-bootlog-libreelec-h6-aarch64-12-0-nightly-20231214-8a29159-orangepi-one-plus-txt).

Also tested LibreELEC-H6.**arm-11.0**-nightly-20231214-ec82496-orangepi-one-plus. See [bootlog](https://gist.github.com/maartenjd/9cb8a2588ab80ec8f92b28cb1961d6ef#file-bootlog-libreelec-h6-arm-11-0-nightly-20231214-ec82496-orangepi-one-plus-txt).

Finally also booted LibreELEC-H6.arm-10.0.4-orangepi-one-plus (stable). See [bootlog](https://gist.github.com/maartenjd/45b460d4fedd9e067ce3ecf52dcbebba#file-bootlog-libreelec-h6-arm-10-0-4-orangepi-one-plus-txt).

Unfortunately still the same "Error in mount_flash" with the 11 and 12 nightlies (see screenshot). 
[image]

Note the particularities in the 11 en 12 bootlogs, that do not show in 10:

- U-Boot (2023.10 and 2022.10):
ERROR:   PMIC: Found unknown PMIC 00
ERROR:   RSB: read command: 0x103
ERROR:   RSB: write command: 0x3
ERROR:   RSB: read command: 0x3
ERROR:   RSB: write command: 0x3
ERROR:   RSB: read command: 0x3
ERROR:   RSB: write command: 0x3

- U-boot:
Loading Environment from FAT... Unable to read "uboot.env" from mmc0:1...

- Kernel:
[    0.000000] Kernel command line: boot=UUID=1512-2531 disk=UUID=5830d71a-14e9-409c-b6b1-992f76f28ae7 console=ttyS0,115200 console=tty1
[    0.000000] **Unknown kernel command line parameters** "boot=UUID=1512-2531 disk=UUID=5830d71a-14e9-409c-b6b1-992f76f28ae7", will be passed to user space.

jernejsk, 2023-12-19:
So it seems that the issue is in PMIC bus switch from I2C (LE10, Linux 5.10) to RSB (LE11, Linux 6.1). PMIC is responsible for power regulation to most subsystems, including SD card. That would explain issues you see in boot log.

Currently the only idea that I have is to lower RSB frequency, since it's much higher than I2C. However, that needs to be changed in U-Boot and Linux. I'll build test image when I'll have some time.

maartenjd, 2024-01-24:
https://forum.armbian.com/topic/20227-gave-up-waiting-for-root-file-system-device/

maartenjd, 2024-04-28:
Issue continues to exist on Latest Beta Version 12.0 (2024.04). Did you already have a chance to look at the RSB-timings, [jernejsk](https://github.com/jernejsk)? I made some build with some uneducated guesses for timings, but to no avail.

maartenjd, 2024-05-04:
Today I built LE12 with [ Orangepi One Plus Rollback to i2c patch](https://github.com/armbian/build/commit/0df2112319ec8f898632eacd70c0b24c6f020b9d).
Had a few successful boots. Will provide serial output later on.

maartenjd, 2024-05-31:
[ComradeSerge](https://github.com/ComradeSerg)
Can you try this image, compiled with i2c rollback patch?
[link to image](https://nextcloud.poortman.dynu.net:8443/s/T8iGdBrbneLAo64)

heitbaum, 2024-10-29:
Is this issue still occurring?

maartenjd, 2024-10-29:
Yes, still the same issue with LibreELEC-H6.aarch64-12.0-devel-20241029210940-a0b3874-orangepi-one-plus.

I2C-patch applied in:

./build.LibreELEC-H6.aarch64-12.0-devel/build/linux-6.6.46/arch/arm64/boot/dts/allwinner/sun50i-h6-orangepi.dtsi
./build.LibreELEC-H6.aarch64-12.0-devel/build/u-boot-2024.01/arch/arm/dts/sun50i-h6-orangepi.dtsi

U-Boot SPL 2024.01 (Oct 29 2024 - 20:37:19 +0100)
DRAM: 1024 MiB
Trying to boot from MMC1
NOTICE:  BL31: v2.10.0	(release):12.0.0
NOTICE:  BL31: Built : 16:58:06, May  3 2024
NOTICE:  BL31: Detected Allwinner H6 SoC (1728)
NOTICE:  BL31: Found U-Boot DTB at 0xa070328, model: OrangePi One Plus
ERROR:   PMIC: Found unknown PMIC 00
ERROR:   RSB: read command: 0x103
ERROR:   RSB: write command: 0x3
ERROR:   RSB: read command: 0x3
ERROR:   RSB: write command: 0x3
ERROR:   RSB: read command: 0x3
ERROR:   RSB: write command: 0x3

With previous version I sometimes had succesfull boots. This time 0 out of 6.

heitbaum, 2025-02-27:
Is this still an issue with the LE13 nightly images? Latest will be a 202502xx image.

maartenjd, 2025-02-27:
I will bake a build and check.

maartenjd, 2025-03-01:
Problem still exists with LibreELEC-H6.aarch64-13.0-nightly-20250226-23a3c47-orangepi-one-plus-image.

See bootlog below. I added the environment-settings as well (I managed to interrupt the 0 seconds bootdelay and changed it to 5s). Also the output of some uboot commands (mmc, fatls and ext2ls).

On the attached display same notification as earlier:
`*** Error in mount_flash: mount_common : Could not mount UUID=2702-5016 ***`
`### Starting debugging shell foor boot step mount_flash.... type exit to quit ### `
`#`

==============================================================

Image: LibreELEC-H6.aarch64-13.0-nightly-20250226-23a3c47-orangepi-one-plus

U-Boot SPL 2025.01 (Feb 27 2025 - 00:59:05 +0000)
DRAM: 1024 MiB
Trying to boot from MMC1
NOTICE:  BL31: v2.12.0(release):23a3c47-dirty
NOTICE:  BL31: Built : 23:54:48, Feb 26 2025
NOTICE:  BL31: Detected Allwinner H6 SoC (1728)
NOTICE:  BL31: Found U-Boot DTB at 0xa071c18, model: OrangePi One Plus
**ERROR:   RSB: read command: 0x103**

U-Boot 2025.01 (Feb 27 2025 - 00:59:05 +0000) Allwinner Technology

CPU:   Allwinner H6 (SUN50I)
Model: OrangePi One Plus
DRAM:  1 GiB
Core:  49 devices, 16 uclasses, devicetree: separate
WDT:   Not starting watchdog@7020400
MMC:   mmc@4020000: 0
Loading Environment from FAT... Unable to read "uboot.env" from mmc0:1... 
In:    serial@5000000
Out:   serial@5000000
Err:   serial@5000000
Net:   No ethernet found.
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found /extlinux/extlinux.conf
Retrieving file: /extlinux/extlinux.conf
1:	LibreELEC
Retrieving file: /KERNEL
append: boot=UUID=2702-5016 disk=UUID=792f8921-4262-4964-b62b-038824c25c30 console=ttyS0,115200 console=tty1
Retrieving file: /sun50i-h6-orangepi-one-plus.dtb
Moving Image from 0x40080000 to 0x40200000, end=0x41830000
## Flattened Device Tree blob at 4fa00000
   Booting using the fdt blob at 0x4fa00000
Working FDT set to 4fa00000
   Loading Device Tree to 0000000049ff4000, end 0000000049fff7b7 ... OK
Working FDT set to 49ff4000

Starting kernel 

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.12.15 (ubuntu@2dc9ca5c08b1) (aarch64-libreelec-linux-gnu-gcc-14.2.0 (GCC) 14.2.0, GNU ld (GNU Binutils) 2.44) #1 SMP PREEMPT Thu Feb 27 00:33:41 UTC 2025
[    0.000000] Machine model: OrangePi One Plus
[    0.000000] OF: reserved mem: Reserved memory: No reserved-memory node in the DT
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] cma: Reserved 384 MiB at 0x0000000066c00000 on node -1
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[... 270 lines cut ...]
fdtfile=allwinner/sun50i-h6-orangepi-one-plus.dtb
fdtoverlay_addr_r=0x4FE00000
kernel_addr_r=0x40080000
kernel_comp_addr_r=0x44000000
kernel_comp_size=0xb000000
loadaddr=0x42000000
mmc_boot=if mmc dev ${devnum}; then devtype=mmc; run scan_dev_for_boot_part; fi
mmc_bootdev=0
partitions=name=loader1,start=8k,size=32k,uuid=${uuid_gpt_loader1};name=loader2,size=984k,uuid=${uuid_gpt_loader2};name=esp,size=128M,bootable,uuid=${uuid_gpt_esp};name=system,size=-,uuid=${uuid_gpt_system};
pxefile_addr_r=0x4FD00000
ramdisk_addr_r=0x4FF00000
scan_dev_for_boot=echo Scanning ${devtype} ${devnum}:${distro_bootpart}...; for prefix in ${boot_prefixes}; do run scan_dev_for_extlinux; run scan_dev_for_scripts; done;
scan_dev_for_boot_part=part list ${devtype} ${devnum} -bootable devplist; env exists devplist || setenv devplist 1; for distro_bootpart in ${devplist}; do if fstype ${devtype} ${devnum}:${distro_bootpart} bootfstype; then part uuid ${devtype} ${devnum}:${distro_bootpart} distro_bootpart_uuid ; run scan_dev_for_boot; fi; done; setenv devplist
scan_dev_for_extlinux=if test -e ${devtype} ${devnum}:${distro_bootpart} ${prefix}${boot_syslinux_conf}; then echo Found ${prefix}${boot_syslinux_conf}; run boot_extlinux; echo EXTLINUX FAILED: continuing...; fi
scan_dev_for_scripts=for script in ${boot_scripts}; do if test -e ${devtype} ${devnum}:${distro_bootpart} ${prefix}${script}; then echo Found U-Boot script ${prefix}${script}; run boot_a_script; echo SCRIPT FAILED: continuing...; fi; done
scriptaddr=0x4FC00000
serial#=82c00007a65f027d
soc=sunxi
stderr=serial@5000000
stdin=serial@5000000
stdout=serial@5000000
uuid_gpt_esp=c12a7328-f81f-11d2-ba4b-00a0c93ec93b
uuid_gpt_system=b921b045-1df0-41c3-af44-4c6f280d3fae

Environment size: 2824/65532 bytes

jernejsk, 2025-03-15:
Sorry it took me so long to get back to this, but I have now test image to fix this issue. Can you try this one?
https://jernej.libreelec.tv/test/one-plus/

It was build from https://github.com/jernejsk/LibreELEC.tv/commits/aw_i2c_fixes/

If it works, please also test sleep and wake up. For example, go to sleep via power menu and wake it up via power button on the board.

maartenjd, 2025-03-16:
Hello Jernej. Thanks for your efforts. System boots OK now. See attached log.

Suspend via Kodi-menu works, then wake-up with HW button works OK too.

Power-off from running Kodi with HW-button OK. Power-on with HW-button from powerdown state OK (although I the first time there was an issue (see log) but that seems/seemed to be Kodi.bin-related.

I can control with USB-keyboard and mouse. However, control via CEC not working. May also be Kodi-related? Or issue with HDMI-connector? Tried LE10 and there CEC-control works.

Once again thank you for solving/mitigating this RSB-issue, Jernej!

[Boot-LibreELEC-H6.aarch64-13.0-devel-20250315190025-1bcaafe-orangepi-one-plus.txt](https://github.com/user-attachments/files/19273818/Boot-LibreELEC-H6.aarch64-13.0-devel-20250315190025-1bcaafe-orangepi-one-plus.txt)

jernejsk, 2025-03-16:
Interesting, could be some kernel related stuff, but I need to check. Anyway, please open separate issue for CEC.

Thanks for testing!

closed 2025-03-16 by HiassofT